arendst/Tasmota #6325: Request add device energy meter dds578r

lzahradil, 2019-08-30:
Hi, 
is it possible to add support for the electricity meter DDS578R?

MODBUS-RTU communication protocol Description and manual: https://www.dropbox.com/s/o0csudhh3otop8l/DDS578R.pdf?dl=0

RS485 Modbus Rtu DIN Rail 3 Phase 4 Wire LCD Digital Wattmeter Electronic Power Consumption Energy Meter 5-100A 380V AC 50Hz
https://s.click.aliexpress.com/e/5w4aliIM

Jason2866, 2019-08-31:
Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-08-31 by lzahradil

MiG-41, 2020-03-31:
Modbus communication with these meters like DDS6619-017, DDS6619-016, DDS578MR, DDS578R , DDS024MR and modifications of reaper7/SDM_Energy_Meter is possible. Managed to do in DDS024MR.... Most important things are:
- modifing addresses to correspond from manual
- modifing node adres of counter ( default is 01 , but these have different)
- change the serial communication to Even parity

vakarthik, 2020-07-07:
@MiG-41 I have a DDS024MR meter. checked with modscan and output is fine. I use the following script in tasmota (compiled as per instructions in docs)

`>D

>B
=>sensor53 r

>M 1
+1,3,M,0,9600,MODBUS,1,1,18040000,18040002,18040004`

my meter address is 18. and using nodemcu default Rx, Tx pins and max485 for conversion. 

Not getting any data. any help is appreciated

kugelkopf123, 2020-07-07:
So you get no output on `Sensor53 d1` ?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

vakarthik, 2020-07-07:
> 
> 
> So you get no output on `Sensor53 d1` ?
> 
> Sent with [GitHawk](http://githawk.com)

i get this 

```
17:43:11 CMD: sensor53 d18
17:43:11 RSL: stat/tasmota_3906F7/RESULT = {"Time":"2020-07-07T17:43:11","SML":{"CMD":"dump: 1"}}
```

kugelkopf123, 2020-07-07:
> > So you get no output on `Sensor53 d1` ?
> > Sent with [GitHawk](http://githawk.com)
> 
> i get this
> 
> ```
> 17:43:11 CMD: sensor53 d1
> 17:43:11 RSL: stat/tasmota_3906F7/RESULT = {"Time":"2020-07-07T17:43:11","SML":{"CMD":"dump: 1"}}
> ```

Okay. I cant find any datasheet or Manual with Modbus Registers for this one. Normally its ID,ReadRegister,RegisterNumber -1.
Oh and btw. When you use the "M" (upper case) your Serial connection is 8E1 for Modbus instead of "m" 8N1 for Modbus.  So perhaps thats your problem? You should have a look in the Manual and use the right parity for your Device. 

With "Sensor53 d1" you can see what you modbus device returns. While using the dumplog, the Values in the Main Menu doesnt change. You have to stop Dump logging via "sensor53 d0"

But first lets have a look on what the Meter returns.

vakarthik, 2020-07-08:
thanks for hte input.
i get output from modscan using the following settings: baud: 9600, even parity, 8 bits and one stop bit

```
04:39:17 CMD: sensor53 d1
04:39:17 RSL: stat/tasmota_3906F7/RESULT = {"Time":"2020-07-08T04:39:17","SML":{"CMD":"dump: 1"}}
04:39:27 CMD: sensor53 d0
04:39:27 RSL: stat/tasmota_3906F7/RESULT = {"Time":"2020-07-08T04:39:27","SML":{"CMD":"dump: 0"}}
```

i dont see any output from meter. is the above commands correct ? 
this is my wiring 
https://imgur.com/a/AqMshsM

when i try to connect using modscan, when connection established, a small phone symbol appears on the meter. When iconnect using nodemcu i dont get anything. should i check the wiring. i get 3.3Vwith respect to GND in both Rx and Tx

kugelkopf123, 2020-07-08:
You are missing the descriptor.

Try the following.
```
>D
>B
=>sensor53 r
>M 1
+1,3,m,0,9600,MODBUS,1,5,18040000,18040002,18040004
1,180400UUuuxxxxxxxx@i0:1,Voltage L1,V,Voltage_L1,0
# 
```

The line: 
`1,180400UUuuxxxxxxxx@i0:1,Voltage L1,V,Voltage_L1,0`

is just a role model that you can use for the time being so that the Wemos doesn't crash. If you are using a Wemos d1 Mini or Nodemcu, set your device type to Generic(18) in the device settings.

For 8N1 parity, you should use the lowercase letter "m". 

GPIO3 and GPIO1 are perfectly fine. However, I would set the query interval a bit longer in the beginning. I set it to 5 in the example. 5x100ms waiting time. 
Is the ID 18 already a hexadecimal number or is it decimal?
I'm not sure if the device ID has to be transmitted as HEX. I think so. But I only use the device ID "2", so I can't say anything specific.

`Sensor53 d1` is only for debugging, or to find out if you get an answer from the counter. If yes, what it answers you. With this answer, you can adjust your descriptor. BTW. the "1" from "d1" is the number of the Meter that you had defined in your skript. For you its the frist. So 1 is right.
The descriptor compares what you receive with the line you entered in the script. 
Therefore you should first find out with Sensor53 d1 what the answer from your counter is. 
For example:
counter answer in Webconsole with sensor53 d1: `02 03 04 03 78 ff eb 48 d1`
my Descirptor: 
`5,020304UUuuxxxxxxxxxx@i6:100,Power Total,kW,Power_Total,3`

UUuu are the places I read out and the places marked with "x" are ignored. "UUuu extract an unsigned word (high order byte first)" There are several possibilities depending on how your counter works.
You can find out which options are available in the Tasmota Docs. 

If you end the debugging with `sensor53 d0`, the descriptor is used and you will see the corresponding line and value in the main menu. 

Good luck!

vakarthik, 2020-07-08:
thanks for the help. much appreciated. 
@ set your device type to Generic(18) in the device settings.
i tried setting it to generic(18). but it reverts back to sonoff basic. not sure why. 

```
>D
>B
=>sensor53 r
>M 1
+1,3,M,0,9600,MODBUS,1,5,12040000,12040002,12040004
1,120400UUuuxxxxxxxx@i0:1,Voltage L1,V,Voltage_L1,0
1,120400UUuuxxxxxxxx@i1:1,Voltage L2,V,Voltage_L2,0
1,120400UUuuxxxxxxxx@i2:1,Voltage L3,V,Voltage_L3,0
# 
```
The above script was not working. (device id hex value is 12) now the crash is resolved and script stays.

`sensor53 d1` does not return any value.

I will try with some other board and different max485 chip and update this issue.

kugelkopf123, 2020-07-08:
> thanks for the help. much appreciated.
> @ set your device type to Generic(18) in the device settings.
> i tried setting it to generic(18). but it reverts back to sonoff basic. not sure why.
>
> ``` [...]

That your board crashes when you change the device property to generic is not normal. You should check again if it's the board or  you should completely erase the board and then upload a fresh tasmota version. 

And: You still use 8E1 as parity. Is this your intention? Because I think your meter uses 8N1.  Like this:

```
>D
>B
=>sensor53 r
>M 1
+1,3,m,0,9600,MODBUS,1,5,12040000,12040002,12040004
1,120400UUuuxxxxxxxx@i0:1,Voltage L1,V,Voltage_L1,0
1,120400UUuuxxxxxxxx@i1:1,Voltage L2,V,Voltage_L2,0
1,120400UUuuxxxxxxxx@i2:1,Voltage L3,V,Voltage_L3,0
# 
```

zeljko-ciric, 2025-11-30:
Maybe is little bit late but may help someone:

In user_cofig_override.h add:

> #ifndef USE_SCRIPT
> #define USE_SCRIPT
> #endif
> #ifndef USE_SML_M
> #define USE_SML_M
> #endif
> #ifdef USE_RULES
> #undef USE_RULES
> #endif
> #ifdef USE_MODBUS_BRIDGE
> #undef USE_MODBUS_BRIDGE
> #endif
> #ifdef USE_MODBUS_BRIDGE_TCP
> #undef USE_MODBUS_BRIDGE_TCP
> #endif

1. Use r(aw): +1,12,m,0,9600,,13,5,r010400000040 to read first 64 registers + 01040100,01040400 for energy total 
2. Expand buffer to 256 (1,=so3,256,512)
1,010480ffffffff@i0:1,Voltage P1,V,Voltage_P1,2
read 1st answer
1,010480x4ffffffff@i0:1,Voltage P2,V,Voltage_P2,2 
use x4 to skip 4 Bytes (N is decimal number!!!), then convert 4 bytes to float (ffffffff)

Last two are little bit diferent.

> >D
> >B
> =>sensor53 r
> =>sensor53 l2
> ; disable publishing at MQTT tele period on boot
> smlj=0
> >S
> ; re-enable publishing at MQTT tele period after 10 seconds
> if upsecs>10
> then
> smlj|=1
> endif
> >M 1
> 
> +1,12,m,0,9600,,13,5,r010400000040,01040100,01040400
> 1,=so3,256,512
> 1,010480ffffffff@i0:1,Voltage P1,V,Voltage_P1,2
> 1,010480x4ffffffff@i0:1,Voltage P2,V,Voltage_P2,2
> 1,010480x8ffffffff@i0:1,Voltage P3,V,Voltage_P3,2
> 
> 1,010480x12ffffffff@i0:1,Current,A,Current,2
> 1,010480x16ffffffff@i0:1,Current P1,A,Current_P1,2
> 1,010480x20ffffffff@i0:1,Current P2,A,Current_P2,2
> 1,010480x24ffffffff@i0:1,Current P3,A,Current_P3,2
> 
> 1,010480x32ffffffff@i0:1,Total Active Power,W,Total_Active_Power,2
> 1,010480x36ffffffff@i0:1,Active Power P1,W,Active_Power_P1,2
> 1,010480x40ffffffff@i0:1,Active Power P2,W,Active_Power_P2,2
> 1,010480x44ffffffff@i0:1,Active Power P3,W,Active_Power_P3,2
> 
> 1,010480x48ffffffff@i0:1,Total Reactive Power,VAr,Total_Reactive_Power,2
> 1,010480x52ffffffff@i0:1,Reactive Power P1,VAr,Reactive_Power_P1,2
> 1,010480x56ffffffff@i0:1,Reactive Power P2,VAr,Reactive_Power_P2,2
> 1,010480x60ffffffff@i0:1,Reactive Power P3,VAr,Reactive_Power_P3,2
> 
> 1,010480x64ffffffff@i0:1,Total Apparent Power,W,Total_Apparent_Power,2
> 1,010480x68ffffffff@i0:1,Apparent Power P1,VA,Apparent_Power_P1,2
> 1,010480x72ffffffff@i0:1,Apparent Power P2,VA,Apparent_Power_P2,2
> 1,010480x76ffffffff@i0:1,Apparent Power P3,VA,Apparent_Power_P3,2
> 
> 1,010480x80ffffffff@i0:1,Power Factor,,Power_Factor,2
> 1,010480x84ffffffff@i0:1,Power Factor P1,,Power_Factor_P1,2
> 1,010480x88ffffffff@i0:1,Power Factor P2,,Power_Factor_P2,2
> 1,010480x92ffffffff@i0:1,Power Factor P3,,Power_Factor_P3,2
> 
> 1,010480x108ffffffff@i0:1,Frequency,Hz,Frequency,2
> 
> 1,010404ffffffff@i1:1,Active Energy,kWh,Active_Energy,2
> 
> 1,010404ffffffff@i2:1,Reactive Energy,kVArh,Reactive_Energy,2
> #