arendst/Tasmota #6331: STATUS not reporting sensor/counter data on generic type module

kandev, 2019-08-31:
Program Version	6.6.0(sensors)
Build Date & Time	2019-07-06T13:40:10
Core/SDK Version	2_4_2/2.2.1(cfd48f3)

Flashed on ESP8266. I have Module Generic(18) and set up AM2301 sensor on GPIO14, counters on GPIO4 and GPIO5. Both thesensor and the counters works - values visible in web GUI and acurately sent via mqtt telemetry (.../tele/SENSOR) but are missing from the STATUS report.

Missing that data makes gathering the sensor status via http impossible.

Jason2866, 2019-08-31:
Itn s there 
```
18:39:41 MQT: tele/sonoff-5D3631/STATE = {"Time":"2019-08-31T18:39:41","Epoch":1567269581,"Uptime":"3T05:50:33","UptimeSec":280233,"Vcc":2.695,"Heap":29,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"Jason_Home_WLAN","BSSId":"00:A0:57:2A:BD:19","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:05"}}
18:39:41 MQT: tele/sonoff-5D3631/SENSOR = {"Time":"2019-08-31T18:39:41","Epoch":1567269581,"AM2301":{"Temperature":25.3,"Humidity":70.4},"BME280":{"Temperature":25.4,"Humidity":60.5,"Pressure":970.6,"SeaPressure":1014.0},"PressureUnit":"hPa","TempUnit":"C"}

```

kandev, 2019-08-31:
Yup. My mistake. All there!
Thank you!

closed 2019-08-31 by kandev