arendst/Tasmota #6338: OBI2 Plug (Type 61) Led1 does not follow Relay1

rubberduck09, 2019-09-02:
As in Issue [4567](https://github.com/arendst/Sonoff-Tasmota/issues/4567) i have the issue that the Power LED defined as Led1 does not reflect the Status of Relay1.

The Second Led in this device has been defined as LedLink and does exactly what it should do.

I tried configure a own template and if i redefine Led1 as Relay2 then i get a second switchbutton in the Web-Interfaces which perfectly switches the LED.

What also worked then with the Led defined as Relay2:
seriallog 0
rule1 on power1#state do power2 %value% endon
rule1 1
ledmask 0x0000
setoption13 1

Then the Led follows the Relay1. But then there is a hyperliquid 2nd Button in the WebIf.

effelle, 2019-09-02:
@rubberduck09 when you open a bug or future request you need to fill the template or your issue will be marked as invalid.
@arendst and @ascillato: he discussed with me on our Discord, and I have adapted one of our rules on Cookbook to suit his needs. I'm not sure if there is another solution to this configuration.

arendst, 2019-09-02:
What means led1 does not follow relay1?

Does it mean if relay is on, led is off and reverse? If so, simply define led as ``Led1i`` instead of ``Led1``.

rubberduck09, 2019-09-02:
If you switch on or off the Relay1 then the Led1 does not go on and off following the status of the Relay1

arendst, 2019-09-02:
In that case pls provide output of commands ``status 0`` and ``template``

rubberduck09, 2019-09-02:
17:42:26 CMD: status 0
17:42:26 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"0000","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:42:26 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"Software/System restart","Uptime":"0T00:51:30","StartupUTC":"2019-09-02T14:50:56","Sleep":50,"CfgHolder":4617,"BootCount":32,"SaveCount":163,"SaveAddress":"F4000"}}
17:42:26 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019.07.06 13:17:19","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
17:42:26 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["0118 999 881 999 119 7253",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000A009","280500000100060000000000000000000000","00000000"]}}
17:42:26 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":504,"Free":496,"Heap":14,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540A1","FlashMode":3,"Features":["00000407","0FDAE396","001783A0","22B617CD","01003BC0"]}}
17:42:26 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3007","IPAddress":"10.117.17.165","Gateway":"10.117.17.2","Subnetmask":"255.255.255.0","DNSServer":"10.117.17.1","Mac":"84:F3:EB:AA:EB:BF","Webserver":2,"WifiConfig":4}}
17:42:26 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AAEBBF","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
17:42:26 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 02 15:42:26 2019","Local":"Mon Sep 02 17:42:26 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"06:33","Sunset":"20:06"}}
17:42:26 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-02T17:42:26"}}
17:42:26 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-02T17:42:26","Uptime":"0T00:51:30","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"0118 999 881 999 119 7253","BSSId":"00:18:0A:32:BD:EF","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}}
17:42:30 CMD: template
17:42:30 RSL: stat/sonoff/RESULT = {"NAME":"Selbstgebaut","GPIO":[0,0,0,0,21,17,0,0,158,52,0,0,0],"FLAG":0,"BASE":61}

arendst, 2019-09-02:
Can't see anything wrong. Nice issue. You say "Hausgemacht" so you have a schematic how you connected the hardware together?

arendst, 2019-09-02:
Or you might want to upgrade to development version 6.6.0.9 first as things might have changed.

rubberduck09, 2019-09-02:
"Selbsgebaut" is only the template - i did not change the hardware. only soldiered in some wires to be able to flash.
Do you have a sonoff_de.bin file for me to test?

ascillato, 2019-09-02:
>Do you have a sonoff_de.bin file for me to test?

You can follow this steps of the wiki to compile Tasmota. If you don't want to install any software in your PC you can use online compilers like Gitpod or TasmoCompiler. More info in the wiki at: https://github.com/arendst/Sonoff-Tasmota/wiki/Flashing#flashing-and-compiling-from-source

effelle, 2019-09-02:
We don't make builds in foreign languages for develop branch. You can download the latest dev from [here](http://thehackbox.org/tasmota/) or use our [gitpod](https://github.com/arendst/Sonoff-Tasmota/wiki/Compiling-Tasmota-on-Gitpod) compiler.

rubberduck09, 2019-09-02:
LED works with this update. Thank you very much.

closed 2019-09-02 by rubberduck09