arendst/Tasmota #6339: Shelly 2.5 switch interlock not working

amdnox, 2019-09-02:
### ISSUE DESCRIPTION - TROUBLESHOOTING
I flashed the Shelly 2.5 and issued the template 
{"NAME":"Shelly 2.5","GPIO":[56,255,17,255,21,83,0,0,6,82,5,22,156],"FLAG":2,"BASE":18}
as well as 
rule1 on energy#current>0.500 do backlog power1 0; power2 0 endon
rule 1
rule 5
and 

SwitchMode 1

via the console.
Now there is still the 2 Toggle buttons on the main page and I can still switch both relays on at the same time.
Thats not good!!  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : _____
- [x ] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [x ] Development IDE - Compiler / Upload tools used : ____ / ____
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

18:08:14 CMD: STATUS 0
18:08:14 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:08:14 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"External System","Uptime":"0T00:29:19","StartupUTC":"2019-09-02T16:38:55","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":48,"SaveAddress":"F9000"}}
18:08:14 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:08:14 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HSAP",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
18:08:14 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
18:08:14 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"rollo","IPAddress":"192.168.20.61","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.1","Mac":"A4:CF:12:BA:6E:CD","Webserver":2,"WifiConfig":4}}
18:08:14 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BA6ECD","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:08:14 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 02 17:08:14 2019","Local":"Mon Sep 02 18:08:14 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:08","Sunset":"19:31"}}
18:08:14 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:08:14 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-02T18:08:14","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":46.4},"ENERGY":{"TotalStartTime":"2019-09-02T16:34:04","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000},"TempUnit":"C"}}
18:08:14 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-02T18:08:14","Uptime":"0T00:29:19","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"HSAP","BSSId":"F0:9F:C2:27:D2:AB","Channel":11,"RSSI":62,"LinkCount":2,"Downtime":"0T00:00:07"}}}
```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
18:08:14 CMD: STATUS 0
18:08:14 RSL: stat/sonoff/STATUS = {"Status":{"Module":0,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
18:08:14 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-basic.bin","RestartReason":"External System","Uptime":"0T00:29:19","StartupUTC":"2019-09-02T16:38:55","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":48,"SaveAddress":"F9000"}}
18:08:14 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:08:14 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["HSAP",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
18:08:14 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540EF","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
18:08:14 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"rollo","IPAddress":"192.168.20.61","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.1","Mac":"A4:CF:12:BA:6E:CD","Webserver":2,"WifiConfig":4}}
18:08:14 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_BA6ECD","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
18:08:14 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Sep 02 17:08:14 2019","Local":"Mon Sep 02 18:08:14 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:08","Sunset":"19:31"}}
18:08:14 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
18:08:14 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-02T18:08:14","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature":46.4},"ENERGY":{"TotalStartTime":"2019-09-02T16:34:04","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000},"TempUnit":"C"}}
18:08:14 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-02T18:08:14","Uptime":"0T00:29:19","Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"HSAP","BSSId":"F0:9F:C2:27:D2:AB","Channel":11,"RSSI":62,"LinkCount":2,"Downtime":"0T00:00:07"}}}

```
**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-09-02:
How or where did you tell the software to use interlock?

ascillato2, 2019-09-02:
@amdnox

Have you tried in the console the command `Interlock` ? (https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#control)

amdnox, 2019-09-02:
I fixed my issue 
I entered interlock 0 and afterwards interlock 1 into the console and now it works properly.
Maybe it could be written at the wiki of the 2.5? or am I stupid and over read it?

ascillato2, 2019-09-02:
Great to know that now it is working for you.

closed 2019-09-02 by ascillato2