arendst/Tasmota #6340: Shelly 2.5 ENERGY_POWER_LIMIT only works for output 1

cinnayyannic, 2019-09-02:
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Shelly 2.5
- [x] Tasmota binary firmware version number used : 6.6.0.x / (pre-compiled)

### TO REPRODUCE
Set a power limit value. Connect load to output 2 of Shelly 2.5 > power limit value

### EXPECTED BEHAVIOUR
Output 2 should be turned off if power limit is reached.
I think the problem is that for Shelly 2.5 the power is measured as sum of output 1 and 2.
It would be nice to have a separate power measurement for each output.

If there is power measurement per output the power limit value should be the max power / output. If max power / output is reached the corresponding output should be turned off.

If individual measurement per output is not possible i would expect that at least both outputs are turned off if them sum of both outputs is > power limit value.

ascillato2, 2019-09-02:
Closing this issue as it is **duplicated** from #6160. Sorry.

Please, read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

cinnayyannic, 2019-09-02:
I dont think it is a duplicate of #6160 

The issue is not about dual power metering but about the non functional power limit.
Even without dual power metering i would expect that both outputs are turned off for safety reasons if the power limit is reached. 

For over temperature it is done the right way
`SetAllPower(POWER_ALL_OFF, SRC_OVERTEMP);`

In case of over temperature both outputs are turned off. I think the same should be done for the power limit until dual metering is implemented.

arendst, 2019-09-03:
Will implement SetAllPower too in near future.

jziolkowski, 2019-09-03:
Will that command be available to end user, say as POWERALL?

cinnayyannic, 2019-09-03:
First of all thanks for the fast fix 👍 
Now in case of overpower SetAllPower will take care that all channels are turned off.

But i think now the Max power retry acts wrong.
Lets say output 1 was turned on and output 2 was turned off. Output1 is causing the over power situation. Now max power retry will kick in and turn both outputs on.  SetAllPower(POWER_ALL_ON, SRC_MAXPOWER);

Also not really a good behaviour. I think the power should be restored to the last known user requested output state.

arendst, 2019-09-03:
@jziolkowski just added command ``power0 0/1/2/Off/On/Toggle`` to control all outputs at once.

cinnayyannic, 2019-09-03:
Great should work now unless user changed power state while in over power retry condition.

Just an idea:
```
sonoff.ino

power_t lastPoweredState = POWER_ALL_OFF

void SetDevicePower(power_t rpower, int source)
{
   ...
      if (count > 1) {
        mask = ~Settings.interlock[i];    // Turn interlocked group off as there would be multiple relays on
        power &= mask;
        rpower &= mask;
      }
    }
  }
   
  ...

  if (rpower != POWER_ALL_OFF)
  {
      lastPoweredState = rpower; // remember last known applyed powered ON state
  }

  ...
}

void RestoreAllPower(int source)
{
  if (power != lastPoweredState) {
    SetDevicePower(lastPoweredState, source);
    MqttPublishAllPowerState();
  }
}
```
Afterwards just call RestoreAllPower(SOURCE) no need to remember state in xdrv_03_energy.ino or any future source and always latest power ON state.

arendst, 2019-09-04:
Thx. Done.

closed 2019-09-04 by cinnayyannic