arendst/Tasmota #6347: Teckin SP22 - Power monitor not working

VinceMasuka, 2019-09-03:
Hi

I've some Teckin SP22 sockets over here. Funny thing is, I recently flashed v6.6.0.9 [version: sonoff, STAGE/2.2.2-dev(38a443e)] on one, recognizing that power monitor is not working anymore. I've some other SP22 around [holding v6.3.0.15, 2_3_0/1.5.3(aec24ac9)] and power monitor is just working fine there. Sockets are configured to "Teckin (52)". I re-flashed the faulty one over and over again, but it won't do anything on the power monitor side. I'm neither getting the picture what went wrong with this SP22 nor what to do to resolve it.

Any suggestions would be greatly appreciated!

Regards
Vince

ascillato2, 2019-09-04:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

VinceMasuka, 2019-09-04:
I think, I'll just crack it up and flash it via serial. Thanks.

closed 2019-09-04 by VinceMasuka