arendst/Tasmota #6348: Security - CVE-2019-12586, CVE-2019-12587, CVE-2019-12588?

petergeneric, 2019-09-03:
### BUG DESCRIPTION
I can't find any information on whether the current version is impacted by CVE-2019-12586, CVE-2019-12587 or CVE-2019-12588. The first two appear to be limited to networks with EAP, but the 3rd appears to be a general vulnerability to crash the esp8266

The issues are described in detail at https://github.com/Matheus-Garbelini/esp32_esp8266_attacks - excerpt:

* **[Zero PMK Installation (CVE-2019-12587)](https://matheus-garbelini.github.io/home/post/zero-pmk-installation/)** - Hijacking ESP32/ESP8266 clients connected to enterprise networks;
* **[ESP32/ESP8266 EAP client crash (CVE-2019-12586)](https://matheus-garbelini.github.io/home/post/esp32-esp8266-eap-crash/)** - Crashing ESP devices connected to enterprise networks;
* **[ESP8266 Beacon Frame Crash (CVE-2019-12588)](https://matheus-garbelini.github.io/home/post/esp8266-beacon-frame-crash/)** - Crashing ESP8266 Wi-Fi devices.

Follow the links on each vulnerability for more details.

This vulnerabilities were found in SDKs of ESP32 and ESP8266. Their version were ESP-IDF v4.0-dev-459-g7a31cb7 and NONOS-SDK v3.0-103-g7a31cb7 respectivelly at the time of the vulnerabilities discovery.

meingraham, 2019-09-03:
Tasmota uses the Arduino SDK for the ESP82xx. Tasmota can be compiled with different Cores (2.3.0, 2.4.x, 2.5.x, pre-2.6). If the vulnerability is in the particular Core, then Tasmota will likely be affected.

ascillato, 2019-09-04:
Hi,

Thanks a lot for sharing this information. 

By default, Tasmota don't use SDK 3.x and for the latests arduino cores, it uses SDK 2.2y that doesn't have enterprise network support.

So, the precompiled bins should be fine. The SDK 3.x has several bugs that also have wifi disconnections, that is why is not used by default in Tasmota.

ascillato, 2019-09-05:
As explained in the Arduino Repository, (https://github.com/esp8266/Arduino/issues/6436#issuecomment-528192690) using core 2.5.2 you can have the beacon crash attack. If you use latest STAGE core, this is already fixed.

In Tasmota you will need to compile under core pre2.6.0. At this moment this core is the default in platformio.ini

Matheus-Garbelini, 2019-09-05:
Hello, thanks for the discussion. I've updated the blog description with all the dates and commit patches that fixes the issues. Espressif has also made backports:
* NONOS SDK Stable release 3.0.1 (July 15, 2019)
* NONOS SDK Development master (July 3, 2019)
* NONOS SDK Backport 2.2.X (July 3, 2019)
* NONOS SDK Backport 3.0.0 (July 3, 2019)
* Arduino ESP866 Development Master (July 5, 2019)

ascillato2, 2019-09-05:
Thanks a lot for reporting. Very appreciated.

Closing this issue as it is solved by using latest arduino core. Platform.ini file has already this latest core by default:

https://github.com/arendst/Sonoff-Tasmota/blob/a1e9c2d2ac6ccc3ff7da4aae260501a1267f7175/platformio.ini#L182L183

@arendst 

May be we can deploy a new release version with core pre2.6.0?

ascillato2, 2019-09-05:
( keeping this issue opened until we have patched release binaries available for download )

andrethomas, 2019-09-06:
@ascillato @ascillato2 

Added http://thehackbox.org/tasmota/json/ to build dev bins using @Jason2866 's modified version of the staged Arduino core.

ascillato, 2019-09-06:
Cool, Thanks! :+1:

closed 2019-10-05 by ascillato2