arendst/Tasmota #6349: Counter / Pulse Timer as a Stopwatch with MQTT

drbasst, 2019-09-04:
Hello everybody,

I'm not sure where to post my question because it contains probably a bug as well as a general question.

First, let me explain what I'd like to to:

I would like to measure how often a sensor is activated (easy), measure the time between the activation and post the results via mqtt.
like: 
1 ; 1,5sec
2; 1,4sec
3: 2,1sec
...
The reason is, that I want to monitor the total count of cycles as well as the current speed.

To get two different counters on one sensor, I helped myself with a wire on the board, so that the sensor sends its signal to two pins. (I'm using a Wemos D1 mini)
One of those is defined as Counter, one is a pulse Timer.

So far so good.

Now the potential bug:
The pulse Timer works fine, but shows the right value only for a few seconds. Afterwards, it shows 
 4294.967295 s until the sensor is triggered again. Then, the new right(!) value is shown for a few seconds. Then the display on the webinterface and the Status via MQTT switches back to 4294.967295 s
Is this because of a mistake of mine or is it a bug? (Using Version 6.6.0)

My further question:
Like I said, I would like to send a Message via MQTT on each sensor activation containing the number and the time. The general "State" Posts contain the right information, but are only send in fixed time intervals.

I read a lot about rules etc. but came to the point, that it seems that I can not grab the counters /pulse timers values as a VAR or MEM even though they are in the systems "state" information.
Is this possible to get them anyhow?

Or is there another approach for this problem?

I think this functionality could be useful for others too, because it would be much easier to detect peaks in water or power consumption.

I hope that I did explain it well enough. If not, or if information is missing, please tell me.

Thanks a lot!

ascillato2, 2019-09-04:
Please, could you be so kind on completing the [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) in order to have more information so as to properly help you?

Remember to read the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md). Thanks.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

drbasst, 2019-09-04:
Sure:

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [?] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Wemos D1 mini (clone?)_____
- [x] Tasmota binary firmware version number used : latest pre-compiled from http://thehackbox.org/tasmota/
- [n.a.] Development IDE - Compiler / Upload tools used : ____ / ____
- [ x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

14:21:45 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
14:21:45 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:02:37","StartupUTC":"2019-09-04T13:19:08","Sleep":50,"CfgHolder":4617,"BootCount":6,"SaveCount":23,"SaveAddress":"3F5000"}}
14:21:45 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.9(138ed6d-sonoff)","BuildDateTime":"2019-09-04T13:09:15","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
14:21:45 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Thomas-Luftschloss2",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A64000000000000","00000000"]}}
14:21:45 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":3548,"Heap":20,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164068","FlashMode":2,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
14:21:45 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3797","IPAddress":"192.168.3.136","Gateway":"192.168.3.11","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.254","Mac":"80:7D:3A:78:EE:D5","Webserver":2,"WifiConfig":4}}
14:21:45 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_78EED5","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
14:21:45 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 04 13:21:45 2019","Local":"Wed Sep 04 14:21:45 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:11","Sunset":"19:27"}}
14:21:45 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-04T14:21:45","Epoch":1567603305,"COUNTER":{"C1":2,"C2":4294.967295},"SI7021":{"Temperature":27.9,"Humidity":44.7},"TempUnit":"C"}}
14:21:45 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-04T14:21:45","Epoch":1567603305,"Uptime":"0T00:02:37","UptimeSec":157,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"Thomas-Luftschloss2","BSSId":"F0:9F:C2:FA:20:F0","Channel":6,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

drbasst, 2019-09-11:
Hi! I just saw, that there is still the "awaiting feedback"-label attached. Is there any information still missing?

drbasst, 2019-10-07:
Can anyone help me please before this thread gets closed?

At least it would be great to know, if the behavior is considered to be a bug.

Thanks a lot!

meingraham, 2019-10-07:
@arendst ?

ascillato2, 2019-10-09:
Hi, 

Seems that you are having noise in the measurement. To have support on that, please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you.

Thanks.

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-10-09 by ascillato2

smart4h, 2019-10-22:
Hi @drbasst. I had exactly  the same problem. My workaround is downgrade Tasmota to 5.14 (last with millis). Then everything works as it should. It might be some noise problem in fact, but I didn't have time to dig in it. I m using  Tasmota to count pulses from energy-meter. Pulse timer is for calculate  instantaneous active power [Watts].

arendst, 2019-10-22:
> The pulse Timer works fine, but shows the right value only for a few seconds. Afterwards, it shows
4294.967295 s until the sensor is triggered again. Then, the new right(!) value is shown for a few seconds. Then the display on the webinterface and the Status via MQTT switches back to 4294.967295 s
Is this because of a mistake of mine or is it a bug? (Using Version 6.6.0.9)

It's as designed. The timer is reset to max value after every reading (gui or JSON) to prepare it for new measurement. It's also an indication that if the timer reports 4294.967295 it did not receive new pulses.

Considering the situation where a pulse could take much longer (4294 seconds = 71 minutes at most) than a gui update (which is about 2.3 seconds) I think this reset after a gui update is not correct. I'll change it to reset after 60 minutes if no pulse is received.