arendst/Tasmota #6351: reset boot count

GiladShoham, 2019-09-04:
I installed my sonoffs during the renovation of my house.
This means they were installed on the walls for a few months without wifi connection (the house was empty).
Now each of my sonoffs has thousands of boots, which make it much harder to see if it has new reboots now that I'm living in the house.
The only way to get around it is either have some table with the old boot counter and compare it every time - not a practical solution since I have about 40 devices.
Somehow restore completely all the devices which means I need to re-configure everything from scratch which is a lot of work. (actually I'm not sure what exactly I need to do - flash tasmota again? with wipe? take everything outside the wall? send a full reset command?)
What I really want is to send a command in the console to reset the counter.
This is the same as #684 which is an old closed issue. (I didn't just responded there since I believe no one will monitor it after 2 years).

here is a partial print screen from the tasmoadmin, just to see the numbers..

Thanks.

[image]

arendst, 2019-09-04:
I'll reconsider

ascillato2, 2019-09-04:
Hi, 

Your version is very old and as the required issue template was not completed, we can't know which core are you using and which settings are you using and which is the reboot reason. 

Those high values are not normal. 

First you should upgrade your devices to latest Tasmota (6.6.0.9) with core 2.5.2 (precompiled bin at http://thehackbox.org/tasmota/020502/sonoff.bin) or pre2.6.0 (by compiling Tasmota yourself). This latest core is very fast and reliable. 

Then, to reset those values you can send the command `reset 5` that will wipe all settings except wifi credentials. Be aware that this command will erase your rules, etc. But will help to start clean. 

---------------------------

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

GiladShoham, 2019-09-05:
Thanks for the response.
I'm using tasmota 6.4.1 with core 2.3.0.
I plan to upgrade all of them (Already upgraded few), I wonder whether to use core 2.5.2 or 2.3.0 (it looks like the default is 2.3.0, is there a reason why 2.5.2 is not the default?)
I know I can use reset 5, but it will require a lot of work to re-configure all 40 switches.
That's why I want something similar but only for the boot counter.

arendst, 2019-09-05:
I'll add a ``reset something`` soon to just clear the bootcount.

GiladShoham, 2019-09-05:
Thanks @arendst .
When is it expected to be available to install (not from source).

arendst, 2019-09-05:
Now! Just download the pre-compiled dev binaries from the link in the README file.

s-hadinger, 2019-09-05:
Theo, some users often need to do a complete flash erase via serial. Would it be possible (and woudl it work) to add a command like `Reset Full` that would erase not only Tasmota settings but also flash blocks used by Espressif SDK, especially pages FC to FF?

ascillato, 2019-09-05:
Now the command `reset 99` resets the bootcount to zero.

Thanks :+1:

arendst, 2019-09-05:
@s-hadinger it's already implemented. One of the 6 options does it. I'm out so cannot tell you which one.

s-hadinger, 2019-09-05:
Ha. Got it.

It looks like `Reset 2` does erase everything from end of code to end of physical flash.

GiladShoham, 2019-09-05:
> Now! Just download the pre-compiled dev binaries from the link in the README file.

@arendst Wow that was quick! Thanks a lot!
Is it safe to use the dev version for all my switches? are they stable enough?

arendst, 2019-09-05:
Yep

ascillato2, 2019-09-05:
Closing this issue as it has been added. Thanks.

closed 2019-09-05 by ascillato2

GiladShoham, 2019-09-09:
@arendst Worked like a charm, thank you very much!