arendst/Tasmota #6356: Hackbox 2.3.0 Binaries Not Compiling

meingraham, 2019-09-05:
http://thehackbox.org/tasmota/020300

The only binary that is up to date is minimal. All others have not been updated since August 28.

Mike

qingz2004, 2019-09-05:
There is a compilation error with core 2.3.0.

> Indexing .pioenvs/sonoff/lib3b6/libSD.a
> Indexing .pioenvs/sonoff/lib2f3/libesp-knx-ip-0.5.2.a
> Compiling .pioenvs/sonoff/FrameworkArduino/HardwareSerial.cpp.o
> In file included from lib/NeoPixelBus-2.5.0.09/src/internal/NeoEsp8266UartMethod.cpp:28:0:
> lib/NeoPixelBus-2.5.0.09/src/internal/NeoEsp8266UartMethod.h: In member function 'void NeoEsp8266AsyncUart<T_UARTFEATURE, T_UARTCONTEXT>::UpdateUart(bool)':
> lib/NeoPixelBus-2.5.0.09/src/internal/NeoEsp8266UartMethod.h:267:9: error: 'swap' is not a member of 'std'
>          std::swap(_pixelsSending, _pixels);
>          ^
> *** [.pioenvs/sonoff/libdaa/NeoPixelBus-2.5.0.09/internal/NeoEsp8266UartMethod.cpp.o] Error 1
> Indexing .pioenvs/sonoff/libFrameworkArduinoVariant.a
> ===================================== [FAILED] Took 98.13 seconds =====================================
>

joba-1, 2019-09-05:
swap is member of std, but depends on type to swap and c++ dialect. Maybe a build flag for c++11 or c++17 is missing?

arendst, 2019-09-05:
NeoPixelBus was updated recently and is failing on core 2.3.0

I'll try to patch it ASAP.

Thx for research so far.

closed 2019-09-05 by ascillato2