arendst/Tasmota #6357: How do add the EPT Font when building Tasmota?

DeviousPenguin, 2019-09-05:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING

Hi, I'm using a SSD1306 OLED screen with Tasmota Stable 6.6 and it works well with the default GFXFont.

I would quite like to add the EPDFont as detailed in the wiki:

https://github.com/arendst/Sonoff-Tasmota/wiki/Displays  

 >The EPD fonts use about 9k space, which can be selected at compile time using #ifdef directives.
> SSD1306 - 1.15k

I've searched the `my_user_config.h` file for how to add EPD fonts but there is nothing there.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : sonoff 4ch
- [x] Tasmota binary firmware version number used : 6.6 self-compiled
- [s] Development IDE - Compiler / Upload tools used : vscodium / platformio
- [x] Provide the output of command ``display`` :
```
DISPLAY OUTPUT HERE:

RESULT = {"Display":{"Model":2,"Mode":0,"Dimmer":1,"Size":1,"Font":3,"Rotate":0,"Refresh":2,"Cols":[16,8],"Rows":4}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

not relevent
```
**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2019-09-05:
the EPD fonts are enabled per default.
examples
GFX font size 1=> displaytext [f0s1]Hallo
EPD font 12 => displaytext [f1]Hallo
EPD font 24 => displaytext [f2]Hallo
EPD font 8 => displaytext [f3]Hallo

ascillato2, 2019-09-05:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-05 by ascillato2

meingraham, 2019-09-05:
`Tasmota Stable 6.6`

Is a later development branch required for the newer display driver features?

DeviousPenguin, 2019-09-05:
> the EPD fonts are enabled per default.

Nice, that would explain why I could not see any font related includes in my my_user_config.h file.

> examples
> GFX font size 1=> displaytext [f0s1]Hallo
> EPD font 12 => displaytext [f1]Hallo
> EPD font 24 => displaytext [f2]Hallo
> EPD font 8 => displaytext [f3]Hallo

Ok, next time I'm at the Sonoff I'll give that a try, when I tried using `DisplayFont 1` or `DisplayFont 2` the text did not change, it just stayed on the GFX font.