arendst/Tasmota #6358: Sonoff Mini "bricking" after Full Firmware Update

meingraham, 2019-09-05:
I cannot provide template information. I'm just reporting three instances in the past several days of seeing this from users.

A user uses Sonoff DIY to flash sonoff-basic and it completes successfully. However, when performing an update to sonoff.bin via web OTA, the device does not come back online. No AP is broadcasting. It cannot be reset with 40s button press. The only recovery is to flash via serial.

ascillato, 2019-09-05:
The problem seems that some incorrect data remains in the flash. 

Should be recommendable to do a `reset 5` to clean the flash BEFORE doing an OTA for the first time.

We should test that

ascillato2, 2019-09-06:
Could you test that?

effelle, 2019-09-07:
I asked to a friend of mine to try that. Will report as soon as I receive the feedback. In the meantime perhaps @digiblur can test it for us since he has a sonoff R3 too?

effelle, 2019-09-07:
Got a reply right now: update to sonoff-basic using `reset 5` DID NOT brick the device.

digiblur, 2019-09-07:
Had a Sonoff mini I did via DIY.  Did a reset 5 command and it came back without issue.  Then I flash http://thehackbox.org/tasmota/release/sonoff-basic.bin via the OTA URL method and it went through without issue.  Any other tests?

meingraham, 2019-09-07:
@digiblur 

1. Itead firmware >> Sonoff DIY sonoff-basic.bin
2. `Reset 5`
3. OTA update to sonoff.bin

Still works?

digiblur, 2019-09-07:
I won't be able to test that.  No itead firmware anymore.

meingraham, 2019-09-07:
Just buy another batch of your beloved Sonoff Minis :wink:

digiblur, 2019-09-08:
Ha!  Only reason I bought two was since they had a sale for $5 or so.

paybuzz, 2019-09-10:
Does anyone know if will there be a firmware update to resolve these Sonoff MINI DIY model so they will just work with standard firmware without the need to issue any further commands?

Jason2866, 2019-09-11:
You can compile your own sonoff firmware. Disable all the features you do not need.
The firmware size has to be <500k. There will be no other solution, because there is the 500k limit. The sonoff-basic version can be used for daily use. It is not a intermediate version.

meingraham, 2019-09-11:
Added this suggestion to the Sonoff DIY article.

Vendo232, 2019-09-15:
[image]

Team

I did Tasmota OTA on 6.6.0.12 sonoff-basic and I can wake up the device. i can flash via serial successfully but the unit will cycle in Serial monotor and won`t brotcast SSID, what can I do?

meingraham, 2019-09-15:
When you flashed serial did you erase flash first?

Vendo232, 2019-09-15:
yes I did. is there any other way to erase flash?

[image]

meingraham, 2019-09-15:
When it first boots, does it stay booted long enough so you can enter `Reset 5`?

Are you using a pre-compiled binary?

Does the device go into this reboot cycle upon a "fresh" flash or after attempting to configure the device?

Vendo232, 2019-09-15:
HALLELUJAH!!!!

I flashed **blank_1MB.bin** from ESP EASY FLASHER

then flashed tasmota and they are BACK!!!

meingraham, 2019-09-15:
We have suspected the NodeMCU PyFlasher "wipe" may not be as thorough as it sometimes needs to be. When using esptool.py we can make the erase explicit and as a matter of fact, we sometimes recommend doing it twice (each one performs a "write" to flash and also calibrates Wi-Fi. In other words, it's much more thorough an erase. Plus, when using esptool, we also recommend a power cycle between erase and upload. PyFlasher obviously doesn't do that either. Nevertheless, PyFlasher is still a great tool for a huge majority of users.

Jason2866, 2019-09-16:
You didnt follow wiki howto flash ;-)
You used DIO as flash mode. It is written on many places in Tasmota github and wiki that the one and only flash mode is **DOUT**

meingraham, 2019-09-17:
Wiki refined to provide detailed guidance. Nothing to be done beyond that I'm afraid.

closed 2019-09-17 by meingraham

fdsailer, 2019-12-17:
Did update as described in the wiki

1. Itead firmware >> Sonoff DIY sonoff-basic.bin
2. Reset 5 -> 3 times as someone suggested that it my be helping
3. OTA update to sonoff.bin

Now the device is not respondig. It seems that as soon as connected to the Wifi it will reboot again.
40s button and also the 4 times power on cycle for 3s did not help so far.

Any other sugestions then flashing via the serial way ?

Vendo232, 2019-12-17:
Flash 1MB empty bin file, Google it

fernajuf, 2019-12-17:
3 Sonoffs minis bricking. One of them I see on the network "sonoffDiy" but I can not access it. The other 2 don't even see them on the net.

Vendo232, 2019-12-17:
Solder serial and flash with empty 1MB bin than flash tasmota. Good luck, you are almost there

fernajuf, 2019-12-18:
In the first Sonoff Mini: I soldered the cables. I connect it and the LED flashes. I have run "esptool.py flash_id" and it doesn't connect.
In the second Sonoff Mini, I have also soldered the cables but the LED does not flash.

JackyX2k, 2019-12-30:
I have the same issue, 4 Mini's flashed, 3 out of 4 bricked, flashed all with tasmota-wifiman.bin, one worked just fine and manged to connect to my WiFi Network then updated with Sonoff basic worked flawlessly.  The other 3 flashed the same way, all of them as Showing on My DNS server but I can't connect to them, any suggestions?  Its really frustrating!

marcwagner, 2019-12-30:
same for me. bricked my second mini today. 

those pads are tiny. any tips to solder the serial connectors?

Vendo232, 2019-12-30:
Solder serial and flash with empty 1MB bin than flash tasmota. Good luck, you are almost there

http://www.pratikpanda.com/wp-content/uploads/2016/05/blank_1MB.zip

sbeneme, 2020-01-09:
Hi,
I'm in a similar situation. I've flashed 2 sonoff mini with tasmota-wifiman.bin(maybe wrong) now are death.
help

sredni, 2020-01-11:
@Vendo232 got right but may be more detailed. 
1. Solder 4 wires to mini board 
[image]
2. Connect it to your uart/ftdi but **do not connect** it to your computer yet (make sure that your adapter uses 3v3 not 5v).
3. Download/install esptool.py
4. Download empty flash image (http://www.pratikpanda.com/wp-content/uploads/2016/05/blank_1MB.zip)
5. Download latest tasmota (i suggest to use lite version - after all it may be easy updated via ota)
6. Hold button on your sonoff mini and connect uart/ftdi to computer
7. Burn empty image: 
```
esptool.py --port <path_to_port_which_sonoff_is_connected>  --chip esp8266 write_flash -fs 1MB -fm dout 0x0 <path_to_blank_image>/blank_1MB.bin
```
Output should looks like:
```
esptool.py v2.8
Serial port <path_to_port_which_sonoff_is_connected>
Connecting....
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: d8:f1:5b:82:82:dd
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
Compressed 1048576 bytes to 1039...
Wrote 1048576 bytes (1039 compressed) at 0x00000000 in 0.1 seconds (effective 88756.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
8. Disconnect uart/ftdi from computer
9. Hold button on your sonoff mini and connect uart/ftdi to computer
10. Burn tasmota:
```
esptool.py --port <path_to_port_which_sonoff_is_connected>  --chip esp8266 write_flash -fs 1MB -fm dout 0x0 <path_to_tasmota_image>/tasmota-lite.bin 
```
Output should looks like:
```
esptool.py v2.8
Serial port <path_to_port_which_sonoff_is_connected>
Connecting....
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: d8:f1:5b:82:82:dd
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 474576 bytes to 328553...
Wrote 474576 bytes (328553 compressed) at 0x00000000 in 29.0 seconds (effective 130.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
11. Sonoff led should start blinking and you can be happy now because you just bring back your sonoff to life :)

JackyX2k, 2020-01-29:
>    1. Disconnect uart/ftdi from computer
>    2. Hold button on your sonoff mini and connect uart/ftdi to computer
>    3. Burn tasmota:

Done, I had revived all my Sonoff's and working now ... Great every one and
thanks ...

mf01, 2020-01-31:
I'm at another project but have also this message.

```
Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.
```

Is this an issue or should the firmware run as expected?

sredni, 2020-02-04:
@mf01 Depends on what you are doing. If you trying to upload non empty image esptool try’s to check what is in the image and adjust some flashing settings. But when the image is empty this warning can be ignored.

georgelza, 2020-12-31:
... 4 units, 3 is dead,
#1 and #2, tried to solder.. well guess if you end damaging the components next to this damm small pins it won't work again. to even see the place to solder i use a magnifying glass, and still battle.

Tried the OTA, first battle to get it into DIY mode... eventually lucky and get it (these are new units without the jumper).
then assign wifi credentials, At this point I can do curl's and node-red command push'es. get responses. nw try and flash, originally used the pythin Simpllehttpserver, always end with broken pipe during download, so tried the npn one, well that works, file is transferred and then it complains about a POST api/devices/otaFlash and now it's dead/bricked, nothing works.
at times i get it to present a wifi, called ESP_**** but thats of no help,

working on a MAC.

i have 1 unit left, would love a step by step that will work from a MAC.

G

DSew20, 2021-03-22:
I had the same issue. Flashing minimal bricked the mini r2. But, you can recover by soldering to the header pins (which are tiny on the R2) and I got the first one running.

I had better luck with my 2nd mini r2. My OTA sequence was:

1. Connect the Mini R2 to Sonoff eWeLink mobile app. Use the "Compatible mode" for initial connection
2. Go to 10.10.7.1 and add the same wifi SSID as the PC you are going to use to flash via DIY tool
3. Download the Sonoff DIY tool. It should detect the Mini as its on the same network
4. Flash tasmota-lite via the DIY tool
5. Log-in to the SSID provided by the flashed tasmota on your mobile and go to 192.168.4.1 on the same device
6. Update the wifi settings so that the tasmota can connect to your network
7. Log-in to the tasmota once restarted. Flash to tasmota-minimal
8. Once restarted, flash to full tasmota

Its a bit of a pain doing all of these steps, but worth trying before heating up your soldering iron.

Hope this helps others.

tonygh77, 2021-12-23:
> @Vendo232 got right but may be more detailed.

> 4. Download empty flash image (http://www.pratikpanda.com/wp-content/uploads/2016/05/blank_1MB.zip)

Hello, how to download empty flash?

Vendo232, 2021-12-23:
This app has it included

https://github.com/letscontrolit/ESPEasy/releases

ramanraja, 2024-02-23:
I got many devices bricked after doing an OTA upgrade. The firmware was a custom build.
Problem: I had chosen  `board = esp8266_4M2M` in platformio.ini. But my device was actually a 1MB device.  
The OTA uploads were successful, but the device never came back alive again -I had to flash through serial port all over again.
Solution: choose the correct board type `esp8266_1M` and build again. After this change, my OTAs never failed.