arendst/Tasmota #6359: MOES TYWE1S based dimmer.. works but stops soon after flashing

vajonam, 2019-09-05:
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : _____
- [x] Tasmota binary firmware version number used : tried both  (pre-compiled or self-compiled ?)
- [x] Development IDE - Compiler / Upload tools used : platformio / esptool.py
- [x] Provide the output of command ``status 0`` :

```
STATUS 0 OUTPUT HERE:

21:35:58 CMD: STATUS 0 
21:35:58 SRC: WebConsole from 192.168.100.1
21:35:58 RSL: stat/sonoff/STATUS = {"Status":{"Module":54,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
21:35:58 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:00:20","StartupUTC":"2019-09-05T20:35:38","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":22,"SaveAddress":"F6000"}}
21:35:58 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.9(basic)","BuildDateTime":"2019-09-05T16:05:24","Boot":7,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
21:35:58 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["gaul-hw",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A64000000000000","00000000"]}}
21:35:58 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":458,"Free":544,"Heap":29,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashMode":3,"Features":["00000809","8F082383","043083A0","20021706","010001C0","00000001"],"Drivers":"1,2,3,4,6,9,10,16,20,21,22,24","Sensors":"3,4"}}
21:35:58 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-3506","IPAddress":"10.1.1.7","Gateway":"10.1.1.253","Subnetmask":"255.255.255.0","DNSServer":"10.1.1.253","Mac":"B4:E6:2D:66:AD:B2","Webserver":2,"WifiConfig":4}}
21:35:58 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_66ADB2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:35:58 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Sep 05 20:35:58 2019","Local":"Thu Sep 05 21:35:58 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:12","Sunset":"19:25"}}
21:35:58 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-05T21:35:58","Epoch":1567715758}}
21:35:58 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-05T21:35:58","Epoch":1567715758,"Uptime":"0T00:00:20","UptimeSec":20,"Vcc":3.435,"Heap":29,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":62,"MqttCount":0,"POWER":"ON","Dimmer":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"gaul-hw","BSSId":"8A:8A:20:87:77:6B","Channel":1,"RSSI":74,"LinkCount":1,"Downtime":"0T00:00:07"}}}
21:35:59 TYA: Set WiFi LED 3 (0)
21:35:59 TYA: Send "55aa000300010306"
21:36:00 TYA: Set WiFi LED 3 (0)
21:36:00 TYA: Send "55aa000300010306"
21:36:00 TYA: Send "55aa00000000ff"
21:36:01 TYA: Set WiFi LED 3 (0)
21:36:01 TYA: Send "55aa000300010306"
21:36:02 TYA: Set WiFi LED 3 (0)
21:36:02 TYA: Send "55aa000300010306"
21:36:02 HTP: Main Menu
21:36:03 TYA: Set WiFi LED 3 (0)
21:36:03 TYA: Send "55aa000300010306"
21:36:03 HTP: Configuration
21:36:04 TYA: Set WiFi LED 3 (0)
21:36:04 TYA: Send "55aa000300010306"
21:36:04 HTP: Configure Other
21:36:05 TYA: Set WiFi LED 3 (0)
21:36:05 TYA: Send "55aa000300010306"
21:36:05 WIF: Checking connection...
21:36:05 WIF: Connected
21:36:05 HTP: Configuration
21:36:06 TYA: Set WiFi LED 3 (0)
21:36:06 TYA: Send "55aa000300010306"
21:36:06 HTP: Configure Logging
21:36:07 TYA: Set WiFi LED 3 (0)
21:36:07 TYA: Send "55aa000300010306"
21:36:08 TYA: Set WiFi LED 3 (0)
21:36:08 TYA: Send "55aa000300010306"
21:36:08 HTP: Configure Logging
21:36:08 LOG: SerialLog 0, WebLog 4, SysLog 0, LogHost , LogPort 514, TelePeriod 300
21:36:08 HTP: Configuration
21:36:09 TYA: Set WiFi LED 3 (0)
21:36:09 TYA: Send "55aa000300010306"
21:36:09 HTP: Main Menu
21:36:10 TYA: Set WiFi LED 3 (0)
21:36:10 TYA: Send "55aa000300010306"
21:36:10 HTP: Console
```

This doesn't switch doesn't seem to power up on its own when connected to just 3.3v, so I cannot get the console output. 

Soon after flashing it works for about 1 minute or so. Works means I can click the togggle button hear the relay click on. Connect to MQTT and control brightness via WebUI and MQTT.  this adjusts the led array  on the switch to show the brightness. 

Then suddenly it stops works randomly only way to get to work again is to reflash. I have tried 6.6.0.9 compiled on my own. 

I am thinking this has something to with the MCU not being reset/heartbeat? 

I have saved the original 1M image before I started dumping. Any help is appropriated.

vajonam, 2019-09-05:
On the web console .
```
21:52:38 CMD: TuyaMCU 21,1
21:52:38 SRC: WebConsole from 192.168.100.1
21:52:38 RSL: stat/sonoff/RESULT = {"Command":"Unknown"}
```

What am I missing?

vajonam, 2019-09-05:
I am also using the sonoff-basic/fimrware.bin.

meingraham, 2019-09-05:
Please try 6.6.0.10 pre-compiled.

Please close this issue and address further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-05 by vajonam

ascillato2, 2019-09-05:
The sonoff-basic.bin is a basic version of Tasmota. Does not have tuya support. Please, use sonoff.bin that is the full Tasmota version. Thanks

Wolfmanxxx, 2019-10-29:
I'm running sonoff.bin ver 6.7.1
after setting "weblog 4" my log keeps spitting out
`TYA: Set WiFi LED 3 (0)
TYA: Send "55aa000300010306"
TYA: Set WiFi LED 3 (0)
TYA: Send "55aa000300010306"
TYA: Send "55aa00000000ff"
TYA: Set WiFi LED 3 (0)
TYA: Send "55aa000300010306"
TYA: Set WiFi LED 3 (0)
TYA: Send "55aa000300010306"`
tried the command `SerialSend5 55aa0001000000`
and received
`CMD: Group 0, Index 5, Command "SERIALSEND", Data "55aa0001000000"
RSL: stat/sonoff/RESULT = {"SerialSend":"Done"}
CFG: Saved to flash at F6, Count 43, Bytes 4096
TYA: Set WiFi LED 3 (0)
TYA: Send "55aa000300010306"
TYA: Set WiFi LED 3 (0)
TYA: Send "55aa000300010306"
TYA: Send "55aa00000000ff"`