arendst/Tasmota #6362: Cyclic lack of response when server is unavailable

mikemanha, 2019-09-06:
Hi there,

I’m using many Tasmota devices which has also some manual interfaces like buttons, switches and touch switches.
Great FW. All working perfectly if the server is not occupied or disconnected.
Sometimes the server(Raspberry) is out of order(damaged SD, updates, overheating, or any other reason). When it’s happened all Tasmota devices are periodically not responding to manual control. About 5 seconds you can use the switches to control lights directly, and then about 6-8 seconds it’s not responding, and then again 5 seconds that you can control manually… and so on.
If you want to turn the light on you have to push the switch only in the 5 seconds window when it can respond, otherwise nothing will happened.

It’s happened to all types with different modules(Tuya and sonof).
Also tested it with TPlink and linksys routers- same result.
Also tested core 2.3.0 and 2.5.2 same result. (with core 2.5.2 had additional disconnection issues if WIFI signal is lower than 50%, but it’s different issue).

I’m not sure. But it looks like the devices are trying to reconnect periodically and during this time can’t read other requests.

tasmota config for example:

> 
Program Version	6.6.0(release-basic)
Build Date & Time	2019-07-06T13:11:28
Core/SDK Version	2_3_0/1.5.3(aec24ac9)
Uptime	16T04:25:15
Flash write Count	1854 at 0xF7000
Boot Count	22
Restart Reason	Power on
Friendly Name 1	Sonoff
Friendly Name 2	Sonoff2
Friendly Name 3	Sonoff3
	 
AP1 SSId (RSSI)	arc12 (62%)
Hostname	l_bath_m-3499
IP Address	192.168.0.167
Gateway	192.168.0.1
Subnet Mask	255.255.255.0
DNS Server	192.168.0.1
	 
MQTT Host	192.168.0.163
MQTT Port	1883
MQTT User	mqnet
MQTT Client	DVES_466DAB
MQTT Topic	l_bath_m
MQTT Group Topic	sonoffs
MQTT Full Topic	cmnd/l_bath_m/
MQTT Fallback Topic	cmnd/DVES_466DAB_fb/
Emulation	None
mDNS Discovery	Disabled
ESP Chip Id	4615595
Flash Chip Id	0x144051
Flash Size	1024kB
Program Flash Size	1024kB
Program Size	422kB
Free Program Space	580kB
Free Memory	17kB

>

It seem like a critical issue if having trouble controlling manually. It supposed to be the most solid option, especially when the server is down.

Is there any way I can eliminate this lack of response using different config, or it’s more complex issue which requires FW update?

Jason2866, 2019-09-06:
Looks like mqtt lag. Please try to switch off mqtt for testing if the 5 sec delay is gone.
If yes try latest development version.

ascillato2, 2019-09-06:
Hi,

This lag is already fixed with latest Arduino core (pre2.6.0) (see #1992 latest comments)

Besides this, it is recommendable to move to this latest core due to #6348 

This last core is faster than the previous and solves this anoying lag you are experiencing.

Until a next release of Tasmota is made with this core, it is recommended that you compile latest Tasmota (6.6.0.10) by yourself. In the actual codebase (dev branch) this new core is set by default in platformio.ini (so, you don't need to edit or modify anything)

If you want to compile locally in your PC, it is recommended to use Visual Studio code + Platformio extension.

If you want to compile online, you can use gitpod.

All steps are explained in the wiki. Anyway, if you need help on this, please find us in the Tasmota Support Chat. Thanks.

Closing this issue as it is **duplicated**. Sorry.

--------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-06 by ascillato2

ascillato, 2019-09-06:
Until a new release is made, I attach here for you, latest Tasmota (6.6.0.10) with core pre2.6.0, so you can test it now. You can do an OTA upgrade if you like.

[sonoff.zip](https://github.com/arendst/Sonoff-Tasmota/files/3583908/sonoff.zip)

If you continue experience this or any other lag, please try to erase all flash with esptool.py as explained in the wiki and flash this file again with esptool.py

Thanks.

ascillato, 2019-09-06:
I attach also this sonoff-lite version that is the same without some sensors support in order to be a small firmware, so you can make a direct OTA without the need of using the 2 steps OTA with the minimal.

[sonoff-lite.zip](https://github.com/arendst/Sonoff-Tasmota/files/3583926/sonoff-lite.zip)

mikemanha, 2019-09-06:
thanks Jason, and Ascillato.
Indeed, if removing the mqtt configuration the lag is disappeared.
I have installed the latest dev, but the lag issue is still there.

I will try the files which you have attached.

Any estimation when will be the new release?

Thanks again.

ascillato, 2019-09-06:
>Any estimation when will be the new release?

Sorry, no idea.

mikemanha, 2019-09-06:
I have updated the FW with the attached file and no lag if the broker is down!

thanks

ascillato, 2019-09-06:
Great!!!! Thanks for reporting back! Very appreciated :+1:

ascillato, 2019-09-06:
If you or anyone needs the Tasmota version from today (6/sep/2019) with core STAGE but in another language, you can download it from https://github.com/ascillato/Sonoff-Tasmota_KNX/releases/tag/v6.6.0.10