arendst/Tasmota #6364: EPT Fonts not working on my SSD1306 OLED screen

DeviousPenguin, 2019-09-06:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

<Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.>

### ISSUE DESCRIPTION - TROUBLESHOOTING

Hi, I'm using a SSD1306 OLED screen with Tasmota Stable 6.6 it works but only with the default GFX font. I would like to use the larger EPT font as detailed here:

https://github.com/arendst/Sonoff-Tasmota/wiki/Displays  

I've tried both the `DisplayFont 2` command and also `DisplayText [f2]Hello world` commands but it makes no difference the font stays the same. It's always the GFX font. I can attach a pic if needed.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : sonoff 4ch
- [x] Tasmota binary firmware version number used : 6.6 self-compiled
- [x] Development IDE - Compiler / Upload tools used : vscodium / platformio
- [x] Provide the output of command ``DisplayFont 2`` :
```
DisplayFont OUTPUT HERE:

{"DisplayFont":2}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

not relevent
```
**(Please, remember to close the issue when the problem has been addressed)**

ascillato2, 2019-09-06:
Hi,

Can you provide also STATUS 0 output? Thanks

arendst, 2019-09-06:
For SSD1306 need dev version and then I'm still not sure it is supported.

gemu2015, 2019-09-06:
it works perfectly with the dev version!

ascillato, 2019-09-06:
@DeviousPenguin 

Please, try with latest Tasmota (6.6.0.10) Thanks

DeviousPenguin, 2019-09-06:
> Hi,
> 
> Can you provide also STATUS 0 output? Thanks

Hi sure no problem:
```js
STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LinkWIFI",""],"TelePeriod":60,"Resolution":"A98180C0","SetOption":["400A8009","280500000100060000000000000000000000","00000220"]}}
STATUS4 = {"StatusMEM":{"ProgramSize":484,"Free":516,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0F6000C6","240011A0","00000417","000000C0"]}}
STATUS5 = {"StatusNET":{"Hostname":"garage_lights_9C99AD-1181","IPAddress":"192.168.1.26","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.27","Mac":"60:01:94:9C:99:AD","Webserver":2,"WifiConfig":4}}
STATUS6 = {"StatusMQT":{"MqttHost":"diskstation.local","MqttPort":8883,"MqttClientMask":"garage_lights_%06X","MqttClient":"garage_lights_9C99AD","MqttUser":"new_tasmota","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 06 15:14:41 2019","Local":"Fri Sep 06 16:14:41 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"05:02","Sunset":"20:34"}}
STATUS10 = {"StatusSNS":{"Time":"2019-09-06T16:14:41","Switch1":"OFF","BME280":{"Temperature":16.81,"Humidity":68.68,"Pressure":1009.07},"PressureUnit":"hPa","TempUnit":"C"}}
STATUS11 = {"StatusSTS":{"Time":"2019-09-06T16:14:41","Uptime":"0T21:31:34","Vcc":3.486,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","Wifi":{"AP":1,"SSId":"LinkWIFI","BSSId":"7C:DA:90:79:18:FA","Channel":4,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```

> For SSD1306 need dev version and then I'm still not sure it is supported.

Ah OK I was unaware of this. I'll give the Dev version a try and see if it works then.

> Please, try with latest Tasmota (6.6.0.10) Thanks

Ok, I'll try the latest, I assume I should use dev branch here?

ascillato, 2019-09-06:
>Ok, I'll try the latest, I assume I should use dev branch here?

Yes, exactly

closed 2019-09-06 by ascillato2

DeviousPenguin, 2019-09-06:
Ok, now moved to the Dev branch, and the fonts are now working. Thanks :)