arendst/Tasmota #6368: Wiki: Add minimum firmware required to use a command

effelle, 2019-09-07:
**Have you looked for this feature in other issues and in the wiki?**
Yes, I have.

**Is your feature request related to a problem? Please describe.**
New features implemented on firmware requires, obviously, an update on device. Lately I have seen users complain on Discord about commands not working, like `LedLink` or `LedState`, due to an old firmware installed. 
Please consider not all user want to update the firmware for a series of reasons: they think is a complicated task or are simply scared by the idea if something goes wrong they need to remove the device from the plaster/wall/box to perform a serial flash again.

**Describe the solution you'd like**
When an user steps in Discord, Istart a triage session. A simple command already help me to have a first picture of the issue
`backlog module; template; status 2; gpio; rule1; rule2; rule3` 
but even if I am used to Tasmota, I can't always remember when a command/function were introduced. 
If I think a command could not be on a specific firmware version then I suggest an OTA before continuing.
On firmware files I can normally see what issue has introduced a new feature, what I don't know is on what firmware version was made the commit.
I would like to add a new column or a note on description field on [ command wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands), because all I have now is a basic information that at least version 6.6.0 is needed on `How to Use Commands` paragraph. 

**Describe alternatives you've considered**
Leave it that way and force an user to update to latest firmware to get assistance. Normally I suggest to migrate to dev branch.

**Additional context**
It will be a time consuming task and it's just an idea. If @theo or @ascillato have a way to get this kind of information easily then will be another useful tool for us. 
If not... I will continue to offer assistance as we do normally.

mprins, 2019-09-08:
From the sentence **Commands listed are available from version 6.6.0** it's pretty clear what the minimum viable version of software is for any and all of those commands. However since it is a wiki you are free to go ahead and update the table with an extra column and add those initial versions. It looks like a lot of work digging through all the [release notes](https://github.com/arendst/Sonoff-Tasmota/releases), the [wiki page history](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands/_history) and the [changelog](https://github.com/arendst/Sonoff-Tasmota/blob/development/sonoff/_changelog.ino)

jziolkowski, 2019-09-08:
https://github.com/arendst/Sonoff-Tasmota/blob/development/tools/decode-config.py#L415

this might be helpful. Maybe not for all commands, but it shows in which version what features were added

meingraham, 2019-09-08:
decode-config - nice version "reference". Will definitely use this!

effelle, 2019-09-08:
@mprins thanks for your comment but sorry, no. The majority of the commands are pre-6.6. That message on wiki was introduced to try to force the user to the latest master available, now 6.6.0 before was 6.5.0 (and so on). If an user want to use rules he don't need to upgrade to the latest master. 
What you probably don't know is @meingraham had already created a column with similar information in the past but we lost the data after a wiki upgrade.  
As I stated on OP there are users that are not comfortable with OTA or serial update. 
Considering how many user we assist on Discord every day the more triage tools  we have, the better.

@jziolkowski Thanks Jacek, I think is what I was looking for.

@meingraham Michael, perhaps that decoder is what we need to rebuild the base. Any chance to find the latest wiki edit with the old column to start to make a spreadsheet?

closed 2019-09-08 by effelle

arendst, 2019-09-08:
If you lost the version data from the commands web page you might be able to get it back when you click the link saying ``695 revisions``. It should allows to see all 695 revisions of the page.

effelle, 2019-09-08:
Theo, thanks to stepping in. 
As Michael told me the wiki layout was changed and is not easy to retrieve that column without creating some weird things. Since was his works we will works on team (as we normally do) to rebuild the information. Now I'm curious... What do you think about that?

arendst, 2019-09-11:
Any progress on this?
What are the plans to re-introduce the version information?

Considering the small amount of characters allowed on one line I prefer a text like **6.6.0.6** in the parameter column.

Like:

Command | Parameters
-- | --
Power0<a id="Power0"></a>|**6.6.0.9**<BR>Control the power state simultaneously for all relays on the device<BR> `0` / `off` = turn OFF <BR> `1` / `on` = turn ON <BR> `2` / `toggle` = if relay is `ON` switch to `OFF` and vice versa
PayUser<a id="PayUser"></a>|**until 1.1.0**<BR>Send bitcoins to user

Just an idea.

If you agree just let me know and I'll update the ones I can trace back.

meingraham, 2019-09-11:
@arendst 

I agree. In hindsight, adding the extra column inevitably ended up being hidden requiring horizontal scrolling or reduces the width of the Parameters column making that a jumbled mess.

Norbert's decode-config turns out to be a nice change log for many commands as he uses the firmware version to group supported commands.

Mike

arendst, 2019-09-11:
Great.

So are you planning a bulk update or can I just add my known version info to some single entries?

meingraham, 2019-09-11:
`Commands listed are available from version 6.6.0.` - And change this to explain the new "versioning".

Go ahead and get started. More can be rolled in as time permits/new information uncovered.

Care needs to be taken for any commands/setoptions introduced and later changed; particularly setoptions.

effelle, 2019-09-11:
Great @arendst!
I'm definetly in, I was just waiting your feedback to start with. 
I agree to put version in description, more readable. 
I will discuss with @meingraham how to proceed to track down the missing command versions.

effelle, 2019-09-11:
@arendst, @meingraham: 
I started to add the minimum version on `parameters` column as requested, gathering information from releases pages and issue pages or using decode-config.py. 
In many cases the default version added would be the `5.10.0`, because when I can't find information about older releases I use `decode-config` as fallback. The tool was deployed for the first time starting from that version.
For now I have edited just the [control](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#control) paragraph, and before I continue, I would like to know if I can use version `5.10.0` tag as minimum requirement for all the commands I can't find myself, and then lately (and eventually) update the page.

arendst, 2019-09-12:
I suggest to leave the version out if it is 5.10.0 This would mean it is available a long time and probably way before 5.10.0