arendst/Tasmota #6374: Sonoff Basic R2 crashes after setting PWM to GPIO

willhoh, 2019-09-08:
### BUG DESCRIPTION
When you set PWM to GPIO in Tasmota 6.6.0, Sonoff Basic R2, this crashes. 

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Sonoff Basic R2
- [x] Tasmota binary firmware version number used : 6.6.0 / (pre-compiled or self-compiled ?)
- [x] Development IDE - Compiler / Upload tools used : bin, TasmotaCompilier / NodeMCU PyFlasher
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
state 0  (befor set PWM to GPIO14)
12:00:46 CMD: state 0
12:00:46 RSL: stat/sonoff/RESULT = {"Time":"2019-09-08T12:00:46","Uptime":"0T00:00:12","Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"sunsys","BSSId":"E8:DF:70:ED:CA:FF","Channel":13,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}

state 0 (immediately after setting PWM1i to GPIO14)
12:07:10 CMD: state 0
12:07:10 RSL: stat/sonoff/RESULT = {"Time":"2019-09-08T12:07:10","Uptime":"0T00:04:51","Heap":15,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":92,"POWER1":"OFF","POWER2":"ON","Dimmer":40,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"sunsys","BSSId":"E8:DF:70:ED:CA:FF","Channel":13,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:07"}}

the following was recommended to me in the Discord to call up 
Backlog Module; GPIO; status 2
22:04:31 CMD: Backlog Module; GPIO; status 2
22:04:31 RSL: stat/sonoff/RESULT = {"Module":"1 (Sonoff Basic)"}
22:04:31 RSL: stat/sonoff/RESULT = {"GPIO1":"0 (None)","GPIO2":"0 (None)","GPIO3":"0 (None)","GPIO4":"0 (None)","GPIO14":"46 (PWM1i)"}
State
22:05:03 CMD: State
22:05:03 RSL: stat/sonoff/RESULT = {"Time":"2019-09-05T22:05:03","Epoch":1567717503,"Uptime":"0T00:00:43","UptimeSec":43,"Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER1":"ON","POWER2":"OFF","Dimmer":10,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"sunsys","BSSId":"E8:DF:70:ED:CA:FF","Channel":13,"RSSI":92,"LinkCount":1,"Downtime":"0T00:00:04"}}

```

### TO REPRODUCE
with version 6.6.0 set PWM to a GPIO on a Sonoff Basic R2.

### EXPECTED BEHAVIOUR
I want to control the brightness and thus the PWM of Tasmota via an external button. But Tasmota has to tell me that the button is pressed (for a longer time). I hoped for this through the rules. 

### ADDITIONAL CONTEXT
Wegen einem Tip im Discord habe ich versuch die Tasmota etwas anzupassen und bin dabei auf TasmotaCompiler gestoßen. Mit dem habe ich sämtliche Module ausgeschaltet und Stück für Stück wieder hinzugefügt. Dabei habe ich herrausgefunden, dass das Rules Feature das Problem darstellt. Ohne dieses (aber mit Scripts) geht es. 

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-09-08:
It is a known issue of sonoff basic r2 having a very very weak power supply.
You cant add anything that needs extra current or use parts in Tasmota that draws more current than
just switching a relais. PWM is software based and needs a lot of current because it is necessary  (and done) to disable dynamic sleep function which saves a lot of current.
Sorry no solution from software side. Only solution is replacing the weak sonoff basic!

willhoh, 2019-09-08:
I just don't understand why it worked great with the 6.4.1.13 then. I also wrote this in the issue.

Jason2866, 2019-09-08:
In later versions more features where added (need more CPU power -> needs more current)
So if 6.4.x works for you, stay at this build.

ascillato2, 2019-09-08:
Try to erase before flash with esptool.py as explained in the wiki and flash again with the precompiled bin called sonoff_basic.bin

closed 2019-09-08 by ascillato2