arendst/Tasmota #6375: SwitchMode: disable physical switch

schefa11, 2019-09-08:
It would be great if it is possible to disable the "switch operation" when the physical switch is actuated. Of course, the MQTT message still needs to be sent.
In this way you can use any switch for any actuator without changing the state of the actually connected actuator.

Jason2866, 2019-09-08:
It is already there. Please read https://github.com/arendst/Sonoff-Tasmota/wiki/Buttons-and-Switches

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-08 by ascillato2

BassT23, 2020-08-09:
I want to reopen this issue;

I want to disable the physically switch of my shellys, so my kids cant play around with it.
I try the poweronstate 4 option, but I also want to use it with off mode. So thats no option for me.

Best would be a switchmode option that disable any input.

So I could send command over Alexa -> iobroker -> device

ascillato, 2020-08-09:
This issue is very old. What you are asking is already implemented. Please, check setoptions on the Tasmota docs. Thanks.

ascillato, 2020-08-09:
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md)

Thanks.

---------------------------

### Support Information 
See [Docs](https://tasmota.github.io/docs/) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)

schefa11, 2020-08-10:
@BassT23:
It is month ago but after a few tries i managed it. But don't remember exactly but I think the most important step was to set the switchtopic. E.g. enter "switchTopic MyRoom" in the console of your tasmota device.
If setOption61 is disabled (default) then the relay should not operate when the button is pressed.
Wording is a little confusing: (https://github.com/arendst/Tasmota/pull/4562#issuecomment-446230001)

I hope I could help