arendst/Tasmota #6376: SSPI settings not stored on newer core versions

tremichl, 2019-09-08:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
Setting up SSPI ports for E-Paper Display Waveshare 2.9 cannot be stored if core version is higher than 2.3.0. It seems that Wemos mini D1 is rebooting sometimes after saving settings, and than all port settings are erased. But USE_DISPLAY_EPAPER_29 is disabled for core 2.3.0 

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : Wemos D1 mini
- [x ] Tasmota binary firmware version number used : 6.6.0.11/ pre-compiled 
- [ x] Development IDE - Compiler / Upload tools used : NodeMCU pyFlasher
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
15:17:05 CMD: status 0
15:17:05 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
15:17:05 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Exception","Uptime":"0T00:17:36","StartupUTC":"2019-09-08T13:59:29","Sleep":50,"CfgHolder":4617,"BootCount":5,"SaveCount":13,"SaveAddress":"F7000"}}
15:17:05 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.11(da35643-display)","BuildDateTime":"2019-09-08T13:05:11","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
15:17:05 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TNet01",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A64000000000000","00000000"]}}
15:17:05 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":525,"Free":476,"Heap":21,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FAAA39F","00369FA0","22B617C1","010008C0","00000001"],"Drivers":"1,2,4,5,6,8,9,10,13,16,18,19,22,24","Sensors":"1,2,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
15:17:05 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1339","IPAddress":"10.152.52.160","Gateway":"10.152.52.254","Subnetmask":"255.255.255.0","DNSServer":"10.152.52.254","Mac":"BC:DD:C2:24:85:3B","Webserver":2,"WifiConfig":4}}
15:17:05 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_24853B","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
15:17:05 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 08 14:17:05 2019","Local":"Sun Sep 08 15:17:05 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:16","Sunset":"19:19"}}
15:17:05 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-08T15:17:05"}}
15:17:05 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-08T15:17:05","Uptime":"0T00:17:36","UptimeSec":1056,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"TNet01","BSSId":"44:4E:6D:17:3A:FE","Channel":1,"RSSI":78,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE
Install Coreversion higher 2.3.0
Assigne GPIO Ports SSPI MOSI, SSPI SCLK, SSPI CS and save

### EXPECTED BEHAVIOUR
Device is several rebooting and settings are erased

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Using core 2.3.0 does work as expected, but display cannot be used with it

**(Please, remember to close the issue when the problem has been addressed)**

gemu2015, 2019-09-08:
could you please specify which pins you tried to use for the individual signals?

Jason2866, 2019-09-08:
@gemu2015  I tried too, same error behaviour. 
GPIO12 as MOSI, GPIO13 as SCLK and GPIO14 as CS

tremichl, 2019-09-08:
Same as Janson2866 and in addition GPIO13=MOSI GPIO14=CLK GPIO4=CS

gemu2015, 2019-09-08:
ok evaluating 

but disabling 
#define USE_DISPLAY_MODES1TO5

=>#undef USE_DISPLAY_MODES1TO5
seems to solve it for the time being 

please confirm

Jason2866, 2019-09-08:
Can confirm, does not reset BUT display isnt working!

gemu2015, 2019-09-08:
my display works well ???

displaymode1to5 is fixed with this change
this caused the crash

    //if (Settings.display_rotate) {
      if (!disp_screen_buffer_cols) { DisplayAllocScreenBuffer(); }
    //}

PR #6379 (2019-09-08): epaper display 29 fix
## Description:

**Related issue (if applicable):** fixes #6376 >

was related to diplaymode1to5

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2 and 2.5.2
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).
closed 2019-09-08 by arendst