arendst/Tasmota #6381: enable mqtt password via Backlog command

GiladShoham, 2019-09-08:
I have a bunch of sonoff t1 flashed with 6.6.0.11.
I tried to send a backlog command to set the MQTT user/pass as described in the wiki: [using-backlog](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#using-backlog)
The host/user has been set correctly, however, the password (which is disabled by default) do not become enabled. do I have to specify something else to make it enabled?

Thanks a lot.

Jason2866, 2019-09-08:
What command and payload do you have tried?

GiladShoham, 2019-09-08:
As described in the example
`Backlog mqtthost <yourhost>; mqttuser <user>; mqttpassword <password>;`
Also tried password only
`Backlog mqttpassword <password>;`

meingraham, 2019-09-08:
Try it without the `Backlog` - `mqttpassword <password>`

Likely won't matter, but just to confirm.

Jason2866, 2019-09-08:
The last `;` is not needed. Maybe it does harm

GiladShoham, 2019-09-08:
tried both.
not working.
It looks like it changed the pass but didn't make it enable.
here is a result in the console
```
21:44:01 CMD: mqttpassword mpass
21:44:01 RSL: access-trail/stat/RESULT = {"MqttPassword":"mpass"}
```
Then it restarts the device, but the password still disabled.

arendst, 2019-09-11:
How do you check if the password is enabled?

arendst, 2019-09-11:
I normally use mosquitto with anonymous access.

Just checked mosquitto security and it works as expected. Once a user and password is configured in mosquitto and the password file is activated only a device with a valid user/password is allowed to connect to mosquitto. If not allowed you will receive this:
```
14:10:14 MQT: Attempting connection...
14:10:14 MQT: Connect failed to domus1:1883, rc 5. Retry in 10 sec
```
So it's likely a mosquitto misconfiguration on your side.