arendst/Tasmota #6382: subscribe topic after restart 1 empty

timtimsson, 2019-09-08:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
_A clear and concise description of what the bug is._

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x ] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x ] Device used (i.e. Sonoff Basic) : Wemos d1
- [x ] Tasmota binary firmware version number used : 6.6.10 / (pre-compiled or self-compiled ?)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [0] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

21:37:08 CMD: Status 0
21:37:08 MQT: /ESP_DEV3/stat/STATUS = {"Status":{"Module":18,"FriendlyName":["ESP_DEV3","Sonoff2"],"Topic":"ESP_DEV3","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":0,"PowerRetain":1}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoff4096","OtaUrl":"http://10.1.1.200:80/api/arduino/sonoff4096.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:42","StartupUTC":"2019-09-08T19:17:26","Sleep":50,"CfgHolder":2309,"BootCount":15,"SaveCount":183,"SaveAddress":"3F5000"}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.10(sensors)","BuildDateTime":"2019-09-05T23:02:24","Boot":31,"Core":"2_5_2","SDK":"2.2.1(cfd48f3)"}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"10.1.1.200","LogPort":514,"SSId":["bronco","bronco"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000080E9","280500000100060000005A64000000000000","00000000"]}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS4 = {"StatusMEM":{"ProgramSize":708,"Free":2360,"Heap":15,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640E0","FlashMode":3,"Features":["00000809","EFEEE3B7","00B7B3A5","B7FFBFCD","01DA9BC0","00000007"],"Drivers":"1,2,3,4,5,6,8,9,10,12,13,14,16,17,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,25,26,28,34,37,39,40,42,43,51,52"}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS5 = {"StatusNET":{"Hostname":"ESP_DEV3-6529","IPAddress":"10.1.1.168","Gateway":"10.1.1.129","Subnetmask":"255.255.255.128","DNSServer":"10.1.1.129","Mac":"5C:CF:7F:00:19:81","Webserver":2,"WifiConfig":4}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.1.1.138","MqttPort":1883,"MqttClientMask":"ESP_DEV3-%06X","MqttClient":"ESP_DEV3-001981","MqttUser":"Raspi4","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 08 19:37:08 2019","Local":"Sun Sep 08 21:37:08 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+02:00","Sunrise":"06:33","Sunset":"19:41"}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS10 = {"StatusSNS":{"Time":"2019-09-08T21:37:08","Epoch":1567971428,"ANALOG":{"A0":0}}}
21:37:08 MQT: /ESP_DEV3/stat/STATUS11 = {"StatusSTS":{"Time":"2019-09-08T21:37:08","Epoch":1567971428,"Uptime":"0T00:19:42","UptimeSec":1182,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"bronco","BSSId":"44:4E:6D:6F:0D:29","Channel":1,"RSSI":82,"LinkCount":1,"Downtime":"0T00:00:08"}}}
21:37:44 MQT: /ESP_DEV3/tele/STATE = {"Time":"2019-09-08T21:37:44","Epoch":1567971464,"Uptime":"0T00:20:18","UptimeSec":1218,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"bronco","BSSId":"44:4E:6D:6F:0D:29","Channel":1,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:08"}}
21:37:44 MQT: /ESP_DEV3/tele/SENSOR = {"Time":"2019-09-08T21:37:44","Epoch":1567971464,"ANALOG":{"A0":0}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

```

### TO REPRODUCE

hi guys,
i use subscribe to send textmessages to my oled... it works fine, but ater restart the subscribe string is empty - why this ?
Thanks for your help!
Tim

laurentdong, 2019-09-09:
You need to re-subscribe after restart, so better to put it in rule, like:
`rule1 on system#boot do subscribe OLED, .......`

meingraham, 2019-09-09:
Updated wiki to highlight that `Subscribe` needs to be "reinstated" at boot.

closed 2019-09-10 by timtimsson