arendst/Tasmota #6387: HX711 loses calibration point after a reset

lalo-uy, 2019-09-09:
I have commented in bug #6312 but don't know how to reopen it.

I can't fully understand the way the  weight is calculated.

On load cells there are 2 parameters. the offset and the scale.
The offset if the value read from the sensor with no weight on the system.
That one is the hx_weight at the end of HX_CAL_RESET step ( xsns_34_hx711.ino line 295)  ,  when you are sure no weight is on the tray, but it is not saved any place.
After calibration the offset and the scale should be saved, so on boot the system can calculate the correct weight without intervention.

hx_offset is related to TARE function, and is independent ( bad name choice ) .

I don't see the usefulness of the  Settings.weight_reference once the calibration is done.
May be a good place to save the offset.

arendst, 2019-09-22:
Pls define "reset".

hx_weight is the current measured weight of the load. It varies with every load change. Saving it to flash with every load change would wear out flash pretty soon making the hardware unusable.

If you perform a restart the current weight is saved just before the restart and restored after the restart.

If you power off and power on the unit you will loose the knowledge of the current weight and will reinit to zero. To safely power cycle the unit while a load is still present perform command ``Sensor34 7`` first. This save the current measured load and restore after restart.

lalo-uy, 2019-09-23:
There is an Offset that is the weight of the scale components, with no
useful load.
I understand that that offset must be saved during calibration, no eeprom
wear.
If you know the offset and the scale coefficient at power on the actual
useful weight can be computed.
There is no "safely power cycle" in a blackout.
Hope to have a working version, and diff in 2 weeks , when I finish my
actual project.

andrethomas, 2019-09-23:
As indicated by @arendst the only way to store it at the moment would be to store it in the same flash locations used to store configuration data.

If I were you I would use one of these two options, probably in this order.

1. Get your home automation software (node-red in my case) to store the value and get the firmware to retrieve the value via mqtt after an mqtt connect event occurs.

2. If you're feeling "programmy" then think of things that have battery-backed SRAM such as the DS3232M RTC... not so much for the RTC part but for the fact that it has battery-backed SRAM which you can use to store data continuously and not lose anything during an intended or unintended power cycle.

ascillato2, 2019-09-23:
Closing this issue as it has been answered.

---------------------------

### Support Information ([Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md))
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-23 by ascillato2

tobox, 2019-12-09:
@andrethomas how could I realize your option 1?

>     1. Get your home automation software (node-red in my case) to store the value and get the firmware to retrieve the value via mqtt after an mqtt connect event occurs.

My Wemos D1 based scale restarted after a power cycle and now shows 0.000kg. I am logging the values on my home automation, so I know the last value. But from all the Sensor34 commands, how would I set the current load value? The scale is in a remote location, so I cannot remove the weight from the scale.

Regards

Seba200086, 2021-06-21:
@tobox, @andrethomas I have exactly the same question: how to realize option 1?

>     1. Get your home automation software (node-red in my case) to store the value and get the firmware to retrieve the value via mqtt after an mqtt connect event occurs.

I just made a weight sensor (4 load sensors, HX711, ESP8266 with Tasmota) put it under the oven feeder (I was going to measure the fuel level in the tank) and realised that after restarting the ESP8266 I get 0,000 kg....

Question is, how to "tell" the ESP8266 via MQTT that there is e.g. 70kg on the scale at the moment?

tobox, 2021-06-21:
I don't know if it's possible. But you can do what I did: just use the RAW value via mqtt and calibrate the values on your automation system. 

I agree that having the proper values directly in tasmota/mqtt would be nicer.

Seba200086, 2021-06-21:
Thanks for the tip.  Indeed, if only the RAW does not reset (to 0) after a reboot then this is the solution.

PR #7171 (2019-12-10): Report raw value from HX711 to overcome auto-tare functionality.
## Description:

**Related issue (if applicable):** fixes #6387 (partly)

Currently, when the device is switched off (intentionally or due to
power cut), Tasmota only supports starting with the last known weight
(Sensor34 7) or starting with 0 kg. In either case, this is sometimes
not the desired functionality. For off-site logging of weights, it is
more useful to report the raw, absolute value from the HX711 and to do
calibration/tare in upper layers.

Note: although it is called "Raw" value, the value is both averaged
and scale [... cut]