arendst/Tasmota #6388: Ability to prefer one SSID over another, documention improvements

dagbdagb, 2019-09-09:
**Have you looked for this feature in other issues and in the wiki?**
Yes.

**Is your feature request related to a problem? Please describe.**
When moving a set of devices with no connected button or switch from one ESSID to another, it would prove useful to have both ESSIDs configured at the same time and be able to indicate which one to prefer, which one to use as fallback, and what to happen if neither is present.
Having a fallback allows for the device to fail to a previously working config, which is very useful for a device without a physical user interface.

**Describe the solution you'd like**
Rather than preferring a stronger AP over a weaker, I would like to say:
- "prefer ESSID A now and in the future"
- "if ESSID A is unavailable, use ESSID B"
- "if neither A nor B is available, start a local accesspoint, but only if a local user/password has been configured". (For example)

**Describe alternatives you've considered**
I haven't found a way to prefer an ESSID, only to prefer an AP with a stronger signal. My APs broadcast multiple ESSIDs.

**Additional context**
.... and comments...
https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#Wi-Fi provides some commands to impact the wifi config. It uses terms like 'smartconfig' and 'wifi manager' without defining either, and exactly what 'disable Wi-Fi configuration' implies is not explained anywhere I could find. In short, this part of the (otherwise great) documentation is weak.

I understand that the available space on our devices in many cases limits what is possible and what is not. 

And finally, thank you very much for Tasmota. 

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-09-09:
@dagbdagb 

OK, added some [WiFiConfig](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wificonfig) links. Now quit your whining :wink:

Mike

P.S. Try as I may, I could not find any iOS SmartConfig apps. Another reason why WiFiManager is more broadly used.

andrethomas2, 2019-09-20:
Closing this issue as it has been answered.

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-20 by andrethomas2