arendst/Tasmota #6389: Set up a tip jar for Tasmota

dagbdagb, 2019-09-09:
**Have you looked for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Yes. I can't find a way to pay for a cold beverage for the developers of this software.

**Describe the solution you'd like**
A virtual tip jar of some kind? Paypal?

**Describe alternatives you've considered**
Expressing my gratitude for the software in bug reports and enhancement requests might come across as trying to buy/flatter my way to a solution I would like. I still do it...

**Additional context**
nah...

**(Please, remember to close the issue when the problem has been addressed)**

digiblur, 2019-09-09:
I have used the one on the main readme page of the project. But I do believe there is an option to put one up top in GitHub?

https://paypal.me/tasmota

dagbdagb, 2019-09-09:
@digiblur Did not find that one on first try. Actually, took a while to find it on the second try. Thank you!

closed 2019-09-09 by dagbdagb