arendst/Tasmota #6392: SUPPORT_IF_STATEMENT does not work with core 2.3.0

qingz2004, 2019-09-09:
Here is the error message during compilation:

> Linking .pioenvs/sonoff/firmware.elf
> .pioenvs/sonoff/src/sonoff.ino.cpp.o:(.text._Z19ExecuteCommandBlockPKci+0x10): undefined reference to `strpbrk'
> .pioenvs/sonoff/src/sonoff.ino.cpp.o: In function `ExecuteCommandBlock(char const*, int)':
> sonoff.ino.cpp:(.text._Z19ExecuteCommandBlockPKci+0xd2): undefined reference to `strpbrk'
> collect2: error: ld returned 1 exit status
> *** [.pioenvs/sonoff/firmware.elf] Error 1
>

Jason2866, 2019-09-09:
To keep core 2.3.0 up and running is getting harder and harder....
Can you try/test core pre 2.6 in your setup?
It performs very well in our tests and some new features are only possible with.

qingz2004, 2019-09-09:
It works with both core 2.4.2 and core 2.6, but the size is much bigger. I try to limit the file to 500M with all the sensors and options I need.

laurentdong, 2019-09-09:
In xdrv_10_rules.ino, please try to replace: 
`     char *pEndOfCommand = strpbrk(pos, "\x1e;");
`
with:
`      char *pEndOfCommand1e = strchr(pos, '\x1e');`
`      char *pEndOfCommand = strchr(pos, ';');`
`      if (NULL != pEndOfCommand1e &&`
`        (NULL == pEndOfCommand || pEndOfCommand > pEndOfCommand1e)) {`
`        pEndOfCommand = pEndOfCommand1e;`
`      }`

qingz2004, 2019-09-09:
@laurentdong That works. Thanks!

arendst, 2019-09-10:
I'll fix it now

closed 2019-09-10 by arendst