arendst/Tasmota #6393: Wemos D1 Mini xtensa-lx106-elf-gcc: error: libraries\esp8266\esp8266.a: No such file or directory

frablaser, 2019-09-10:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING
_A clear description of what the issue is and be as extensive as possible_
  
When I try to compile to the Wemos D1 mini I got the Error:
xtensa-lx106-elf-gcc: error: libraries\esp8266\esp8266.a: No such file or directory
I reinstalled the : C:\Users\******\AppData\Local\Arduino15 directory and downloaded the Esp8266 cards again, but it didn't work.

[image]

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [*] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [*] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [*] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [*] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [*] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [*] Device used (i.e. Sonoff Basic) : _____
- [*] Tasmota binary firmware version number used : ____ / (pre-compiled or self-compiled ?)
- [*] Development IDE - Compiler / Upload tools used : ____ / ____
- [*] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
xtensa-lx106-elf-gcc: error: libraries\esp8266\esp8266.a: No such file or directory

exit status 1
Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
xtensa-lx106-elf-gcc: error: libraries\esp8266\esp8266.a: No such file or directory

exit status 1
Erreur de compilation pour la carte LOLIN(WEMOS) D1 R2 & mini

```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-09-10:
Setup error of Arduino IDE. Do exactly as described in wiki. If you do different you get errors...
https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

Better IDE for a big project like Tasmota is PlatformIO. 
https://github.com/arendst/Sonoff-Tasmota/wiki/Flashing#flashing-and-compiling-from-source

frablaser, 2019-09-10:
Hi thanks to your feedback,

I mixed the "Copy all files from the Tasmota release Source code folder lib into your sketchbook\libraries folder" with the C:\Program Files (x86)\Arduino\libraries... 

That where bumped into...  Now it works!

By the way, the PlatformIO have a very bad user friendly IDE,very hard to understand, I'm a guy working with Microsoft stuff since 30 years, I use Vs Code beginning from the start, but that one isn't easy to understand. I didn't like it on my first try... I need to use it on simple project to get the learning curve.

Jason2866, 2019-09-10:
I use VSC with PlatformIO plugin. If you are used to VSC give it a try.

arendst, 2019-09-11:
Close as it has been solved.