arendst/Tasmota #6395: size of clock on display SSD1306

nromensky, 2019-09-10:
sorry if this has been discussed already .... how to make clock larger on  SSD1306 display, please see photo

[image]

Jason2866, 2019-09-11:
All possibilitys are explained in wiki
https://github.com/arendst/Sonoff-Tasmota/wiki/Displays

arendst, 2019-09-11:
Currently you cannot. 

Making it larger will make it unreadable as most characters will fall of the screen. There is no scaling in between.

[image]

arendst, 2019-09-11:
Now you can.

nromensky, 2019-09-12:
thx Theo! unfortunately I do not see significant difference in relation to this topic.

any chance you could make clock in format HH:MM which will fill whole screen, currently we have HH:MM:SS + date, which uses only half the screen

closed 2019-09-23 by nromensky