arendst/Tasmota #6398: Sonoff Touch 1 CH turns automatically on every few minutes

Moskito99, 2019-09-10:
Hi!

I flashed a Sonoff Touch 1 CH with FTDI and Tasmota Firmware 6.6.0.10.
Everything runs fine beside the automatically on and off.

Here is a log from one switch:

```
15:33:54 MQT: tele/sonoff_-_kuechenlicht/STATE = {"Time":"2019-09-10T15:33:54","Epoch":1568126034,"Uptime":"0T00:06:42","UptimeSec":402,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"LINUX.HOME","BSSId":"10:C3:7B:A2:C2:B0","Channel":13,"RSSI":86,"LinkCount":1,"Downtime":"0T00:00:04"}}
15:33:57 WIF: Checking connection...
15:33:57 WIF: Connected
15:34:17 WIF: Checking connection...
15:34:17 WIF: Connected
15:34:37 WIF: Checking connection...
15:34:37 WIF: Connected
15:34:47 APP: Button1 multi-press 1
15:34:47 SRC: Button
15:34:47 MQT: stat/sonoff_-_kuechenlicht/RESULT = {"POWER":"OFF"}
15:34:47 MQT: stat/sonoff_-_kuechenlicht/POWER = OFF
15:34:55 CFG: Saved to flash at FB, Count 213, Bytes 4096
```
I changed the option "SetOption1" to 1 and the automatically switch is gone, but the log messages about the Button 1 multi-press is still there...
I don't know if this is the right behaviour for this Sonoff switch....

```
15:46:08 WIF: Checking connection...
15:46:08 WIF: Connected
15:46:28 WIF: Checking connection...
15:46:28 WIF: Connected
15:46:47 APP: Button1 multi-press 1
15:46:48 WIF: Checking connection...
15:46:48 WIF: Connected
15:47:08 WIF: Checking connection...
15:47:08 WIF: Connected
```

meingraham, 2019-09-10:
Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/issues/new/choose) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

Please close this issue and address this to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Moskito99, 2019-09-10:
Move to Discord Chat.

closed 2019-09-10 by Moskito99

jardous, 2019-12-05:
I have the same issue. The button press is falsely triggered every minute...

18:14:20 APP: Button1 multi-press 1
18:14:24 WIF: Checking connection...
18:14:24 WIF: Connected
18:14:32 SRC: Button
18:14:32 RSL: stat/touch-switch-1/RESULT = {"POWER":"ON"}
18:14:32 RSL: stat/touch-switch-1/POWER = ON
18:14:32 CFG: Saved to flash at F7, Count 565, Bytes 4096
18:14:44 WIF: Checking connection...
18:14:44 WIF: Connected
18:15:04 WIF: Checking connection...
18:15:04 WIF: Connected
18:15:16 APP: Button1 multi-press 1
18:15:17 SRC: Button
18:15:17 RSL: stat/touch-switch-1/RESULT = {"POWER":"OFF"}
18:15:17 RSL: stat/touch-switch-1/POWER = OFF
18:15:17 CFG: Saved to flash at F6, Count 566, Bytes 4096
18:15:24 WIF: Checking connection...
18:15:24 WIF: Connected
18:15:29 APP: Button1 multi-press 1
18:15:41 SRC: Button
18:15:41 RSL: stat/touch-switch-1/RESULT = {"POWER":"ON"}
18:15:41 RSL: stat/touch-switch-1/POWER = ON
18:15:41 CFG: Saved to flash at F5, Count 567, Bytes 4096
18:15:44 WIF: Checking connection...
18:15:44 WIF: Connected
18:16:04 WIF: Checking connection...
18:16:04 WIF: Connected
18:16:06 APP: Button1 multi-press 1
18:16:17 SRC: Button
18:16:17 RSL: stat/touch-switch-1/RESULT = {"POWER":"OFF"}
18:16:17 RSL: stat/touch-switch-1/POWER = OFF
18:16:17 CFG: Saved to flash at F4, Count 568, Bytes 4096
18:16:24 WIF: Checking connection...
18:16:24 WIF: Connected
18:16:44 WIF: Checking connection...
18:16:44 WIF: Connected
18:17:04 WIF: Checking connection...
18:17:04 WIF: Connected

Unfortunately this seems to be HW problem...
(Sonoff T1 R2 EU)