arendst/Tasmota #6400: Working with Sonoff SP31 No Current Rating - Disabled Serial Logging Already

PlxntKid, 2019-09-11:
<GUIDE>

<This troubleshooting issue template is meant to help Tasmota users with difficult problems. It is aimed to be opened if using the wiki and the support chat could not solve the issue. The Github Issue tracker is NOT a general discussion forum>
  
<Please DO NOT OPEN AN ISSUE:>
  <If you have general questions or you need help on Tasmota usage, go to the Tasmota support chat>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a new device, please use the Tasmota Template Feature. See wiki for that>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### ISSUE DESCRIPTION - TROUBLESHOOTING

I have recently flashed a Sonoff SP31 and am having trouble getting it to report current.

I have flashed the most recent version of the sonoff.bin and I have also disabled serial logging and run the following commands in console:

SerialLog 0
Sleep 50
SetOption39

I have made sure to run a wattage test and set the powerset command as appropriate for the voltage I was receiving.

[image]

As you can tell by the uploaded image, I am not able to see any current across the device.

I have spoken with DigiBlurDIY and he has advised that I submit a support request and see if there is anything that can be done.

Thank you!

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : SP31
- [X] Tasmota binary firmware version number used : 2.5.2 / (pre-compiled or self-compiled ?)
- [X] Development IDE - Compiler / Upload tools used : NodeMCU PyFlasher / OTA Web Upload
- [X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

02:57:11 RSL: stat/sonoff/STATUS = {"Status":{"Module":41,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
02:57:11 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/020300/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:18:58","StartupUTC":"2019-09-11T01:38:13","Sleep":50,"CfgHolder":4617,"BootCount":8,"SaveCount":27,"SaveAddress":"FB000"}}
02:57:11 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.12(33baadf-sonoff)","BuildDateTime":"2019-09-10T22:04:05","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
02:57:11 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["dd-wrt-2.4",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100068000005AFF000000000000","00000000"]}}
02:57:11 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":509,"Free":492,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
02:57:11 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5177","IPAddress":"192.168.1.149","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"CC:50:E3:1B:B4:39","Webserver":2,"WifiConfig":4}}
02:57:11 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1BB439","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
02:57:11 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 11 01:57:11 2019","Local":"Wed Sep 11 02:57:11 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:20","Sunset":"19:13"}}
02:57:11 RSL: stat/sonoff/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
02:57:11 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-11T02:57:11","ENERGY":{"TotalStartTime":"2019-09-11T01:53:48","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":120,"Current":0.000}}}
02:57:11 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-11T02:57:11","Uptime":"0T00:18:58","UptimeSec":1138,"Heap":15,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":44,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"dd-wrt-2.4","BSSId":"CC:40:D0:8B:F5:CB","Channel":6,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:06"}}}
```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

02:54:42 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 50 8B 24 61 00 09 D8
02:54:43 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 52 24 89 61 00 09 D8
02:54:43 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 53 BC EF 61 00 09 D7
02:54:43 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 55 56 54 61 00 09 D8
02:54:43 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 56 F0 B8 61 00 09 D7
02:54:43 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 58 89 1E 61 00 09 D8
02:54:43 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 5A 23 82 61 00 09 D8
02:54:43 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 5B BB E8 61 00 09 D7
02:54:43 DMP: F2 5A 02 E6 F8 01 D8 01 00 3F 3E 02 55 F2 50 BF E0 5D 56 4C 61 00 09 D8

```
**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2019-09-11:
Investigating...

arendst, 2019-09-11:
First impression is it should work.

I just loaded up the same firmware as you (sonoff.bin, core 2.3.0, commit 33baadf) with no problem. See correct log below:
```
00:00:00 CFG: Loaded from flash at FB, Count 109
00:00:00 Project sonoff Pow3 Version 6.6.0.12(33baadf-sonoff)-2_3_0
00:00:04 WIF: Connecting to AP1 indebuurt2 in mode 11N as pow3...
00:00:08 WIF: Connected
00:00:08 HTP: Web server active on pow3 with IP address 192.168.2.194
00:00:09 MQT: Attempting connection...
00:00:09 MQT: Connected
00:00:09 MQT: tele/pow3/LWT = Online (retained)
00:00:09 MQT: cmnd/pow3/POWER = 
00:00:09 MQT: tele/pow3/INFO1 = {"Module":"Sonoff Pow R2","Version":"6.6.0.12(33baadf-sonoff)","FallbackTopic":"cmnd/DVES_668F99_fb/","GroupTopic":"sonoffs"}
00:00:09 MQT: tele/pow3/INFO2 = {"WebServerMode":"Admin","Hostname":"pow3","IPAddress":"192.168.2.194"}
00:00:09 MQT: tele/pow3/INFO3 = {"RestartReason":"Software/System restart"}
00:00:09 MQT: stat/pow3/RESULT = {"POWER":"ON"}
00:00:09 MQT: stat/pow3/POWER = ON
10:44:14 MQT: tele/pow3/STATE = {"Time":"2019-09-11T10:44:14","Uptime":"0T00:00:18","UptimeSec":18,"Heap":14,"SleepMode":"Dynamic","Sleep":0,"LoadAvg":151,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"indebuurt2","BSSId":"A0:AB:1B:7D:42:AC","Channel":5,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}
10:44:14 MQT: domoticz/in = {"idx":267,"nvalue":0,"svalue":"282.3;5870.4;0.0;0.0;2;0","Battery":100,"RSSI":10}
10:44:14 MQT: tele/pow3/SENSOR = {"Time":"2019-09-11T10:44:14","ENERGY":{"TotalStartTime":"2019-09-06T00:00:00","Total":6.153,"Yesterday":0.068,"Today":0.022,"Period":0.01,"Power":2.20,"ApparentPower":9.83,"ReactivePower":9.60,"Factor":0.22,"Voltage":227.9,"Current":0.043}}
10:44:24 CMD: weblog 4
10:44:24 MQT: stat/pow3/RESULT = {"WebLog":4}
10:44:24 DMP: 55 5A 02 F7 60 00 03 62 00 40 10 06 83 72 51 A6 58 27 21 B7 E1 7E C8 7E
10:44:24 DMP: 55 5A 02 F7 60 00 03 62 00 40 10 06 83 72 51 A6 58 27 21 B7 E1 7E C8 7E
10:44:24 DMP: 55 5A 02 F7 60 00 03 62 00 40 10 04 CD D1 51 A6 58 27 21 B7 E1 7E C8 25
10:44:24 DMP: 55 5A 02 F7 60 00 03 62 00 40 10 04 CD D1 51 A6 58 27 21 B7 E1 7E C8 25
```
From your (valid checksummed) DMP log it seems no load is detected by the S31 hardware. The first byte F2 tells that it took the hardware too long to measure any load. 

This means the load is too small to measure. Did you try a load of at least 10W?

digiblur, 2019-09-11:
Learned something new on that first byte there!  I was checking my logs on a couple S31's and they looked like Theo's with the 55 5A, etc.  One thing we did notice is @PlxntKid 's S31 has the newer case that Sonoff is showing as the new UL listed S31, but yet his doesn't say UL on the back.  That was the only difference hardware wise though when comparing it to other internal photos.

@PlxntKid does it say SP31 or S31 on the back?  I noticed you reference as SP in your post above.

arendst, 2019-09-11:
From the CSE7766 datasheet info regarding byte 1:

[image]

Where Abnormal is decoded as:

[image]

and byte 21:

[image]

With info in bits 4, 5 and 6:

[image]

PlxntKid, 2019-09-11:
I have attached a load of 143W and set the configuration using powerset.  I am still not seeing any difference on the current rating.

Is it possible that the current reading chip is somehow faulty?

PlxntKid, 2019-09-11:
I have already ordered two more s31's and I am hoping to flash those tomorrow to see if I have a repeat result.

digiblur, 2019-09-11:
Back to when I was hoping you had more than one. Good idea!

ascillato2, 2019-09-23:
Adding the label On Hold until the user has time to test.

closed 2019-09-23 by ascillato2

PlxntKid, 2019-09-24:
Yo guys!

I got a new outlet and flashed the same as before.

Everything is working with power monitor.

I am sure that it actually was a busted unit!

digiblur, 2019-09-24:
@meingraham nailed it. Smart guy indeed!

cesarmorisco, 2022-04-10:
Hello, I have a doubt about Tasmota because it disables the serial port in 10 minutes. And how to leave it active to communicate with a microcontroller. Thank you

barbudor, 2022-04-10:
@cesarmorisco by design Tasmota disable the serial console after 10 minutes
Normally you wouldn't use the Serial console to talk with a MCU but [Serial Bridge](https://tasmota.github.io/docs/Commands/#serial-bridge) or [TasmotaClient](https://tasmota.github.io/docs/TasmotaClient/)

Now, if you insist in keeping the serial console active you can have a rule such as `ON Time#minute|5 DO SerialLog 2 ENDON`

cesarmorisco, 2022-04-10:
Thanks
How do I connect a Microcontroller pic What do I have to do.
I thank

barbudor, 2022-04-10:
You are on the wrong topic.
You can open a new discussion but start reading the links I provided