arendst/Tasmota #6401: TuyaDevices

RobsonW70, 2019-09-11:
Just small question because always I see Sonoff
Is the firmware compatible for Tuya Devices? (like light switch curtains switch, sockets etc)
Do u have any list of supported devices with that firmware?
Please

Jason2866, 2019-09-11:
Reading wiki does help 😀 https://github.com/arendst/Sonoff-Tasmota/wiki
Devices see here: https://blakadder.github.io/templates/

RobsonW70, 2019-09-11:
Thank you for information
I am newbie ;) and sometimes I can have problem with checking what is with was.
I don's see many Tuya devices - because maybe something i s named different thatn should be 
I found only curtains - WF-CS01_EU
and I'm not sure for lights and dimmers:
Lights can be MMoes or Luminea
For dimmers it is harder ;(
Can u help me how can I identify what is suitable for my Tuya devices?

meingraham, 2019-09-11:
@RobsonW70 

Many, if not the majority, of the devices in the templates repository are Tuya devices. Scan the photos to see if you can find something that looks similar to yours and then check the product links to confirm whether it is. Even it it's a different brand, many times they are actually the same device.

If the device you have has a Wi-Fi module that is based on the ESP82xx, then you can flash Tasmota on it. HOWEVER, sometimes the device uses a secondary MCU to control some or all of the features of the device. Tasmota has support for many of these commands through the TuyaMCU command. However, the "library" of protocols used may not be complete. When a new one is encountered, Tasmota is updated to incorporate that new protocol.

So, for your Tuya devices, it is likely that it is supported now. But, if it is not, then it can be supported.

We ALWAYS recommend that you backup the original firmware from the device before erasing the flash and uploading Tasmota. This way, if you find that your device uses a new protocol, the original firmware can be restored so the protocol can be "sniffed" and then use that information to add it to Tasmota's TuyaMCU library.

At this point, please close this issue and address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-20 by andrethomas2