arendst/Tasmota #6403: GUIDE TO CONNECT AWS IOT

jevonch, 2019-09-11:
How to read AWS IoT With Sonoff Tasmota for newbie?

I am new in IoT and of course tasmota helped me alot. 
Im successfully played with Sonoff with basic connection. And it works well as the documentation are clear. 

I found difficulties in reading the guide for AWS IoT
https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT

Especially in this part :
Prepare your AWS IoT credentials. Type the following command: TLSKey. Where should I type this? Currently i'm using Arduino IDE. And flash to Sonoff POW R2. 

Step 1-7 are works well.and found the confuse in this part.

s-hadinger, 2019-09-11:
Happy to take feedback to make the doc less confusing :)

The TLSKey commands are to be typed in the Tasmota Web console, or through Serial connection.

andrethomas2, 2019-09-20:
Closing this issue as it has been answered.

### Support Information 
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [FAQ](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ) for common questions/answers and [links](https://github.com/arendst/Sonoff-Tasmota/wiki/FAQ#i-cannot-find-an-answer-here) if none of your question is in the list
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-20 by andrethomas2

sandrocs, 2019-10-22:
I am also having problems with the TLSkey command. the same is returning that the command is unknown, I believe I did all the steps correctly

s-hadinger, 2019-10-22:
If the command is unknown, it means that you didn't activate TLS and AWS IoT in your compiled Tasmota version.

sandrocs, 2019-10-24:
And how do I enable TLS in Tasmota?
I found that in the my_user_config.h file, you have the following part: #define USE_MQTT_TLS Would it be this ?

Thank you very much in advance

s-hadinger, 2019-10-24:
Yes. Please follow the instructions from this page: https://github.com/arendst/Sonoff-Tasmota/wiki/AWS-IoT

andremulin, 2020-03-18:
Hey guys. I had a equal experience. I tried so much, with different configurations. At finish I could run TLSKey. I tried run TLSKey in a Sonoff Mini run a Tasmota-Lite and never worked. The last try was run the default Tasmota version with the user config changes, and this setup worked.

vellen-vellen, 2020-04-09:
Hello @andremulin  can you please send me a copy the tasmota version you a using.
I am also facing the same issue

andremulin, 2020-04-10:
Hi @vellen-vellen! You can use your version. You only need follow the bellow sequence:
1 - Read the [Compile page](https://tasmota.github.io/docs/Compile-your-build/) and use GitPod to generate your .bin
2 - In GitPod, open tasmota/user_config_override_sample.h and insert theses values:
#ifndef USE_MQTT_TLS
#define USE_MQTT_TLS
#define USE_MQTT_TLS_CA_CERT // Optional but highly recommended
#endif
#ifndef USE_MQTT_AWS_IOT
#define USE_MQTT_AWS_IOT
#endif
#ifdef USE_DISCOVERY
#undef USE_DISCOVERY
#endif

3 - Open tasmota/my_user_config.h and uncomment (remove //) line with #define USE_CONFIG_OVERRIDE

4 - Compile your version based on standard tasmota

5 - Flash your device the normal way; either through serial or OTA. If you use OTA, first flash a sonoff-minimal firmware, then your target firmware.

6 - Flash your device with your firmware

If you need make any changes and compile again, you will need perform the steps 5 and 6 again before these changes can be applied.

s-hadinger, 2020-04-18:
I revamped the AWS IoT doc and provided 2 easy to use CloudFormation templates to simplify the process.
https://tasmota.github.io/docs/AWS-IoT/

vellen-vellen, 2020-04-30:
Thanks a lot

mjollnir14, 2021-01-24:
> 2 - In GitPod, open tasmota/user_config_override_sample.h and insert theses values:

Should be tasmota/user_config_override.h not _sample

ramanraja, 2021-08-09:
@s-hadinger,
Thanks for providing this very useful CloudFormation template.  I guess you are referring to this template:
[https://tasmota-eu-central-1.s3.eu-central-1.amazonaws.com/TasmotaAuth.yaml]
I think it has broken now, with an important piece missing:
  BackLogCommand:
```
    Description: The complete `backlog` command to configure the Tasmota device
    Value: !Sub "BackLog MqttHost ${GeneratePassword.Endpoint}; MqttPort 443; MqttUser tasmota?x-amz-customauthorizer-name=${MQTTAuthorizerName}; MqttPassword ${GeneratePassword.Password}"
```
With this, my Tasmota failed to connect to AWS cloud. 
I realized that TLS was not enabled in the default build. The YAML should include 
`SetOption103 1`
Probably this was there in an earlier version of the template, but is now (August 2021) missing.

s-hadinger, 2021-08-09:
Yes you are completely right. TLS is auto enabled with port 8883 but not with 443. 

I will update the CloudFormation template and maybe add auto enable of tls if port is 443

s-hadinger, 2021-08-16:
Thanks, the template in the S3 Bucket was outdated, it is now updated with the same version as in the documentation.