arendst/Tasmota #6405: IF Command Unknow

rcarreira88, 2019-09-11:
### BUG DESCRIPTION
When I set the Rule1 with the following code, the if throws a **command unknown**.
`Rule1 
ON system#boot DO var1 1 ENDON 
ON AM2301#Humidity<70 DO Backlog Power1 0; Var1 1 ENDON 
ON Button1#State==2 DO Backlog Power1 Toggle; Var1 0 ENDON
ON AM2301#Humidity>70 DO IF (Var1 = 1) Power1 1 ENDIF ENDON` 

I already define the flags USE_EXPRESSION and SUPPORT_IF_STATEMENT at compiled code.

STATUS 0 OUTPUT HERE:
```
23:08:15 MQT: stat/sonoff/STATUS = {"Status":{"Module":4,"FriendlyName":["Bathroom Thermostat"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[7,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}

```
CONSOLE OUTPUT HERE:
```
23:07:19 RUL: AM2301#HUMIDITY>70 performs "IF (Var1==1) Power1 1 ENDIF"
23:07:19 SRC: Rule
23:07:19 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
```

### TO REPRODUCE
Set the code above
I am using the Sonoff TH16

### EXPECTED BEHAVIOUR
When the humidity falls below 70% turn off power1
When the humidity rises above 70% turn on power1
When power1 is on and I press the button(push button,GPIO0), should turn off power1 until the humidity falls below 70% and then resumes the normal behavior

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-09-11:
@laurentdong 

Similar behavior as reported - https://github.com/arendst/Sonoff-Tasmota/pull/6386#issuecomment-529692957

laurentdong, 2019-09-12:
I noticed there is a bug for rule trigger and did a fix. 
PR #6405

closed 2019-09-12 by arendst

qingz2004, 2019-09-13:
Tried with "#define SUPPORT_MQTT_EVENT", still got "MQT: stat/cold/RESULT = {"Command":"Error"}" message.
Please fix it! It causes too much MQTT traffic.
Thanks!

meingraham, 2019-09-13:
Can you revert to not using IF feature until it is?

qingz2004, 2019-09-13:
Without IF feature, I have to use 2 rules each sensor since it takes more code (exceeding 512 characters). I run out of rule space.

meingraham, 2019-09-13:
@qingz2004 

Yes, I too desire to have this feature working properly. We'll get there. Like anything new, it'll take a little polish :wink:

Unfortunately AND/OR in  a trigger - no. You can simulate AND with Vars & Events... but it's not pretty.

Nevertheless, you should be able to put your rules in multiple rule sets and it still functions.

Mike

qingz2004, 2019-09-13:
We have only 3 rule sets. Without combining the conditions either though trigger or IF feature, it's not enough for me.

meingraham, 2019-09-13:
Your rules are greater than 511*3?

As long as a single rule (ON/DO/ENDON) is not longer than 511, you can put them into each of the three rule sets and they will all act in concert.

qingz2004, 2019-09-13:
Not to say 511*3.
Each set may have only just above 256, but I can not merge them into one, or it will be more than 512 characters. 
You got the idea.

meingraham, 2019-09-13:
Please post your rules. I'm curious to see if we can help you make use of more of the rules buffer space.

qingz2004, 2019-09-13:
I can use one set of rule with logical expression in IF state.

```
Rule1
on ADS1115#A0 do if (%value% > %var1% OR %value% < %var2%) TelePeriod 300; var1 %value%; var2 %value%; add1 200; sub2 200 endif endon
on ADS1115#A1 do if (%value% > %var3% OR %value% < %var4%) TelePeriod 300; var3 %value%; var4 %value%; add3 200; sub4 200 endif endon
on tele-ADS1115#A0 do backlog var1 %value%; var2 %value%; add1 200; sub2 200 endon
on tele-ADS1115#A1 do backlog var3 %value%; var4 %value%; add3 200; sub4 200 endon

```

Without IF support, I have to split it into two rule sets since it's over 512 characters.

```
Rule1
on ADS1115#A0>%var1% do backlog TelePeriod 300; var1 %value%; var2 %value%; add1 200; sub2 200 endon
on ADS1115#A0<%var2% do backlog TelePeriod 300; var1 %value%; var2 %value%; add1 200; sub2 200 endon
on tele-ADS1115#A0 do backlog var1 %value%; var2 %value%; add1 200; sub2 200 endon
Rule2
on ADS1115#A1>%var3% do backlog TelePeriod 300; var3 %value%; var4 %value%; add3 200; sub4 200 endon
on ADS1115#A1<%var4% do backlog TelePeriod 300; var3 %value%; var4 %value%; add3 200; sub4 200 endon
on tele-ADS1115#A1 do backlog var3 %value%; var4 %value%; add3 200; sub4 200 endon

```

And I plan to extend this to use A2 and A3 inputs. It will run out of rule set numbers and variable numbers. I can workaround to use memory instead of variable, but no workaround for the rule space.

qingz2004, 2019-09-13:
You are right. I never think of split each rule set and combine then into new rule sets to squeeze them into the limited space.
Looks like I can not accomplish this without a new command. A similar command like PowerDelta or a similar trigger like Analog#A0div10.
Thanks!

meingraham, 2019-09-13:
In these rules, The TelePeriod you execute will trigger all the `tele-ADS1115#An` rules. Why set the `Var` in both when doing one will trigger the other. For example:
```
Rule1
on ADS1115#A0>%var1% do TelePeriod 300 endon
on ADS1115#A0<%var2% do TelePeriod 300 endon
on ADS1115#A1>%var3% do TelePeriod 300 endon
on ADS1115#A1<%var4% do TelePeriod 300 endon
on tele-ADS1115#A0 do backlog var1 %value%; var2 %value%; add1 200; sub2 200 endon
on tele-ADS1115#A1 do backlog var3 %value%; var4 %value%; add3 200; sub4 200 endon
```

Also, try these (I removed the spaces surrounding `<` & `>`) to see if the IF error persists. Try to just turn on Rule1 for now.

Rule1
on tele-ADS1115#A0 do backlog var1 %value%; var2 %value%; add1 200; sub2 200 endon
on ADS1115#A0 do if (%value%>%var1% OR %value%<%var2%) TelePeriod 300; var1 %value%; var2 %value%; add1 200; sub2 200 endif endon

Rule 2
on tele-ADS1115#A1 do backlog var3 %value%; var4 %value%; add3 200; sub4 200 endon
on ADS1115#A1 do if (%value%>%var3% OR %value%<%var4%) TelePeriod 300; var3 %value%; var4 %value%; add3 200; sub4 200 endif endon

qingz2004, 2019-09-13:
var1 and var2 use the value A0, and var3 and var4 use the value A1, thus two triggers for different var sets.

I even tried the sample rule which has no space, still got the same error messages.
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules:-IF-ELSE-ELSEIF-and-AND-OR-Support

meingraham, 2019-09-13:
OK, I just wanted to give that "no space" a try to cover the bases.

If either of these rules are triggered
```
on ADS1115#A0>%var1% do backlog TelePeriod 300; var1 %value%; var2 %value%; add1 200; sub2 200 endon
on ADS1115#A0<%var2% do backlog TelePeriod 300; var1 %value%; var2 %value%; add1 200; sub2 200 endon
```

They will trigger this
```
on tele-ADS1115#A0 do backlog var1 %value%; var2 %value%; add1 200; sub2 200 endon
```
because of the `TelePeriod 300`

Therefore this has the same effect
```
on ADS1115#A0>%var1% do TelePeriod 300 endon
on ADS1115#A0<%var2% do TelePeriod 300 endon
on tele-ADS1115#A0 do backlog var1 %value%; var2 %value%; add1 200; sub2 200 endon
```

The same for the `A1` rules that use `Var3` and `Var4`.

Mike

P.S. `on system#boot do var1 0 endon` is never needed because all `Var` are zero on boot.

qingz2004, 2019-09-13:
The rule on tele-ADS1115#A0 is not necessary, I put it there just for redundancy in case a glitch causes var1 and var2 are set to some data which will cause the compassion trigger never happen, I can reset them every teleperiod.

meingraham, 2019-09-13:
Understood. I was just trying to "make room" in the rule buffers. Getting rid of that rule accomplishes that ;-) And you don't need the `TelePeriod 300` either. Plenty of space :wink:

Now you just don't have enough `Var` :frowning_face:

qingz2004, 2019-09-13:
Right. I wish I could define maximum number of vars in compiling time with a #define.
Thanks!

meingraham, 2019-09-14:
Well... as it turns out, it looks like you can have a maximum of 16. Add `MAX_RULE_VARS` in your `user_config_override.h`. Just don't get carried away because it will consume RAM which is at a premium. Maybe set it to 8 to meet your current needs.

qingz2004, 2019-09-14:
You are right.
MAX_RULE_MEMS can not be changed, but MAX_RULE_VARS can be set up to 16.
https://github.com/arendst/Sonoff-Tasmota/issues/4933

Problem solved!
Thanks!

PR #6406 (2019-09-12): Rule did not been triggered if there's no comparation operator
Rule did not been triggered if there's no comparation operator provided as trigger. The issue related with the new introduced IF support.
For example:
Rule1 ON EVENT#POWERON DO ... ENDON
The rule_name should be assigned as "POWERON" by default.

## Description:

**Related issue (if applicable):** fixes #6405 

## Checklist:
  -  [ X] The pull request is done against the latest dev branch
  - [ X] Only relevant files were touched
  - [ X] Only one feature/fix was added per PR.
  - [ X] The code change is tested and works on core 2.3 [... cut]