arendst/Tasmota #6407: Method to retrieve ALL possible GPIO components via MQTT, including assigned

jziolkowski, 2019-09-12:
The command "gpios" returns all components but without those assigned in module config or template. To be able to edit templates externally, this needs extra work to compile a complete list. 

Such list of all available GPIOs (including these assigned) is available in the comboboxes in template edit - could it be available via MQTT? Maybe just adding a parameter to existing command, or a whole new command? That would greatly help external management tools (like TDM wink wink nudge nudge)

arendst, 2019-09-12:
Not all GPIO are available in all compilations. Depending on defines enabled many GPIO's are not present so your list will almost never be complete.

The ones assigned in module config can be shown with command ``gpio``.

jziolkowski, 2019-09-12:
Yes, I know. That's why I'm polling each device before configuring gpios. I don't want a list of all gpios available in whole Tasmota - but only in that certain device.