arendst/Tasmota #6414: A mysterious issue with light state

GiladShoham, 2019-09-13:
Hi,

Thanks for the great product and work on tasmota.
You are doing a great job (both technical and support!)

## tl;dr
There is a video I created to demonstrate everything.
It's available here:
https://youtu.be/Z9_0bTQEAZo

## environment background
I have a really wierd bug.
I have a lot of switches - 36 switches with 57 gangs.
All are the same (sonoff t1).
All use the same tasmota version: 6.6.0.11
HA is: 0.98.4.
Using mosquitto mqtt as broker.
All configured exactly the same in tasmota (using background command)
All configured exactly the same in HA (using custom script I'll make open source soon, which take Airtable base and create the HA yamls).
So there is no differences at all.

## issue:
Some of the switches are turned on when clicking on HA then the toggle move to off after 2 seconds. (the light itself turned on on the first click and stayed on after the toggle move to off).
If I then click it again, it won't turn off the light, the toggle will move to on, then back to off.
If I click it twice quickly (means I click the turn off before it moved automatically, it will turn off the light.
When clicking on the group main toggle, it will turn on everything, but the problematic ones, will the toggle stayed off (the light did turned on).
On the tasmota ui everything fine (the state is correct).
(see everything in the video link above)

Looks like there is an issue with the reporting back to HA via the MQTT.

I don't know what else to check, but will happy to provide any additional details.

Thanks,
Gilad.

joba-1, 2019-09-13:
Looks like you might be right, but we can‘t tell.

What is the config of the devices (status 0)?

What is in the logs (expected vs unexpected case)?

Please follow the guidelines for opening issues to make life easier for those trying to help...

GiladShoham, 2019-09-13:
HA config:
not working one:
```yaml
  - 
    platform: "mqtt"
    name: "living_room_ceiling_big"
    state_topic: "house/stat/entry/living-room-ceiling-big/POWER1"
    command_topic: "house/cmnd/entry/living-room-ceiling-big/POWER1"
    availability_topic: "house/tele/entry/living-room-ceiling-big/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    payload_on: "ON"
    payload_off: "OFF"
    retain: false
```
working:
```yaml
  - 
    platform: "mqtt"
    name: "living_room_ceiling_small"
    state_topic: "house/stat/entry/living-room-ceiling-small/POWER2"
    command_topic: "house/cmnd/entry/living-room-ceiling-small/POWER2"
    availability_topic: "house/tele/entry/living-room-ceiling-small/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    payload_on: "ON"
    payload_off: "OFF"
    retain: false
  - 
    platform: "mqtt"
    name: "living_room_ceiling_x_small"
    state_topic: "house/stat/entry/living-room-ceiling-small/POWER1"
    command_topic: "house/cmnd/entry/living-room-ceiling-small/POWER1"
    availability_topic: "house/tele/entry/living-room-ceiling-small/LWT"
    payload_available: "Online"
    payload_not_available: "Offline"
    payload_on: "ON"
    payload_off: "OFF"
    retain: false
```

status 0:
not working:
```js
{
    "Module": 28,
    "FriendlyName": [
      "living-room-ceiling-big"
    ],
    "Topic": "living-room-ceiling-big",
    "ButtonTopic": "0",
    "Power": 0,
    "PowerOnState": 3,
    "LedState": 0,
    "LedMask": "FFFF",
    "SaveData": 1,
    "SaveState": 1,
    "SwitchTopic": "0",
    "SwitchMode": [
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0
    ],
    "ButtonRetain": 0,
    "SwitchRetain": 0,
    "SensorRetain": 0,
    "PowerRetain": 1
  }
```

working one:
```js
{
    "Module": 29,
    "FriendlyName": [
      "living-room-ceiling-x-small",
      "living-room-ceiling-small"
    ],
    "Topic": "living-room-ceiling-small",
    "ButtonTopic": "0",
    "Power": 0,
    "PowerOnState": 3,
    "LedState": 0,
    "LedMask": "FFFF",
    "SaveData": 1,
    "SaveState": 1,
    "SwitchTopic": "0",
    "SwitchMode": [
      0,
      0,
      0,
      0,
      0,
      0,
      0,
      0
    ],
    "ButtonRetain": 0,
    "SwitchRetain": 0,
    "SensorRetain": 0,
    "PowerRetain": 1
  }
```

Clicking on on HA:
not working one:
```
10:38:54 MQT: house/stat/entry/living-room-ceiling-big/RESULT = {"POWER":"ON"}
10:38:54 MQT: house/stat/entry/living-room-ceiling-big/POWER = ON (retained)
```
Working one:
```
10:39:34 MQT: house/stat/entry/living-room-ceiling-small/RESULT = {"POWER2":"ON"}
10:39:34 MQT: house/stat/entry/living-room-ceiling-small/POWER2 = ON (retained)
```

What else can help?

GiladShoham, 2019-09-13:
Of found the issue!
It was a stupid one.
When configuring a 1 gang switch, it should be with POWER not POWER1:
```
    state_topic: "house/stat/entry/living-room-ceiling-big/POWER"
    command_topic: "house/cmnd/entry/living-room-ceiling-big/POWER"
```
insted of:
```
    state_topic: "house/stat/entry/living-room-ceiling-big/POWER1"
    command_topic: "house/cmnd/entry/living-room-ceiling-big/POWER1"
```

closed 2019-09-13 by GiladShoham

joba-1, 2019-09-13:
Great! If you ask me, intuitively power should mean all switches and power1 the first.
Probably a historic/backward compatibility thing...

meingraham, 2019-09-13:
`Power` and `Power1` are synonymous. This is _**usually**_ the case for any `Command<x>`. If you do not specify `<x>`, it assumes `1`

GiladShoham, 2019-09-15:
> Power and Power1 are synonymous. This is usually the case for any Command<x>. If you do not specify <x>, it assumes 1

Apparently they act differently..
I'm not sure if it's by purpose, it might be just a bug.
anyway I tested it for about 20 switches. all the single gang switches work only with POWER but not with POWER1

meingraham, 2019-09-15:
If you go to the device's web UI Enter its IP address in a browser), and go to the Console, what behavior do you see if you enter `Power1` and `Power`?

GiladShoham, 2019-09-15:
I can check it when I'll be at home.
but in general, it works for the cmnd but not for stat

meingraham, 2019-09-15:
@GiladShoham 

Actually, check your [`SetOption26`](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#setoption26) setting.

GiladShoham, 2019-09-16:
> Actually, check your SetOption26 setting.

That makes a total sense.
I never changed it, so I guess I have the default.
I can verify if it's important for you.

meingraham, 2019-09-16:
No need to check. You are seeing the default behavior. Honestly, I thought the default was the other way around. But that answers what you were seeing. If you want `Power1` in your setup, change the setting of that options.

Cheers!

Mike