arendst/Tasmota #6417: Regarding the recent SDM120 development

gajotnt, 2019-09-13:
>  * Add driver USE_SDM120_2 with Domoticz P1 Smart Meter functionality as future replacement for USE_SDM120 - Pls test and report

Have been using this for a couple of days, gives me really wront values on the daily usage.
Please dont replace, if you can keep has is, using the SDM similar to POW.

The return values are within what is expected (compare with WASTE) but the usage is way, way off.
[image]

Maybe i configured something wrong, did not change any of the Tariff comands. Because the Usage 1 in domoticz is always @ 0watts and only Usage 2 is registring values.

[image]
[image]

arendst, 2019-09-14:
Partly answering:
- Domoticz Usage1 is energy usage during tariff1 period.
- Domoticz Usage2 is energy usage during tariff2 period.

As you did not change the Tasmota tariff settings from the default settings there are no tariffs in use and only Usage2 will report values being the Peak period as defined by Domoticz:

[image]

In the above case tariff is defined as ``{"Tariff":{"Off-Peak":[23,22],"Standard":[7,6],"Weekend":"ON"}}`` so that should explain your Usage question.

gajotnt, 2019-09-14:
Ok thanks for the input, will configure the tariffs time period and report back later.

ascillato2, 2019-09-23:
Adding the label On Hold until the user has time to test.

closed 2019-09-23 by ascillato2