arendst/Tasmota #6420: Wifi connection lost and false WiFi.status() result

untitledlt, 2019-09-14:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

### BUG DESCRIPTION
Device loses wifi connection (can not ping it from pc) but internal tasmota check does not notice it (`WIF: Checking connection... Connected`). 

Other devices (mac, phone, sonoff) on the same wifi works ok at that time. MQTT broker also works.

Sometimes this happens after few minutes, sometimes after few hours. Reboot helps instantly.

I also have Sonoff TH with tasmota fw. It also sometimes loses wifi connection but i don't have serial hooked to it so no log from it.

Maybe wifi connection status check could be improved with ping to GW as it seams that `WiFi.status()` is not correctly reporting actual status.

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : BlitzWolf BW-SHP5
- [ ] Tasmota binary firmware version number used : 6.6.0 / (pre-compiled)
- [ ] Development IDE - Compiler / Upload tools used : ____ / ____
- [ ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:

10:44:35 CMD: status 0
10:44:35 SRC: WebConsole from 10.0.0.2
10:44:35 RSL: Received Topic /status, Data Size 1, Data 0
10:44:35 RSL: Group 0, Index 1, Command STATUS, Data 0
10:44:35 MQT: stat/shp5/STATUS = {"Status":{"Module":0,"FriendlyName":["SHP5","Sonoff2"],"Topic":"shp5","ButtonTopic":"0","Power":2,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
10:44:36 MQT: stat/shp5/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:01:04","StartupUTC":"2019-09-14T09:43:31","Sleep":50,"CfgHolder":4617,"BootCount":33,"SaveCount":269,"SaveAddress":"F7000"}}
10:44:36 MQT: stat/shp5/STATUS2 = {"StatusFWR":{"Version":"6.6.0(release-sonoff)","BuildDateTime":"2019-07-06T13:10:20","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:44:37 MQT: stat/shp5/STATUS3 = {"StatusLOG":{"SerialLog":4,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["******",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008809","280500000100060500000000000000000000","00000000"]}}
10:44:37 MQT: stat/shp5/STATUS4 = {"StatusMEM":{"ProgramSize":503,"Free":500,"Heap":14,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","22B617CD","01003BC0"]}}
10:
44:38 MQT: stat/shp5/STATUS5 = {"StatusNET":{"Hostname":"shp5-7420","IPAddress":"10.0.0.132","Gateway":"10.0.0.254","Subnetmask":"255.255.255.0","DNSServer":"10.0.0.254","Mac":"2C:F4:32:A6:BC:FC","Webserver":2,"WifiConfig":4}}
10:44:38 MQT: stat/shp5/STATUS6p 14 10:44:38 2019","StartDST":"Sun Mar 31 02:00:00
 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:25","Sunset":"19:06"}}
10:44:39 MQT: stat/shp5/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
10:44:40 MQT: stat/shp5/STATUS10 = {"StatusSNS":{"Time":"2019-09-14T10:44:39","ENERGY":{"TotalStartTime":"2019-09-10T20:34:25","Total":1.492,"Yesterday":0.216,"Today":0.216,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}}
10:44:40 MQT: stat/shp5/STATUS11 = {"StatusSTS":{"Time":"2019-09-14T10:44:40","Uptime":"0T00:01:09","Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"ON","Wifi":{"AP":1,"SSId":"******","BSSId":"C2:6E:1F:50:03:C1","Channel":1,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:02"}}}

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:

10:33:01 WIF: Checking connection...
10:33:01 WIF: Connected
10:33:06 MQT: Attempting connection...
10:33:12 MQT: Connect failed to ***:4000, rc -2. Retry in 10 sec
10:33:23 MQT: Attempting connection...
10:33:30 MQT: Connect failed to***:4000, rc -2. Retry in 10 sec
10:33:33 WIF: Checking connection...
10:33:33 WIF: Connected
10:33:40 MQT: Attempting connection...
10:33:47 MQT: Connect failed to ***:4000, rc -2. Retry in 10 sec

```

Jason2866, 2019-09-14:
Try latest development version with core pre 2.6 if it solves your issue
http://thehackbox.org/tasmota/json/sonoff.bin
If not you can try this bin including a fix PR ( https://github.com/esp8266/Arduino/pull/6484 ) in Arduino core for wifi not reachable after a while
[firmware.zip](https://github.com/arendst/Sonoff-Tasmota/files/3612815/firmware.zip)

untitledlt, 2019-09-14:
Thanks for an answer!
Flashed 6.6.0.12. Let's wait few days to see how stable will it be.

untitledlt, 2019-09-16:
Working stable for 2+ days now. Thanks!
So the problem is in Arduino core, right? 
Next stable tasmota release gonna include fixed Arduino core?

Jason2866, 2019-09-16:
Thx, for the feedback! Yes, it is the Arduino Core. 
@arendst Latest feature/stage has a mandantory code fix (PR https://github.com/esp8266/Arduino/pull/6484) 
The actual stage version does work very reliable (including optimized I2C iram driver) in my test setup.
I would recommend to update the pre 2.6 version to this build.

arendst, 2019-09-16:
Agree.

Jason2866, 2019-09-16:
Will do if the Arduino guys have fixed this issue https://github.com/esp8266/Arduino/issues/6527
At the moment actual feature/stage doesnt compile seems a error in repo (software serial mismatch)

Jason2866, 2019-09-16:
Issue is fixed with commit https://github.com/esp8266/Arduino/pull/6526
Updated core pre 2.6 to latest build with optimized i2c iram

Jason2866, 2019-09-16:
@untitledlt  Next Tasmota release version will be available with core pre 2.6

arendst, 2019-09-16:
Just messed up my platformio environment. Cannot compile with pre-2.6 anymore. pio update stalls with

```
git version 2.14.2.windows.3
error: Pulling is not possible because you have unmerged files.
hint: Fix them up in the work tree, and then use 'git add/rm <file>'
hint: as appropriate to mark resolution and make a commit.
fatal: Exiting because of an unresolved conflict.
Error: VCS: Could not process command ['git', 'pull', '--recurse-submodules']
```

Jason2866, 2019-09-16:
Have you tried to delete the hidden folder `c:\Users\User\.platformio` 
Platformio will rebuild all needed...

arendst, 2019-09-16:
Sorry, didn't report back. I removed the pre-2.6 package and did the pio update.

All's fine now.

araczkowski, 2019-09-16:
I noticed similar (I think) wifi problem on Sonoff TH. Great news that this is solved - I will test too

untitledlt, 2019-09-16:
Thanks guys, you are awesome!

andrethomas2, 2019-09-21:
Closing this issue as it has been answered.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

closed 2019-09-21 by andrethomas2

MAlexanderRobinson, 2019-09-21:
Glad to hear this is fixed. A lot of my devices have the wifi suddenly drop after minutes or hours only to be helped by a reboot. Hopefully this will address my issue. Testing now with latest development version with core pre 2.6

MAlexanderRobinson, 2019-09-21:
> Glad to hear this is fixed. A lot of my devices have the wifi suddenly drop after minutes or hours only to be helped by a reboot. Hopefully this will address my issue. Testing now with latest development version with core pre 2.6

UPDATE: Erased old firmware and flashed with dev version on core 2.5. Lost connection after ~2 hours. :(
Again, unplugging and plugging back in fixes, but this is not good... will keep updating if connection keeps losing.

meingraham, 2019-09-22:
Why revert to Core 2.5.2 when 2.6 was working? Out testing shows that 2.6 is one of the best cores released since 2.3.

MAlexanderRobinson, 2019-09-23:
> Why revert to Core 2.5.2 when 2.6 was working? Out testing shows that 2.6 is one of the best cores released since 2.3.

After updating to STAGE/2.2.2-dev(38a443e), i am happy to report my problems seem to have disappeared.

schrej-zz, 2020-02-17:
Hi,
I still have permanent reconnects of the device.
Looks as if it is reconnecting every 20 sec.

`7:53:14 RSL: Haus/OT/WL_Magichome_01/tele/STATE = {"Time":"2020-02-17T17:53:14","Uptime":"0T00:18:38","UptimeSec":1118,"Heap":12,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":53,"Color":"878787","HSBColor":"0,0,53","Channel":[53,53,53],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:53:14 WIF: Checking connection...
17:53:14 WIF: Connected
17:53:29 RSL: Haus/OT/WL_Magichome_01/tele/STATE = {"Time":"2020-02-17T17:53:29","Uptime":"0T00:18:53","UptimeSec":1133,"Heap":5,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":53,"Color":"878787","HSBColor":"0,0,53","Channel":[53,53,53],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:53:37 WIF: Checking connection...
17:53:37 WIF: Connected
17:53:46 RSL: Haus/OT/WL_Magichome_01/tele/STATE = {"Time":"2020-02-17T17:53:46","Uptime":"0T00:19:10","UptimeSec":1150,"Heap":3,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":53,"Color":"878787","HSBColor":"0,0,53","Channel":[53,53,53],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:54:01 WIF: Checking connection...
17:54:01 WIF: Connected
17:54:03 RSL: Haus/OT/WL_Magichome_01/tele/STATE = {"Time":"2020-02-17T17:54:03","Uptime":"0T00:19:27","UptimeSec":1167,"Heap":12,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":53,"Color":"878787","HSBColor":"0,0,53","Channel":[53,53,53],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:54:18 RSL: Haus/OT/WL_Magichome_01/tele/STATE = {"Time":"2020-02-17T17:54:18","Uptime":"0T00:19:42","UptimeSec":1182,"Heap":12,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":53,"Color":"878787","HSBColor":"0,0,53","Channel":[53,53,53],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:54:21 WIF: Checking connection...
17:54:21 WIF: Connected
17:54:33 RSL: Haus/OT/WL_Magichome_01/tele/STATE = {"Time":"2020-02-17T17:54:33","Uptime":"0T00:19:57","UptimeSec":1197,"Heap":12,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":53,"Color":"878787","HSBColor":"0,0,53","Channel":[53,53,53],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":100,"Signal":-40,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:54:41 WIF: Checking connection...
17:54:41 WIF: Connected
17:54:48 RSL: Haus/OT/WL_Magichome_01/tele/STATE = {"Time":"2020-02-17T17:54:48","Uptime":"0T00:20:12","UptimeSec":1212,"Heap":8,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":53,"Color":"878787","HSBColor":"0,0,53","Channel":[53,53,53],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:07"}}
17:55:01 WIF: Checking connection...
17:55:01 WIF: Connected
`
Firmware is 8.1.0
Status 0:
`17:57:57 RSL: Haus/OT/WL_Magichome_01/stat/STATUS = {"Status":{"Module":34,"FriendlyName":["WL_MagicHome_01"],"Topic":"Haus/OT/WL_Magichome_01","ButtonTopic":"","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
17:57:57 RSL: Haus/OT/WL_Magichome_01/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"tasmotas","OtaUrl":"http://thehackbox.org/tasmota/release/tasmota.bin","RestartReason":"Software/System restart","Uptime":"0T00:23:21","StartupUTC":"2020-02-17T16:34:36","Sleep":50,"CfgHolder":4617,"BootCount":36,"SaveCount":120,"SaveAddress":"FA000"}}
17:57:57 RSL: Haus/OT/WL_Magichome_01/stat/STATUS2 = {"StatusFWR":{"Version":"8.1.0(tasmota)","BuildDateTime":"2019-12-25T12:33:25","Boot":7,"Core":"2_6_1","SDK":"2.2.2-dev(38a443e)","Hardware":"ESP8285","CR":"402/699"}}
17:57:57 RSL: Haus/OT/WL_Magichome_01/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":4,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Schreiner",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["0000880B","0505C8000100060000005AFF000000000000","008000C8","00000000"]}}
17:57:57 RSL: Haus/OT/WL_Magichome_01/stat/STATUS4 = {"StatusMEM":{"ProgramSize":566,"Free":436,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","01001BC0","00007881"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,29","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"}}
[... cut ...]
17:57:57 RSL: Haus/OT/WL_Magichome_01/stat/STATUS10 = {"StatusSNS":{"Time":"2020-02-17T17:57:57"}}
17:57:57 RSL: Haus/OT/WL_Magichome_01/stat/STATUS11 = {"StatusSTS":{"Time":"2020-02-17T17:57:57","Uptime":"0T00:23:21","UptimeSec":1401,"Heap":12,"SleepMode":"Dynamic","Sleep":10,"LoadAvg":99,"MqttCount":0,"POWER":"ON","Dimmer":53,"Color":"878787","HSBColor":"0,0,53","Channel":[53,53,53],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"Schreiner","BSSId":"7C:FF:4D:D0:F3:C7","Channel":11,"RSSI":100,"Signal":-41,"LinkCount":1,"Downtime":"0T00:00:07"}}}`

Jason2866, 2020-02-17:
@schrej-zz  everything okay. If you would have reconnects linkcount would be not 1 😀
Change weblog to 2.

schrej-zz, 2020-02-18:
@Jason2866,
thanks. But then I have another problem.
Device becomes more and more unresponsive via Internet until I finally have to restart.
Worth opening a separate issue?

Jason2866, 2020-02-18:
@schrej-zz  no because, it is not a general issues it is your device (hardware) behaving faulty.

Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.