arendst/Tasmota #6422: Kesen KS-601

iAmShorty, 2019-09-14:
### ISSUE DESCRIPTION - TROUBLESHOOTING
- Almost the same Problem as described here #6023
- Tasmota Firmware flashed, Switch works but resets unexpected
  

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : Kesen KS-601
- [x] Tasmota binary firmware version number used : Sonoff-Tasmota 6.6.0
- [x] Development IDE - Compiler / Upload tools used : ____ / latest esptool
- [x] Provide the output of command ``status 0`` :
```
13:41:50 CMD: status 0
13:41:50 RSL: STATUS = {"Status":{"Module":0,"FriendlyName":["KS-Schlafzimmer"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[1,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
13:41:50 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff-DE.bin","RestartReason":"External System","Uptime":"0T00:17:33","StartupUTC":"2019-09-14T12:24:17","Sleep":50,"CfgHolder":4617,"BootCount":51,"SaveCount":145,"SaveAddress":"FB000"}}
13:41:50 RSL: STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019.07.06 13:49:54","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
13:41:50 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["accesspoint",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008001","280500000100060000000000000000000000","00000000"]}}
13:41:50 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":524,"Free":476,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440C8","FlashMode":3,"Features":["00000407","0FDAE396","001783A0","22B617CD","01003BC0"]}}
13:41:50 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-0402","IPAddress":"xxx.xxx.xxx.xxx","Gateway":"xxx.xxx.xxx.xxx","Subnetmask":"255.255.255.0","DNSServer":"xxx.xxx.xxx.xxx","Mac":"xx:xx:xx:xx:xx:xx","Webserver":2,"WifiConfig":4}}
13:41:50 RSL: STATUS7 = {"StatusTIM":{"UTC":"Sat Sep 14 12:41:50 2019","Local":"Sat Sep 14 13:41:50 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:25","Sunset":"19:06"}}
13:41:50 RSL: STATUS10 = {"StatusSNS":{"Time":"2019-09-14T13:41:50"}}
13:41:50 RSL: STATUS11 = {"StatusSTS":{"Time":"2019-09-14T13:41:50","Uptime":"0T00:17:33","Heap":20,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"OFF","Wifi":{"AP":1,"SSId":"accesspoint","BSSId":"xx:xx:xx:xx:xx:xx","Channel":3,"RSSI":58,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
00:00:00 SRC: Restart
00:00:00 Projekt sonoff KS-Schlafzimmer Version 6.6.0(sonoff)-2_4_2
00:00:00 WIF: Prüfe Verbindung...
00:00:00 WIF: Verbindungsversuch...
00:00:00 WIF: verbinden mit AP1 accesspoint in Modus 11N als sonoff-0402...
00:00:01 WIF: Prüfe Verbindung...
00:00:01 WIF: Verbindungsversuch...
00:00:02 WIF: Prüfe Verbindung...
00:00:02 WIF: Verbindungsversuch...
00:00:03 WIF: Prüfe Verbindung...
00:00:03 WIF: Verbindungsversuch...
00:00:04 WIF: Prüfe Verbindung...
00:00:04 WIF: verbunden
00:00:04 HTP: Web-Server aktiv bei sonoff-0402 mit IP-Adresse xxx.xxx.xxx.xxx
13:45:17 RSL: INFO1 = {"Module":"KS-601","Version":"6.6.0(sonoff)","FallbackTopic":"cmnd/DVES_3DA192_fb/","GroupTopic":"sonoffs"}
13:45:17 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-0402","IPAddress":"xxx.xxx.xxx.xxx"}
13:45:17 RSL: INFO3 = {"RestartReason":"External System"}
13:45:17 RSL: RESULT = {"POWER":"OFF"}
13:45:17 RSL: POWER = OFF
```

### This is the old Layout which works fine

[image]

[image]

[image]

### using this settings
[image]

### This is the new Layout which produces errors

[image]

[image]

[image]

meingraham, 2019-09-14:
Since this is a newer model, it is likely that the GPIO connections have changed. You will need to determine the connections for this device using the [Configuration Procedure for New Devices](https://github.com/arendst/Sonoff-Tasmota/wiki/Configuration-Procedure-for-New-Devices).

Please close this issue and address further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

iAmShorty, 2019-09-14:
I think that's a Bug, not an Issue.

I did every single Step from the Configuration Procedure for New Devices, the GPIO to switch the light is the same as the old one and the only one which works.

meingraham, 2019-09-14:
@iAmShorty 

Even if this turns out to be a bug, the troubleshooting required to narrow down where the issue is is best done via Discord. Please follow up there.

Please provide the output of this command: `Backlog Template; GPIO`

Also, you are using 6.6.0. The latest stable release is 6.6.0.12. Please upgrade using: `Backlog OTAURL http://thehackbox.org/tasmota/json/sonoff.bin; Upgrade 1`

A tremendous amount of new development was done surrounding Tuya Dimmers and Switches resulting in the new [`TuyaMCU`](https://github.com/arendst/Sonoff-Tasmota/wiki/TuyaMCU) command.

Mike

iAmShorty, 2019-09-14:
At the moment i have no chance to head in to discord, will try later

Update is done, Switch is working but has the same problem as before.

This is the requested Output:
15:30:00 SRC: Backlog
15:30:00 RSL: RESULT = {"NAME":"KS-601","GPIO":[17,0,0,0,0,0,0,0,21,56,0,0,0],"FLAG":0,"BASE":18}
15:30:01 SRC: Backlog
15:30:01 RSL: RESULT = {"GPIO":"Not supported"}
15:30:11 WIF: Checking connection...
15:30:11 WIF: Connected

meingraham, 2019-09-14:
Have you tried this template?

https://blakadder.github.io/templates/saiham_KS-601.html

iAmShorty, 2019-09-14:
Tried at the moment, same problem when switching

Jason2866, 2019-09-14:
Do you have another switch of the same type? Device is maybe just defect!
Do you have tested the device before and you know it is working with orig. firmware?

Jason2866, 2019-09-14:
Restart reason is not a reset! Your log `13:45:17 RSL: INFO3 = {"RestartReason":"External System"}`does mean it is caused from extern. Power supply was off.
So it looks like power supply of device is weak. Since Tasmota has a lot of features it draws more current than most of the orig. firmwares

Jason2866, 2019-09-14:
You can try sonoff-basic.bin latest version from here http://thehackbox.org/tasmota/json/sonoff-basic.bin

iAmShorty, 2019-09-14:
I have another device from the same type, delivered yesterday. i thought the old one is broken and ordered the new one. with the new one, the problem is the same.

i checked the device with the original firmware and with tasmota firmware. restoring original firmware and switching via app or the wall switch works fine, but not with tasmota.

connecting the esp to the serial port and switching via app or the touch on the board works fine with both firmware versions. if i put the board back in the wall mount, the behavior is like before and it restarts unexpected.

iAmShorty, 2019-09-14:
> You can try sonoff-basic.bin latest version from here http://thehackbox.org/tasmota/json/sonoff-basic.bin
> This firmware has less features

no change :(

Jason2866, 2019-09-14:
So it is quite sure that power supply of the device is the reason for the issue.
-> Device works with powering over serial adapter (no restart)

iAmShorty, 2019-09-14:
next try

exchanging the board from the floor with the board from the new one, that works fine

meingraham, 2019-09-14:
@Jason2866 Thanks for spotting that.

@iAmShorty please close this issues as it has been resolved and it is not a Tasmota bug.

Cheers!

Mike

iAmShorty, 2019-09-14:
stick the new board in the old wall mount, same behavior, it restarts. i don't believe thats the power supply. if it's the power supply, the old one shouldn't work as well.

Jason2866, 2019-09-14:
Sorry, we cant help. We have no possibilitys to reproduce your issues since we simply do not have this device. If the devices hardware is different your findings are highly welcome. With that finding a solution is probably possible.

iAmShorty, 2019-09-14:
i have no button assigned at the moment, only relay1 to gpio 12, switching via webinterface. same behavior, unexptected restarts

Jason2866, 2019-09-14:
No, useless. Hardware is the reason for.
Last chance was that button on GPIO 0 is doing strange things.
Since you decoupled (deleted config) there is nothing left that can be tried from software side.
Some measuring on hardware is neccersary to find what is going on...

iAmShorty, 2019-09-14:
> Wait one thing left. You have erased the complete flash BEFORE flashing Tasmota

several times

> some measuring on hardware is neccersary to find what is going on...

ok, if you tell me how i can help, i will try my best :)

P.S. ich hab gesehen du kommst auch aus Deutschland, wir können uns auch auf deutsch unterhalten wenn dir das lieber ist

Jason2866, 2019-09-14:
Only english here. Check if power supply is stable. This can not done with a multimeter. You wont see short spikes. (Fast) Osci needed. 
I still think the power supply is to blame. You can try to add a cap 100uF at the 3.3V path near ESP
and a 100nF one.

iAmShorty, 2019-09-14:
> Remote hardware issue finding is most of the time not possible.
> It needs proper equipment and experience. There is no 101 guide for...

If you have the chance to do it and you want to spend time on this, i can send you the second one so that you can test?

Jason2866, 2019-09-14:
Since it has a different (better?!) board design. Sorry i can not spend time in hardware issues.
Have a family and a paid job ;-)

closed 2019-09-14 by iAmShorty

iAmShorty, 2019-09-15:
Trank you..

leecher1337, 2022-02-09:
@iAmShorty  I have a KS-601 with the same board layout. Before flashing Firmware to it, did you ever get this problem resolved or is it just not compatible?

For flashing: Is my assumption correct that the PIN-Header is:
G = GND
R = RX
T = TX
3 = VCC
?