arendst/Tasmota #6423: MQTT broker light, Webpage to display selected mqtt topics.

PepeTheFroggie, 2019-09-14:
Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**

Yes

**Is your feature request related to a problem? Please describe.**

Setting up a mqtt broker and mosquitto to display the values is a nightmare. We dont have a PC running 24/7 or a raspberry pi and we dont want mosquitto installed on every PC and every phone. We want a webpage like the tasmota main page to control switches on different devices and read sensors on all devices.

**Describe the solution you'd like**

I would like tasmota to have an optional mtqq broker "light" installed, and a webpage to display selected values.

**Describe alternatives you've considered**

It dooesent have t be mqtt, anything simple will do.

The alternative would be a webpage served by one tasmota device including text and buttons served from other tasmota devices. With images this would look like a table of
[image]
[image]
No idea how to do that with active buttons.

**Additional context**
_Add any other context or screenshots about the feature request here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-09-14:
Feel free to fork and do as you like, it is allowed since it is open source.

meingraham, 2019-09-14:
Unfortunately, true home automation is not simple. The outcome of automation is to make life simpler ;-)

`Setting up a mqtt broker and mosquitto to display the values is a nightmare.` An opinion, and, in my opinion, pretty inaccurate. I say this as someone who knew NOTHING when I started this journey.

- RPi ZeroW & USB power supply - $5 (you probably already have multiple old smartphone charger)
- Flash RaspianLite - 10 minutes
- Modify two files: ssh (empty) and wpa_supplicant.conf (edit to add Wi-Fi credentials)
- Boot RPi & remotely connect to it
- Install Mosquitto (1 command)
- Modify three files: mosquitto.conf (defaults), passwd, conf.d/listeners (1 line)
- Restart mosquitto

I actually did this yesterday. OK, not my first time, but no more than 30 minutes work. OK, 1 hour with learning curve. Then pretty much forget it.

Mike

Jason2866, 2019-09-14:
Looked in wiki? https://github.com/arendst/Sonoff-Tasmota/wiki/Integrations#device-management

joba-1, 2019-09-15:
@meingraham: where do you get a pi0w with powersupply for $5? I want that please! I paid $10+shipping.

Other than that. Fully agree.

In addition: implementing that (based on mqtt or other) is not possible on an esp8266. You need too much RAM to queue incoming messages and too many simultaneous network connections to publishers and subscribers.

meingraham, 2019-09-15:
@joba-1 

I bought a couple of Pi0W at a large electronics store (MicroCenter in Virginia). For a ZeroW, an old phone microUSB plug (i.e., $0) is enough power for running a lightweight mosquitto instance.

But even for $10...

laurentdong, 2019-09-16:
Actually I'm currently running mosquitto on my router with OpenWRT which works pretty well. My router is TPLink Archer 2600.

meingraham, 2019-09-16:
Many ways to skin the cat... If you have a Linux "server" (e.g., a router with openWRT) already, then install mosquitto and modify 3 files.

The end of "My Cousin Vinny" comes to mind ;-)

closed 2019-09-16 by PepeTheFroggie