arendst/Tasmota #6428: MQT: Connect failed to IP:1883, rc 5. Retry in 10 sec

nightcat91, 2019-09-15:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Freshly flashed MagicHome (Arilux) LC-02 RGBW Controller cant connect to MQTT Broker.

11:26:54 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : MagicHome (Arilux) RGBW Controller LC-02
- [x] Tasmota binary firmware version number used : 6.6.0 self compiled using Visual Code
- [x] Development IDE - Compiler / Upload tools used : Visual Code uploaded with ESPHome Flasher program from a Win10 PC
- [x] Provide the output of command ``status 0`` :
```
11:29:02 CMD: status 0
11:29:02 RSL: stat/magichome01/STATUS = {"Status":{"Module":1,"FriendlyName":["MagicHome01 Bett Nachtlicht"],"Topic":"magichome01","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":7,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":1}}
11:29:02 RSL: stat/magichome01/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"h801s","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:07:33","StartupUTC":"2019-09-15T09:21:29","Sleep":50,"CfgHolder":4617,"BootCount":1,"SaveCount":3,"SaveAddress":"F9000"}}
11:29:02 RSL: stat/magichome01/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-09-14T11:52:05","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
11:29:02 RSL: stat/magichome01/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["NaitsabesNOT","sebsmarthome"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00088029","050500000500060000000000000000000000","00000020"]}}
11:29:02 RSL: stat/magichome01/STATUS4 = {"StatusMEM":{"ProgramSize":471,"Free":532,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14405E","FlashMode":3,"Features":["00000809","00DAA196","000783A0","22B61FCD","01003BC0"]}}
11:29:02 RSL: stat/magichome01/STATUS5 = {"StatusNET":{"Hostname":"magichome01","IPAddress":"10.11.14.101","Gateway":"10.11.1.1","Subnetmask":"255.255.0.0","DNSServer":"10.11.10.108","Mac":"A0:20:A6:1D:84:00","Webserver":2,"WifiConfig":5}}
11:29:02 RSL: stat/magichome01/STATUS6 = {"StatusMQT":{"MqttHost":"10.11.11.1","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1D8400","MqttUser":"magichome01","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:29:02 RSL: stat/magichome01/STATUS7 = {"StatusTIM":{"UTC":"Sun Sep 15 09:29:02 2019","Local":"Sun Sep 15 11:29:02 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99}}
11:29:02 RSL: stat/magichome01/STATUS10 = {"StatusSNS":{"Time":"2019-09-15T11:29:02"}}
11:29:02 RSL: stat/magichome01/STATUS11 = {"StatusSTS":{"Time":"2019-09-15T11:29:02","Uptime":"0T00:07:33","Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":27,"POWER":"OFF","Wifi":{"AP":1,"SSId":"NaitsabesNOT","BSSId":"3A:E8:29:9D:D5:94","Channel":9,"RSSI":84,"LinkCount":1,"Downtime":"0T00:00:03"}}}
```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
11:30:08 CMD: weblog 4
11:30:08 RSL: stat/magichome01/RESULT = {"WebLog":4}
11:30:09 CFG: Saved to flash at F8, Count 4, Bytes 3584
11:30:09 WIF: Checking connection...
11:30:09 WIF: Connected
11:30:09 DNS: MDNS.update
11:30:15 DNS: Query done. MQTT services found 0
11:30:15 MQT: Attempting connection...
11:30:15 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec
11:30:27 DNS: Query done. MQTT services found 0
11:30:27 MQT: Attempting connection...
11:30:27 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec
11:30:31 WIF: Checking connection...
11:30:31 WIF: Connected
11:30:31 DNS: MDNS.update
11:30:39 DNS: Query done. MQTT services found 0
11:30:39 MQT: Attempting connection...
11:30:39 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec
```
**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-09-15:
Sorry cant reproduce. Not a Tasmota issue. Try precompiled sonoff version.

nightcat91, 2019-09-15:
I tried an OTA Update using the sonoff-basic.bin (precompiled binary from the release page) -> did not help the issue
Then I did a configuration reset -> did not help

I then reflashed the device using the sonoff-basic.bin (with wipe of the flash, this is done always by ESPHome Flasher) and went into its access point and typed in my wifi config.
Went onto the web page and typed in the mqtt info, restarted and had the same problem:

00:00:06 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec

and the same log message in my mqtt broker:

1568552588: New connection from 10.11.2.229 on port 1883.
1568552588: Socket error on client <unknown>, disconnecting.
1568552597: New connection from 10.11.2.229 on port 1883.
1568552597: Socket error on client <unknown>, disconnecting.
1568552609: New connection from 10.11.2.229 on port 1883.
1568552609: Socket error on client <unknown>, disconnecting.

nightcat91, 2019-09-15:
Here is the whole serial log from the point of flashing the sonoff-basic.bin on the device

```
Leaving...
Hard Resetting...
Done! Flashing is complete!

Showing logs:
[15:01:18]r[15:01:18]00:00:00 CFG: Use defaults
[15:01:18]00:00:00 Project sonoff Sonoff Version 6.6.0(release-basic)-2_3_0
[15:01:18]00:00:00 WIF: WifiManager active for 3 minutes
[15:01:19]00:00:00 HTP: Web server active on sonoff-1024 with IP address 192.168.4.1
[15:01:57]00:00:39 WIF: Hostname %s-%04d, SSId1 NaitsabesNOT, SSId2 
[15:01:58]00:00:40 APP: Restarting
[15:01:58]
[15:01:58] ets Jan  8 2013,rst cause:1, boot mode:(3,7)
[15:01:58]
[15:01:58]load 0x4010f000, len 1384, room 16 
[15:01:58]tail 8
[15:01:58]chksum 0x2d
[15:01:58]csum 0x2d
[15:01:58]v60000608
[15:01:58]~ld
[15:01:58]
[15:01:58]00:00:00 CFG: Loaded from flash at F8, Count 4
[15:01:58]00:00:00 Project sonoff Sonoff Version 6.6.0(release-basic)-2_3_0
[15:01:59]00:00:00 WIF: Connecting to AP1 NaitsabesNOT in mode 11N as sonoff-1024...
[15:02:05]00:00:06 WIF: Connected
[15:02:05]00:00:06 HTP: Web server active on sonoff-1024 with IP address 10.11.2.229
[15:02:07]14:02:05 RSL: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"6.6.0(release-basic)","FallbackTopic":"cmnd/DVES_1D8400_fb/","GroupTopic":"sonoffs"}
[15:02:07]14:02:05 RSL: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-1024","IPAddress":"10.11.2.229"}
[15:02:07]14:02:05 RSL: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
[15:02:07]14:02:05 RSL: stat/sonoff/RESULT = {"POWER":"OFF"}
[15:02:07]14:02:05 RSL: stat/sonoff/POWER = OFF
[15:02:15]14:02:14 RSL: tele/sonoff/STATE = {"Time":"2019-09-15T14:02:14","Uptime":"0T00:00:16","Vcc":3.149,"Heap":18,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":39,"POWER":"OFF","Wifi":{"AP":1,"SSId":"NaitsabesNOT","BSSId":"3A:E8:29:9D:D5:94","Channel":9,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:06"}}
[15:03:08]14:03:07 RSL: tele/sonoff/LWT = 
[15:03:08]14:03:07 MQT: MqttHost 10.11.11.1, MqttPort 1883, MqttClient DVES_%06X, MqttUser magichome01, Topic magichome01, FullTopic %prefix%/%topic%/
[15:03:09]14:03:08 MQT: Attempting connection...
[15:03:09]14:03:08 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec
[15:03:10]14:03:09 APP: Restarting
[15:03:11]
[15:03:11] ets Jan  8 2013,rst cause:1, boot mode:(3,7)
[15:03:11]
[15:03:11]load 0x4010f000, len 1384, room 16 
[15:03:11]tail 8
[15:03:11]chksum 0x2d
[15:03:11]csum 0x2d
[15:03:11]v60000608
[15:03:11]~ld
[15:03:11]
[15:03:11]00:00:00 CFG: Loaded from flash at F5, Count 7
[15:03:11]00:00:00 Project sonoff Sonoff Version 6.6.0(release-basic)-2_3_0
[15:03:11]00:00:00 WIF: Connecting to AP1 NaitsabesNOT in mode 11N as magichome01-1024...
[15:03:16]00:00:04 WIF: Connected
[15:03:16]00:00:04 HTP: Web server active on magichome01-1024 with IP address 10.11.2.229
[15:03:18]00:00:06 MQT: Attempting connection...
[15:03:18]00:00:06 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec
[15:03:30]14:03:29 MQT: Attempting connection...
[15:03:30]14:03:29 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec
[15:03:42]14:03:41 MQT: Attempting connection...
[15:03:42]14:03:41 MQT: Connect failed to 10.11.11.1:1883, rc 5. Retry in 10 sec
[15:03:43]
Serial port closed!
```

arendst, 2019-09-15:
See https://github.com/arendst/Sonoff-Tasmota/wiki/MQTT-Overview#status-codes

Check your mqttpassword.

nightcat91, 2019-09-15:
How did I not find that info page....bookmarked! 

Very sorry for this!!
There was indeed a mismatch in the username on the mqtt broker side.

Thanks for the help and sorry for the huge bug report for such a small issue

closed 2019-09-15 by nightcat91

ellnic, 2022-02-20:
I had the opposite problem. I've never had the issue before with my own setup, but was recently helping a relative with a new HA installation. I could not get a single Sonoff Mini R2 to connect. The pass was a randomly generated 20 char ASCII. I changed to just upper, lower and numerical after reading this issue, and it resolved it!

JurandirHora, 2022-03-05:
I had the same problem.
I did all the steps indicated in this post, however nothing worked. It only works when I downgrade the tasmota firmware to version 8.5.1.
I suspect that the developer has enabled TLS of the updated version and so it is possible that the MQTT server does not accept the common port (I haven't tested this hypothesis yet!)

barbudor, 2022-03-05:
If you are using TLS, you MUST use port 8883