arendst/Tasmota #6429: Is it only 6.6.0.12 P1 Smart Meter tariff configuration ?

itProfi, 2019-09-15:
Can i use several tariffs with BlitzWolf sockets, for example, for use 2 tariffs, or onle P1 smart meter only?

arendst, 2019-09-15:
Cryptic question but I try to answer:

- You can use tariffs with any energy monitoring device except SDM630 but that is in the works.
- Only one Domoticz P1 smart meter option is supported.

itProfi, 2019-09-15:
Sorry for my nub question, but where i can see 2 tariffs - ?
[image]

arendst, 2019-09-15:
Only in Domoticz (ha,ha)

itProfi, 2019-09-15:
And no chances via web or mqtt? :(

arendst, 2019-09-15:
Nope. Currently not implemented.

itProfi, 2019-09-15:
I and ALL USERS in RUSSIA say "WOW" if it is implemented, because we have 2 tariffs night (23h) and day (7h)...:)

arendst, 2019-09-15:
Just have a look at domoticz which could llok like this:

[image]

with tariff ``{"Tariff":{"Off-Peak":[23,22],"Standard":[7,6],"Weekend":"ON"}}``

itProfi, 2019-09-15:
> Just have a look at domoticz which could llok like this:
> 
> [image]
> 
> with tariff `{"Tariff":{"Off-Peak":[23,22],"Standard":[7,6],"Weekend":"ON"}}`

Pretty::) but we want via mqtt:)

arendst, 2019-09-15:
Fixed:

```
18:04:39 MQT: tele/pow3/SENSOR = {"Time":"2019-09-15T18:04:39","ENERGY":{"TotalStartTime":"2019-09-06T00:00:00","Total":[6.377,0.399,5.979],"Yesterday":0.039,"Today":0.053,"Period":0.02,"Power":1.52,"ApparentPower":10.03,"ReactivePower":9.90,"Factor":0.15,"Voltage":219.1,"Current":0.046}}
```
Tariff info in ``"Total":[6.377,0.399,5.979]`` where first value is total, second is tariff1 total and third is tariff2 total

itProfi, 2019-09-15:
Thanks, I try some later!

itProfi, 2019-09-21:
When i upgrade firmware with lastest commits,
[image]
after EnergyReset1 0, EnergyReset2 0, EnergyReset3 0 -  i see in consol:
"19:48:23 MQT: tele/pump/SENSOR = {"Time":"2019-09-21T19:48:23","ENERGY":{"TotalStartTime":"2019-09-21T19:43:51","Total":[0.02,2608.00,75239.00],"Yesterday":0.00,"Today":0.02,"Period":0,"Power":3,"ApparentPower":14,"ReactivePower":13,"Factor":0.22,"Voltage":228,"Current":0.06}} "
2608.00,75239.00 - what is this and how reset it ?

arendst, 2019-09-21:
Try EnergyReset4 and EnergyReset5

arendst, 2019-09-21:
...and do not change the header of a closed issue please.

Open a new one as now it's unclear to everyone where this is about.

itProfi, 2019-09-21:
Ok, i see, my fault..
When i EnergyReset4 0 and EnergyReset5 0 i see:
"20:12:06 MQT: tele/pump/SENSOR = {"Time":"2019-09-21T20:12:06","ENERGY":{"TotalStartTime":"2019-09-16T11:21:22","Total":[0.02,0.00,595.00],"Yesterday":0.00,"Today":0.02,"Period":0,"Power":3,"ApparentPower":14,"ReactivePower":14,"Factor":0.23,"Voltage":235,"Current":0.06}}"
595 - not reset..(

itProfi, 2019-09-21:
Wnen i try EnergyReset2 0 and EnergyReset3 0 isee:"20:32:21 MQT: tele/pump/SENSOR = {"Time":"2019-09-21T20:32:20","ENERGY":{"TotalStartTime":"2019-09-21T20:32:07","Total":[0.00,4294967240.00,572.00],"Yesterday":0.00,"Today":0.00,"Period":0,"Power":3,"ApparentPower":13,"ReactivePower":13,"Factor":0.22,"Voltage":240,"Current":0.05}}

arendst, 2019-09-21:
EnergyReset4 0,0

itProfi, 2019-09-21:
Yes, i'ts work! 
:20:35:30 MQT: tele/pump/SENSOR = {"Time":"2019-09-21T20:35:30","ENERGY":{"TotalStartTime":"2019-09-21T20:32:07","Total":[0.02,0.00,0.00],"Yesterday":0.00,"Today":0.02,"Period":0,"Power":3,"ApparentPower":14,"ReactivePower":13,"Factor":0.22,"Voltage":235,"Current":0.06}} "
Are write it in wiki in future?

meingraham, 2019-09-21:
@arendst 

Can you provide syntax and explanation of `EnergyReset 4` & `EnergyReset5`? The comma separated parameters differ from 1..3.

itProfi, 2019-09-22:
After 13 hours i see:
"09:40:31 MQT: tele/pump/SENSOR = {"Time":"2019-09-22T09:40:31","ENERGY":{"TotalStartTime":"2019-09-22T06:59:59","Total":[0.27,242.00,0.00],"Yesterday":0.00,"Today":0.27,"Period":0,"Power":4,"ApparentPower":15,"ReactivePower":15,"Factor":0.29,"Voltage":246,"Current":0.06}}"
What is "Total":[0.27,242.00,0.00]? And why tariff2 is zero?

itProfi, 2019-09-22:
When pump is on - i see 
````
09:47:11 MQT: tele/pump/STATE = {"Time":"2019-09-22T09:47:11","Uptime":"0T13:20:52","UptimeSec":48052,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"MALINOVAYA","BSSId":"50:FF:20:05:1D:8E","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
09:47:11 MQT: tele/pump/SENSOR = {"Time":"2019-09-22T09:47:11","ENERGY":{"TotalStartTime":"2019-09-22T06:59:59","Total":[0.29,263.00,0.00],"Yesterday":0.00,"Today":0.29,"Period":2,"Power":759,"ApparentPower":1155,"ReactivePower":870,"Factor":0.66,"Voltage":233,"Current":4.95}}
09:47:21 MQT: tele/pump/STATE = {"Time":"2019-09-22T09:47:21","Uptime":"0T13:21:02","UptimeSec":48062,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"MALINOVAYA","BSSId":"50:FF:20:05:1D:8E","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
09:47:21 MQT: tele/pump/SENSOR = {"Time":"2019-09-22T09:47:21","ENERGY":{"TotalStartTime":"2019-09-22T06:59:59","Total":[0.29,265.00,0.00],"Yesterday":0.00,"Today":0.29,"Period":2,"Power":675,"ApparentPower":1211,"ReactivePower":1005,"Factor":0.56,"Voltage":234,"Current":5.17}}
09:47:31 MQT: tele/pump/STATE = {"Time":"2019-09-22T09:47:31","Uptime":"0T13:21:12","UptimeSec":48072,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"MALINOVAYA","BSSId":"50:FF:20:05:1D:8E","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
09:47:31 MQT: tele/pump/SENSOR = {"Time":"2019-09-22T09:47:31","ENERGY":{"TotalStartTime":"2019-09-22T06:59:59","Total":[0.29,266.00,0.00],"Yesterday":0.00,"Today":0.29,"Period":2,"Power":668,"ApparentPower":914,"ReactivePower":625,"Factor":0.73,"Voltage":232,"Current":3.95}}
09:47:41 MQT: tele/pump/STATE = {"Time":"2019-09-22T09:47:41","Uptime":"0T13:21:22","UptimeSec":48082,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"MALINOVAYA","BSSId":"50:FF:20:05:1D:8E","Channel":9,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:04"}}
09:47:41 MQT: tele/pump/SENSOR = {"Time":"2019-09-22T09:47:41","ENERGY":{"TotalStartTime":"2019-09-22T06:59:59","Total":[0.29,268.00,0.00],"Yesterday":0.00,"Today":0.29,"Period":2,"Power":722,"ApparentPower":1154,"ReactivePower":900,"Factor":0.63,"Voltage":231,"Current":5.00}}
~~~~

arendst, 2019-09-22:
I see the same results. I know the cause. The change implementer forgot the kwh/wh difference on most sensors.

I will investigate.......

arendst, 2019-09-22:
Regarding EnergyReset4 and 5:
- EnergyReset4 takes up to two parameters. It resets or inits the offset of the tariff total usage counters. So ``EnergyReset4 0`` resets tariff1 total usage counter. ``EnergyReset4 0,0`` resets both tariff total usage counter.
- EnergyReset5 does the same for both tariff total return (or produced) counters.

pablozg, 2019-09-22:
> I see the same results. I know the cause. The change implementer forgot the kwh/wh difference on most sensors.
> 
> I will investigate.......

Yes, It's my fault. 

@arendst what do you think about to add to the Total field, the return1 / return2 fields?

Somethink like:
````
"Total":[560.29,410.29,150.00,800.00,1500.00]
````
or better at a new 'export' field:
````
"Export":[800.00,1500.00]
````

meingraham, 2019-09-22:
@arendst what is `return (or produced)` energy totals?

pablozg, 2019-09-22:
Hi @meingraham, some meters like the SDM120 can meter the energy injected to the grid (export or produced), in my case I can inject to energy to the grid because I have a solar instalation, but some people can have a wind turbine, etc.

arendst, 2019-09-22:
I'll add the optional TotalExport

arendst, 2019-09-22:
Just noticed there is already an ExportActive field. I'll add the return values there.

itProfi, 2019-09-22:
Thanks for all!

closed 2019-09-22 by itProfi