arendst/Tasmota #6430: Longer delay - I2C Soil moisture sensor vs Chirp! Plant watering alarm

genestealer, 2019-09-15:
### BUG DESCRIPTION
I2C pre-read I2C delay too short for the "Chirp! Plant watering alarm" version of the "I2C Soil moisture sensor", made by the same person.
The "I2C Soil moisture sensor" is a standalone specifically designed to act as a sensor, thus it can be read faster than Chirp!. 

### REQUESTED INFORMATION
Examples from their website show a 20ms read delay for the I2C sensor and a 1100ms delay for the Chirp. Can we please add this "delay" to xsns_48_chirp.ino.
Ref: https://wemakethings.net/chirp/

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [x] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [x] Device used (i.e. Sonoff Basic) : NodeMCU ESP8266 Sonoff Generic 
- [x] Tasmota binary firmware version number used : self-compiled 2019 09 15 Sonoff-Tasmota-development, due to enabling CHIRP https://github.com/arendst/Sonoff-Tasmota/wiki/CHIRP-Soil-Moisture-Sensor)
- [x] Development IDE - Compiler / Upload tools used : Atom Platform.IO
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE:
00:02:22 CMD: status 0
00:02:22 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:02:22 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:02:17","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":36,"SaveCount":48,"SaveAddress":"F4000"}}
00:02:22 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.12(sonoff)","BuildDateTime":"2019-09-15T18:35:51","Boot":0,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
00:02:22 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT_VLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A64000000000000","00000000"]}}
00:02:22 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":560,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","21001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34,48"}}
00:02:22 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2018","IPAddress":"192.168.2.26","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.21","Mac":"60:01:94:49:C7:E2","Webserver":2,"WifiConfig":4}}
00:02:22 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.21","MqttPort":1883,"MqttClientMask":"sonoff2","MqttClient":"sonoff2","MqttUser":"sonoff2","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:02:22 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:02:22 1970","Local":"Thu Jan 01 00:02:22 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:02:22 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:02:22","CHIRP0":{"Moisture":65535,"Temperature":-0.1,"Illuminance":"500},"TempUnit":"C"}}
00:02:22 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:02:22","Uptime":"0T00:02:17","UptimeSec":137,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":40,"MqttCount":1,"Wifi":{"AP":1,"SSId":"IoT_VLAN","BSSId":"02:1D:AA:DD:B2:08","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:08"}}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
00:03:08 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:03:07","CHIRP0":{"Moisture":65535,"Temperature":-0.1,"Illuminance":"500},"TempUnit":"C"}}

```

### TO REPRODUCE
Use a "Chirp! Plant watering alarm" and not a "I2C Soil moisture sensor"

### EXPECTED BEHAVIOUR
Moisture should change, but is stuck at it's highest value of 65535

Staars, 2019-09-15:
Yes, this should be possible.

To be honest, I am not too surprised. I know about this 20ms-delay, but as the built-in-helper function of Tasmota worked for my device, I decided to go for it in the first place to save some code space.
I can prepare a new PR (probably tomorrow).

Staars, 2019-09-16:
@genestealer 
Can you please just for testing  add the delay into the currently used function:

Please insert:
`delay(1100);`

between the following lines of sonoff/support.ino
```
    if (0 == Wire.endTransmission(false)) {             // Try to become I2C Master, send data and collect bytes, keep master status for next request...
      Wire.requestFrom((int)addr, (int)size);           // send data n-bytes read
```

so the result should be:
```
if (0 == Wire.endTransmission(false)) {             // Try to become I2C Master, send data and collect bytes, keep master status for next request...
delay(1100);
Wire.requestFrom((int)addr, (int)size);           // send data n-bytes read
```

This will definitely NOT(!!!) be the final solution. I just want to figure out, if the delay works. After that, I would rewrite the loop-function of the driver to become non-nlocking.

Staars, 2019-09-16:
I should note, that this might lead to a crash, because the driver will block the main loop for more than 2 seconds, but if we get at least a few readings, this would help.
Set TELEPERIOD to at least 20 seconds.

genestealer, 2019-09-16:
Hi, I added the line re-flashed, set TELEPERIOD to 20 (was 600) and a running Serial log level 3, I see the following:
1. On boot it gets the Moisture and Illuminance, but not the Temperature or FW-version.
2. Illuminance seems to be inverted, with a higher number in darkness ( E.G. 2956 lx in the dark, 700 lx in light)
This seems to be working 👍 
But in the log, what arethe numbers in "CHIRP: timeout: 10, tele: 20"

Serial output:

```
l␏s��N␌␌"␌�␎$��␓��$�l��{l��n�␃
00:00:00 CFG: Loaded from flash at FA, Count 26
00:00:00 APP: Some settings have been reset (2)
00:00:00 SRC: Restart
00:00:00 Project sonoff Sonoff Version 6.6.0.12(sonoff)-STAGE
00:00:00 SHT: Sensor did not ACK command
00:00:00 CHIRP: scan will start ...
00:00:02 I2C: CHIRP: found at 0x20
00:00:02 CHIRP: fw 255
00:00:02 Found 1 CHIRP sensor(s).
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Connecting to AP1 IoT_VLAN in mode 11N as sonoff-2018...
00:00:03 CHIRP: reset all
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 CHIRP: call CapTemp twice
00:00:06 CHIRP: now really read CapTemp for sensor at address 0x20
00:00:08 CHIRP: now really read CapTemp for sensor at address 0x20
00:00:10 CHIRP: call measure light
00:00:11 CHIRP: busy status for light for sensor 255
00:00:11 CHIRP: init measure light for sensor 0
00:00:14 WIF: Connected
00:00:14 HTP: Web server active on sonoff-2018 with IP address 192.168.2.26
00:00:17 CHIRP: call read light
00:00:17 CHIRP: will read light for sensor 0
00:00:18 CHIRP: light read success
00:00:19 CHIRP: paused, waiting for TELE
00:00:20 CHIRP: paused, waiting for TELE
00:00:21 APP: Boot Count 16
00:00:21 CHIRP: paused, waiting for TELE
00:00:21 CFG: Saved to flash at F9, Count 27, Bytes 4096
00:00:22 CHIRP: paused, waiting for TELE},"TempUnit":"C"}
00:00:56 CHIRP: timeout: 10, tele: 20
00:01:08 CHIRP: call CapTemp twice
00:01:08 CHIRP: now really read CapTemp for sensor at address 0x20
00:01:10 CHIRP: now really read CapTemp for sensor at address 0x20
00:01:12 CHIRP: call measure light
00:01:14 CHIRP: busy status for light for sensor 255
00:01:14 CHIRP: init measure light for sensor 0
00:01:16 CHIRP: call read light
00:01:16 CHIRP: will read light for sensor 0
00:01:17 CHIRP: light read success
```

[image]

[image]

Staars, 2019-09-17:
Okay, thanks a lot for your report.

These debug-messages have sometimes strange names and I usually only want to see, if all functions are called or where the driver does crash. In the case you mentioned, I only wanted to see, if the timeout for the next sensor-read-cycle is correctly calculated based on the TELEPERIOD.

Illuminance is a unit-less value, which is described by the hardware developer. I tried to explain this in the WIKI-entry and I know, that 'lx' is misleading here. As I did not find a convincing method to calculate real lux values from these raw values, I decided to pass the raw values. They should be okay to decide if it is dark or light. I am totally open to suggestions, how to deal with it.

Here is my plan:
1. Yesterday I ordered a original CHIRP-sensor, so that I can test on real hardware. I wanted to do real-world-tests with multiple sensors anyway.

2. I started a new driver locally, which already compiles and I hope, that I can test it with my ("no-chirping")-sensor. If this does not break, I would upload it to my fork to make it possible for you to test it.

3. But I would like to wait with a new PR until I have the same sensor in my hands, which might last 2 weeks or so.

Questions to you:
1. Do you have an original chirp-sensor or a chinese replica. I am just curious ;)
2. Can you read temperature from it with an Arduino-sketch and if yes, can you point me to that sketch.

All in all I must admit, that I really misinterpreted the information about the timing differences of the chirp- and non-chirp-version. But thankfully you reported this here.

Staars, 2019-09-17:
I have a new (unfinished) version here:
https://github.com/Staars/Sonoff-Tasmota/blob/chirp/sonoff/xsns_48_chirp.ino

This does still work with my (fast) device and should slow down the read process. Please give it a try.

I will further experiment and hope to test it with my own "real" CHIRP-sensor soon, including test with multiple sensors.
After that (and some clean-up) I would prepare a PR.

genestealer, 2019-09-17:
Hi,
I have both an original "I2C Soil moisture sensor" from Tindie and about 8 of the "Chirp! - plant watering alarm" from China.
I've also added contributed to the https://github.com/Apollon77/I2CSoilMoistureSensor/pull/19

Ops, by bad, looking at the two schematics, the Chirp! is without a thermistor, so cannot report the temperature, it does have an LED so that should work, but not the temperature:
Sensor version: https://github.com/Miceuz/i2c-moisture-sensor/blob/master/i2c-moist-sensor-schematics.pdf
Chirp! version: https://github.com/Miceuz/PlantWateringAlarm/blob/master/schematics.png

genestealer, 2019-09-17:
Just tried your linked unfinished version, downloaded the file and replaced it in the  .\sonoff folder, I assume I should have removed the "delay(1100);" from support.ino. It seems to be working brilliantly, updating in a nice responsive manner.

[image]

[image]

```
00:03:08 CMD: status 0
00:03:08 RSL: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:03:08 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"External System","Uptime":"0T00:03:08","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":20,"SaveCount":33,"SaveAddress":"FB000"}}
00:03:08 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.12(sonoff)","BuildDateTime":"2019-09-17T22:52:34","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
00:03:08 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":3,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IoT_VLAN",""],"TelePeriod":20,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A64000000000000","00000000"]}}
00:03:08 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":559,"Free":444,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B617CD","21001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34,48"}}
00:03:08 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-2018","IPAddress":"192.168.2.26","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.21","Mac":"60:01:94:49:C7:E2","Webserver":2,"WifiConfig":4}}
00:03:08 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_49C7E2","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:03:08 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:03:08 1970","Local":"Thu Jan 01 00:03:08 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:03:08 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:03:08","CHIRP0":{"Moisture":335,"Temperature":-0.1,"Illuminance":"672},"TempUnit":"C"}}
00:03:08 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:03:08","Uptime":"0T00:03:08","UptimeSec":188,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":0,"Wifi":{"AP":1,"SSId":"IoT_VLAN","BSSId":"02:1D:AA:DD:B2:08","Channel":1,"RSSI":100,"LinkCount":1,"Downtime":"0T00:00:20"}}}
```

Staars, 2019-09-18:
@genestealer 
Thanks for reporting.

I will add a small feature to ignore false (== no) temperature readings.

Design wise it could make sense to NOT name the light value ILLUMINANCE, but I would stick with the raw value, because this will stay comparable to the various example sketches. 
IMO a better name would be DARKNESS or DARK LEVEL. Better ideas anyone??

genestealer, 2019-09-18:
My vote would be for "darkness".

Staars, 2019-09-18:
@genestealer
Small update, if you like to test.

Staars, 2019-09-24:
I still have no new hardware (= the "slow" CHIRP!).

@genestealer 
Can you confirm, that the current version works for your setup?:
https://github.com/Staars/Sonoff-Tasmota/blob/chirp/sonoff/xsns_48_chirp.ino

If this is the case, I would launch the PR (and do my personal tests afterwards).

Staars, 2019-09-27:
[image]

`18:20:43 MQT: tele/sonoff/SENSOR = {"Time":"2019-09-27T18:20:43","CHIRP0":{"Moisture":192,"Darkness":3183},"CHIRP1":{"Moisture":244,"Temperature":24.8,"Darkness":29309},"TempUnit":"C"}`

Got now my "real" Chirp and do some tests. I hope to get it finished this weekend.

genestealer, 2019-09-27:
Awesome, brilliant that you also have multiple working on the same bus, sorry I didn't get chance to test with my clone, they should be identical as the original designer's version and will be able to test it next week when I'm back home. Interestingly reading the designers website their biggest hopes were so it would be cloned. "most win for the project would be if it would get cloned by the Chinese" https://wemakethings.net/2012/06/21/plant_watering_alarm/

Staars, 2019-09-28:
Well, this „real slow“ Chirp!-thing is a bit complicated and meanwhile I was even capable of bricking the I2C-communication 🤦‍♂️.

It behaves differently in some points:
1. I could never read a firmware version from it, because I think 255 is simply -1 and this is only an error code. This would be only a minor problem, but it does not really work.
2. Changing of the sensor address with CHIRPSET works differently. On the (non-chirping)-Sensor-only-version this works on the fly. The CHIRP! needs a reset or cold boot.
I fear, that I bricked the I2C-communication somewhere in that process and ATM my CHIRP!-sensor really only chirps :). It is invisible on the I2C-bus and I expect, that I will have to re-flash the firmware.

My conclusion so far:
1. The recommended sensor version is the one without the Chirp.
2. Be careful with address-changes, means: Do it only with one connected sensor and do a reset/power down afterwards. Do it only when you need to!!
3. Do not expect to have a reliable connection to multiple sensors on one bus. Although it did work most of the time, I did not get all sensors on every reboot.

BTW, meanwhile I received two of the chinese clones and both came dead like stones. No chirp, no I2C, no nothing. The rate of defective devices/sensors has increased a lot (for me) in the last few months ,but that is another topic ...

Staars, 2019-09-29:
@genestealer
If you are back to your computer, you might test my last changes:
https://github.com/Staars/Sonoff-Tasmota/blob/chirp/sonoff/xsns_48_chirp.ino

I had a "funny" day with my sensors yesterday and here is my updated point of view:

1. The original Chirp!-firmware does not report a firmware version and I believe, we can assume that fw-version 255 is a Chirp!, which is not too bad.
2. I took a look at the various firmware's for the Chirp! and the I2C-soil-moisture-sensor. With minor changes, it is possible to compile it with the toolchain, that you will get with your usual vanilla Arduino-installation and the board-additions for the Attiny-family.
3. As I am by no means a hardware-guru (= no special programmers at home), I tried to flash with an Arduino-ISP. To my surprise this only worked with one of my "dead" chinese replicas, which came to live. The funny sidenote is, that this has now I2C-soil-moisture-fw on the Chirp!-hardware, which means it reports non-sense for the temperature, but is rock-stable.
Besides with my hardware it is not reliable to flash it that way and I assume, this is related to inferior cabeling. 

I did some small changes to the code and hope, it will work for all I2C-sensor of that family.

Last verdict (for Tasmota-users):
!! DO NOT USE THE CHIRP!, TAKE THE OTHER ONE !! :smile:

meingraham, 2019-09-29:
@Staars Are you updating the Chirp article to include your findings? Is there an alternative sensor to recommend instead?

Staars, 2019-09-29:
Yes, I will update the Wiki.

"The other one" is simply this one (with no catchy name):
https://github.com/Miceuz/i2c-moisture-sensor

I became successfully confused in the beginning and incorrectly assumed, that both versions are identical under the hood - just with or without chirping. But that is not the case and only after this issue was openend, things became clear to me.

But in the end I hope, both sensors can be supported (it seems to work now).

Staars, 2019-09-29:
[image]

Okay, I could recover my "real" Chirp-sensor and will leave some info about it in the WIKI later.

I hope, that most of the problems are solved now.

sirLeone, 2019-10-01:
Where do you get these cheap sensor from ? Ali ?

Staars, 2019-10-02:
I bought two of these chinese versions for testing from ebay(ultra cheap) and banggood (a bit pricier), but both came broken.
I was able to recover one via flashing and this one works now. From the outside they look identical and I do not even know, which one was not flashable. 

So they are definitly  not my first picks, but should work.

Besides this issue can get closed now.

closed 2019-10-02 by ascillato2