arendst/Tasmota #6431: driver USE_SDM630_2 i'm tested

dr-apple, 2019-09-16:
Hi here my report for the driver USE_SDM630_2:

before:

[image]

here with the driver USE_SDM630_2:

[image]

I became only 1 value for volt and the other values....

arendst, 2019-09-16:
Ah, bugger. Forgot to tell it's three phase. Hold on.

dr-apple, 2019-09-16:
Thank you very much, it's work fine:

[image]

closed 2019-09-16 by dr-apple