arendst/Tasmota #6436: Plug dead after enabling Hue emulation

kweronline, 2019-09-16:
### ISSUE DESCRIPTION - TROUBLESHOOTING
After successfully flashing my WOOX R4026 plug with the latest sonoff.bin via tuya-convert, I tried enabling the Hue Bridge Emulation and after that the plug was dead. Before enabling that mode, the plug worked fine.
Ive tried resetting it by using the button combinations/long presses without luck.
  

### REQUESTED INFORMATION

- [X ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [ ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X ] Device used (i.e. Sonoff Basic) : TUYA Based Plug
- [X ] Tasmota binary firmware version number used : pre-compiled v. 6.6.0
- [X ] Development IDE - Compiler / Upload tools used : tuya-convert
- [ ] Provide the output of command ``status 0`` : N/A

CONSOLE OUTPUT HERE:
- N/A

Jason2866, 2019-09-17:
In no way offend! But how do you think anybody can help here?
In general HueEmulation does work fine on many many devices.
There is no open issue that enabling the Emulation crashes devices.

You can try to address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

webtop, 2019-09-19:
Well, if the Hue Bridge Emulation was enabled through the management UI, or via my_user_config.h, then this is technically the correct place to post. Remember that the device was working before flashing.

As for how anyone can help: you could start by asking if the device even turns on, followed by if the UI is reachable. If neither of those provide a YES answer, then a connect to the PC would be useful to see if the flash_id or read_mac command can still be ran (probably using esptool.py).

The point of pushing development is to find these things that don't work, and make them work, and then roll that code into the base code.

Jason2866, 2019-09-19:
@webtop  Agree, but there is every needed info missing. 
There is a reason why we have the issue template
So we can give just a general answer

webtop, 2019-09-19:
@Jason2866 Agree. @kweronline Ball's in your court :)

kweronline, 2019-09-19:
Sorry for the lack of information, but thats all the info I could get, as the Plug itself can't be opened (atleast I didn't manage to open it so far).
I had successfully flashed (via tuya-convert (aka OTA)) and used the Plug normally via the Webinterface. Only after enabling the Hue Emulation and restarting the plug was dead. No turning on(the LED on the plug wouldn't turn on), no WiFi-AP, no connection to the 2ndary Wifi network.

Jason2866, 2019-09-19:
Only solution to get it working again is opening and flashing via serial.

closed 2019-09-23 by ascillato2