arendst/Tasmota #6437: VL53L1X (CJMCU-531) Sensor not responding/not data i2c

SirRedZ, 2019-09-17:
<GUIDE>

<This BUG issue template is meant to REPORT Tasmota software BUGS ONLY>
  
<Please DO NOT OPEN AN ISSUE:>
  <If your Tasmota version is not the latest, please update before posting. Your issue might be already solved. Latest precompiled bins of Tasmota can be downloaded from http://thehackbox.org/tasmota/>
  <If your issue is a flashing issue, please address that to the Tasmota Support Chat>
  <If your issue is compilation problem, please address that to the Tasmota Support Chat>
  <If your issue has been addresed before (duplicated issue), please ask in the original issue>
  <If your issue is wifi problem or mqtt problem, please try first the steps provided in troubleshooting of the wiki>

Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

### BUG DESCRIPTION
Trying to add sensor VL53L1X (cjmcu-531) to report data/distances, not recognized and resets Tasmota v6.6.0 immediately upon configuring SCL & SDA on D1 Mini (clone)

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X ] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : D1 Mini (clone)
- [X] Tasmota binary firmware version number used : _6.6.0 self-compiled 
- [X] Development IDE - Atom/PlatformIO - Arduino IDE 0.87 - Gitpod (latest)
- [X ] Provide the output of command ``status 0`` :
```
20:35:32 CMD: status 0
20:35:32 SRC: WebConsole from 192.168.12.211
20:35:32 MQT: stat/sonoff/STATUS = {"Status":{"Module":18,"FriendlyName":["VL53L1X"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
20:35:32 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software Watchdog","Uptime":"0T00:22:52","StartupUTC":"2019-09-17T00:12:40","Sleep":50,"CfgHolder":4617,"BootCount":11,"SaveCount":26,"SaveAddress":"FA000"}}
20:35:32 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0.12(sonoff)","BuildDateTime":"2019-09-16T22:59:36","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
20:35:32 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":5,"WebLog":5,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wbhnet",""],"TelePeriod":45,"Resolution":"558180C0","SetOption":["00008009","280500000100060000005A64000000000000","00000000"]}}
20:35:32 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":560,"Free":440,"Heap":25,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashMode":3,"Features":["00000809","8FDAE397","043683A0","22B616CD","05001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,8,9,10,14,15,17,18,20,22,26,34,45"}}
20:35:32 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-1005","IPAddress":"192.168.12.177","Gateway":"192.168.12.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.12.1","Mac":"CC:50:E3:48:C3:ED","Webserver":2,"WifiConfig":2}}
20:35:32 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.12.12","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_48C3ED","MqttUser":"rksreygm","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
20:35:32 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Tue Sep 17 00:35:32 2019","Local":"Mon Sep 16 20:35:32 2019","StartDST":"Sun Mar 10 02:00:00 2019","EndDST":"Sun Nov 03 02:00:00 2019","Timezone":99,"Sunrise":"06:55","Sunset":"19:24"}}
20:35:32 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2019-09-16T20:35:32"}}
20:35:32 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2019-09-16T20:35:32","Uptime":"0T00:22:52","UptimeSec":1372,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"wbhnet","BSSId":"58:D5:6E:AA:E3:59","Channel":4,"RSSI":94,"LinkCount":1,"Downtime":"0T00:00:12"}}}
20:35:32 MQT: tele/sonoff/STATE = {"Time":"2019-09-16T20:35:32","Uptime":"0T00:22:52","UptimeSec":1372,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":25,"MqttCount":1,"Wifi":{"AP":1,"SSId":"wbhnet","BSSId":"58:D5:6E:AA:E3:59","Channel":4,"RSSI":90,"LinkCount":1,"Downtime":"0T00:00:12"}}

```
- [x] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
Not applicable

```

### TO REPRODUCE
 Flash D1 Mini with 6.6.0 (custom compiled with sensor enabled as no pre-compiled binaries support this sensor by default.
- I also tried to disable SHT sensor as someone suggested it might be conflicting with some I2C sensors. Same result.

### EXPECTED BEHAVIOUR
I2C sensor found, addressed and reporting data / distance.

### SCREENSHOTS
[image]
Note: Screenshot of 6.6.0 could not be taken as Watchdog reboots occur resetting config after both I2C pins are defined. Either pin can be defined and the software does not reset, including reverse configuration.

### ADDITIONAL CONTEXT

Multiple D1 Mini devices tested, along with custom Arduino sketch to validate sensor is functional as expected. Multiple VL53L1X sensors used also.

https://github.com/SirRedZ/Esp32_8266-VL53L1X-demo

Also tested on ESP32 (without Tasmota)

Sparkfun VL53L1X library: https://github.com/sparkfun/SparkFun_VL53L1X_Arduino_Library/

This library was used for verification/functionality testing.

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-09-17:
Additional context...

The device (with 6.6.0.12, not 6.6.0) runs until the I<sup>2</sup>C GPIO are configured. As soon as the settings are saved, the device bootloops and resets settings.

I was hoping to be able to trap the Exception in INFO3 along with the map file in order to provide some insight on where the exception is occuring. But the restart reason here is just a normal restart.

@SirRedZ - do you think you can set the I<sup>2</sup>C GPIO and capture INFO3 as soon as it boots where you can grab the first boot after the bootloop settles down?

I asked him to include @gemu2015 since he's the driver's author.

Mike

gemu2015, 2019-09-17:
i use the USE_VL53L0X sensor without problems also in new tasmota versions.
the VL53L1X however is completely different and currently not supported!
there is an open pull request for this sensor, you may ask the author about any progress

Jason2866, 2019-09-18:
Please close, not a issue. The sensor is not supported.

closed 2019-09-18 by SirRedZ