arendst/Tasmota #6442: SGP30 aHumidity weird characters prevent proper JSON

jaburges, 2019-09-18:
### BUG DESCRIPTION
SGP30 connected to Sonoff Basic with AM2301
https://github.com/arendst/Sonoff-Tasmota/issues/5841 states that Temp and humidity is required to calculate absolute humidity. After a reflash global_update did in fact pick up these values thus calculating aHumidity - however it still had the odd characters causing 2 issues:
```
tele/nursery/SENSOR = {"Time":"2019-09-18T01:29:28","AM2301":{"Temperature":75.4,"Humidity":49.4},"SGP30":{"eCO2":454,"TVOC":90,"aHumidity": Z�?�Y�?
```

1. the weird characters at the end of the sensor value.
2. The odd characters seem to prevent Tasmota from wrapping the Sensor value in proper JSON (and thus not sending MQTT messages)

Using code from https://paste.ubuntu.com/p/bSPDDbGYBF/ thanks to aheid from HA discord, we could remove the aHumidity value. 

This resulted in the following:
```
tele/nursery/SENSOR = {"Time":"2019-09-18T03:20:36","AM2301":{"Temperature":78.6,"Humidity":46.3},"SGP30":{"eCO2":474,"TVOC":11},"TempUnit":"F"}
```
so the JSON is properly formed and then MQTT could then being sending messages.

The UI on the main webpage still displayed an incorrect character:

```
SGP30 Abs Humidity | �78.6 g/m3
```

### REQUESTED INFORMATION
_Make sure these boxes are checked before submitting your issue. Thank you_

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [X] Searched the problem in the forum (https://groups.google.com/d/forum/sonoffusers)
- [X] Searched the problem in the chat (https://discord.gg/Ks2Kzd4)
- [X] Device used (i.e. Sonoff Basic) : sonoff basic
- [X] Tasmota binary firmware version number used :6.6.0  tested both pre-compiled & self-compiled
- [X] Development IDE - Compiler / Upload tools used : Arduino IDE & Arduino
- [X] Provide the output of command ``status 0`` : 
```
STATUS 0 OUTPUT HERE:

```
- [ ] Provide the output of console when you experience your issue if apply :
_(Please use_ ``weblog 4`` _for more debug information)_
```
CONSOLE OUTPUT HERE:
04:52:28 RSL: stat/nursery/STATUS = {"Status":{"Module":1,"FriendlyName":["Studio-L-B-Lgt"],"Topic":"nursery","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
04:52:28 RSL: stat/nursery/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","RestartReason":"Power on","Uptime":"0T00:01:36","StartupUTC":"2019-09-18T03:50:52","Sleep":50,"CfgHolder":4617,"BootCount":2075,"SaveCount":4160,"SaveAddress":"FB000"}}
04:52:28 RSL: stat/nursery/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-09-17T20:39:56","Boot":4,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
04:52:28 RSL: stat/nursery/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Burgess","Burgess"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00000009","280500000100060000000000000000000000","00000001"]}}
04:52:28 RSL: stat/nursery/STATUS4 = {"StatusMEM":{"ProgramSize":546,"Free":392,"Heap":13,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0FDAE396","001783A0","23B617CD","01003BC0"]}}
04:52:28 RSL: stat/nursery/STATUS5 = {"StatusNET":{"Hostname":"Nursery sonoff","IPAddress":"192.168.1.229","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"X.X.X.X","Mac":"5C:CF:7F:3B:7E:C5","Webserver":2,"WifiConfig":3}}
04:52:28 RSL: stat/nursery/STATUS6 = {"StatusMQT":{"MqttHost":"X.X.X.X","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_3B7EC5","MqttUser":"User","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
04:52:28 RSL: stat/nursery/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 18 03:52:28 2019","Local":"Wed Sep 18 04:52:28 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:30","Sunset":"18:58"}}
04:52:28 RSL: stat/nursery/STATUS10 = {"StatusSNS":{"Time":"2019-09-18T04:52:28","AM2301":{"Temperature":26.3,"Humidity":45.4},"SGP30":{"eCO2":431,"TVOC":8,"aHumidity": W�?�V�?
04:52:28 RSL: stat/nursery/STATUS11 = {"StatusSTS":{"Time":"2019-09-18T04:52:28","Uptime":"0T00:01:36","Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER":"ON","Wifi":{"AP":1,"SSId":"SSID","BSSId":"60:5F:8D:11:F0:46","Channel":1,"RSSI":88,"LinkCount":1,"Downtime":"0T00:00:07"}}}

```

### TO REPRODUCE
SGP30 & AM2301
VCC,  TX,   RX,  GND, GPIO14
VCC, SDA, SCL, GND, AM2301

### EXPECTED BEHAVIOUR
JSON should be formatted correctly 
tested on fresh flash from releases (ie no edits in sketch other than enable SGP30

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

jaburges, 2019-09-18:
Issue may be a little deeper as well as MQTT stops sending messages even when the JSON is formatted.
with just the AM2301 enabled i'm getting MQTT messages. 
adding the SGP30 shows the following in the console, but does not push to MQTT
```
tele/nursery/SENSOR = {"Time":"2019-09-18T05:05:31","AM2301":{"Temperature":78.6,"Humidity":45.8},"SGP30":{"eCO2":526,"TVOC":28},"TempUnit":"F"}
```

Jason2866, 2019-09-18:
Have you tried latest development version? There is a change in log buffer size
https://github.com/arendst/Sonoff-Tasmota/pull/6426

aheid, 2019-09-18:
The issue with the code that generates the JSON is that the condition used to fill the buffer containing the abs temp string is different from the condition used to check if the string should be included in the JSON. I examine the code further than that.

closed 2019-09-18 by arendst

jaburges, 2019-09-18:
wow y'all move fast! thanks

PR #6451 (2019-09-18): fix sgp30
## Description:

**Related issue (if applicable):** fixes #6442 >
fixes abshum json export

## Checklist:
  - [x] The pull request is done against the latest dev branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR.
  - [x] The code change is tested and works on core 2.3.0, 2.4.2, 2.5.2, and pre-2.6
  - [ ] The code change pass travis tests. **Your PR cannot be merged unless tests pass**
  - [x] I accept the [CLA](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md#contributor-license-agreement-cla).