arendst/Tasmota #6443: sonoff basic with PIR/Radar motion sensor MQTT /power state issue

blueboi69, 2019-09-18:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Hi!
I need some help or fix this issue.
Tasmota version: 6.6.0
I have a Sonoff basic module, I connected a RCWL-0516 radar motion sensor to GPIO14 as swtch2 (10).
Switchmode is 1
Switchtopic 0
I have a rule, if a motion is triggered it sends an MQTT message RAIN ON. This works fine.
I'd like to control the sonoff relay from Node-Red via MQTT. This also works fine.
BUT when I send an MQTT command, and the relay state is changed either ON or OFF, it also triggers my rules and also triggers the motion sensor! It doesn't matter if I set a rule or I use a custom switchtopic. Every time I control the relay through MQTT an ON message is triggered. After exactly 3seconds an OFF message is followed. The RCWL-0516 radar motion sensor changes it's state in 2seconds.
I also tried to setup a virtual relay, az relay2 so that the switch2 would somehow trigger a virtual relay as a workaround, but it doesn't matter.

And this is real trouble because if a motion is triggered via the relay, in my HA it is triggered as a new motion and it never tries to turn off the relay.
Ofcourse using the motion sensor connected directly to control the relay is not an option.
Thank you for any help!
  

### REQUESTED INFORMATION

 Rules output here:
  
{"Rule1":"ON","Once":"OFF","StopOnError":"OFF","Free":397,"Rules":"on switch2#state=1 do publish stat/sonoff2/RAIN ON endon on switch2#state=0 do publish stat/sonoff2/RAIN OFF endon"}

  ```
- [ ] Provide the output of command ``Status 0``:
  ```
  STATUS 0 output here:
  

00:12:19 SRC: Rule
00:12:19 RSL: Group 0, Index 1, Command PUBLISH, Data stat/sonoff2/RAIN ON
00:12:19 MQT: stat/sonoff2/RAIN = ON
00:12:20 CMD: status 0
00:12:20 SRC: WebConsole from 192.168.105.1
00:12:20 RSL: Group 0, Index 1, Command STATUS, Data 0
00:12:20 MQT: stat/sonoff/STATUS = {"Status":{"Module":1,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"","SwitchMode":[1,1,1,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
00:12:20 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:12:20","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":9,"SaveCount":138,"SaveAddress":"F6000"}}
00:12:20 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.6.0(sonoff)","BuildDateTime":"2019-07-14T14:21:27","Boot":31,"Core":"2_5_2","SDK":"2.2.2-dev(c0eb301)"}}
00:12:20 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["vatreshno",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","280500000100060000000000000000000000","00000000"]}}
00:12:20 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":555,"Free":448,"Heap":24,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"1440E0","FlashMode":3,"Features":["00000809","0FDAE396","041783A0","22B617CD","01003BC0"]}}
00:12:20 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-0016","IPAddress":"192.168.105.110","Gateway":"192.168.105.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.105.1","Mac":"5C:CF:7F:96:40:10","Webserver":2,"WifiConfig":4}}
00:12:20 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.105.1","MqttPort":4510,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_964010","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
00:12:20 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Thu Jan 01 00:12:20 1970","Local":"Thu Jan 01 00:12:20 1970","StartDST":"Thu Jan 01 00:00:00 1970","EndDST":"Thu Jan 01 00:00:00 1970","Timezone":"+00:00","Sunrise":"07:43","Sunset":"16:03"}}
00:12:20 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:12:20","Switch2":"ON"}}
00:12:20 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:12:20","Uptime":"0T00:12:20","Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"vatreshno","BSSId":"00:0C:42:6C:4D:ED","Channel":4,"RSSI":80,"LinkCount":1,"Downtime":"0T00:00:05"}}}

  
  ```
- [ ] Provide the output of console when you experience your issue if applicable:
  _(Please use_ ``weblog 4`` _for more debug information)_
  ```
  Console output here:
  
  
----Normal movement----
00:24:22 MQT: stat/sonoff2/RAIN = ON
00:24:25 RUL: SWITCH2#STATE=0 performs "publish stat/sonoff2/RAIN OFF"
00:24:25 SRC: Rule
00:24:25 RSL: Group 0, Index 1, Command PUBLISH, Data stat/sonoff2/RAIN OFF
00:24:25 MQT: stat/sonoff2/RAIN = OFF
----------------------------

------- when relay is triggered via MQTT message AND I'm sure there is no movement---------

00:25:15 RUL: SWITCH2#STATE=0 performs "publish stat/sonoff2/RAIN OFF"
00:25:15 SRC: Rule
00:25:15 RSL: Group 0, Index 1, Command PUBLISH, Data stat/sonoff2/RAIN OFF
00:25:15 MQT: stat/sonoff2/RAIN = OFF
00:25:33 SRC: MQTT
00:25:33 RSL: Group 0, Index 1, Command POWER, Data ON
00:25:33 MQT: stat/sonoff/RESULT = {"POWER1":"ON"}
00:25:33 MQT: stat/sonoff/POWER1 = ON
00:25:33 RUL: SWITCH2#STATE=1 performs "publish stat/sonoff2/RAIN ON"
00:25:33 SRC: Rule
00:25:33 RSL: Group 0, Index 1, Command PUBLISH, Data stat/sonoff2/RAIN ON
00:25:33 MQT: stat/sonoff2/RAIN = ON
00:25:33 CFG: Saved to flash at F5, Count 139, Bytes 3584
00:25:36 RUL: SWITCH2#STATE=0 performs "publish stat/sonoff2/RAIN OFF"
00:25:36 SRC: Rule
00:25:36 RSL: Group 0, Index 1, Command PUBLISH, Data stat/sonoff2/RAIN OFF
00:25:36 MQT: stat/sonoff2/RAIN = OFF
--------------------------

**(Please, remember to close the issue when the problem has been addressed)**

meingraham, 2019-09-18:
> - If your Tasmota version is not the latest from the **development branch**, please update your device before submitting your issue. Your problem might already be solved. The latest precompiled binaries of Tasmota can be downloaded from http://thehackbox.org/tasmota/>

blueboi69, 2019-09-18:
Hi! I'm using the latest version of tasmota.
The information page says: 

Program Version	6.6.0(sonoff)
Build Date & Time	2019-07-14T14:21:27
Core/SDK Version	2_5_2/2.2.2-dev(c0eb301)

meingraham, 2019-09-18:
> **latest from the development branch**  
6.6.0.12

> The **_latest_** precompiled binaries

> can be downloaded from http://thehackbox.org/tasmota/

blueboi69, 2019-09-18:
Ok, you are right.
Here we go:

Program Version	6.6.0.12(f8ea106-sonoff)
Build Date & Time	2019-09-18T09:03:15
Core/SDK Version	2_4_2/2.2.1(cfd48f3)
Uptime	0T00:01:07

But after flashing I noticed that if the sonoff basic is powered via usb the issue is not present. This way the relay itself is not messing. After connecting the sonoff to mains and the relay starts to work, the issue comes up again.

meingraham, 2019-09-18:
OK, so you have a questionable voltage regulator. A common problem with these I'm afraid.

Jason2866, 2019-09-18:
If i would make a bet. I would go for device is a Sonoff Basic R2

blueboi69, 2019-09-18:
Oh dear! How can I regonise it is an R2? I use a dc buck converter to make 5V from 3.3V.
Do you think that when the relay is pulled, the 3.3V drops so much that it triggers the radar sensor?

Jason2866, 2019-09-18:
Yeah, the power supply is so weak that some sonoff basic cant run Tasmota because the ESP
needs more power than the original firmware.
Just google for Sonoff Basic r2 and you will find pictures

blueboi69, 2019-09-18:
Allright, thanks! Sorry for opening an issue about that.

closed 2019-09-18 by blueboi69