arendst/Tasmota #6444: Wrong data using P1 smart meter implementation in  Sonoff Pow R2

pablozg, 2019-09-18:
Hi, after few days using the new P1 smart meter for domoticz I observed some wrong data using it with a sonoff Pow R2:

This screenshot is from the new P1 smart meter, as you can see along the day is using both usage (usage1 and usage2): 

[image]

This screenshot is from my custom LUA script in domoticz in the same space time, using the formula I used in my PR.

[image]

But the new sensor is working well with the sdm120 as you can see in this screen shot, so only wrong data is sended with the POW R2.

[image]

Here the status 0 of POW R2:

```
11:25:12 CMD: status 0
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS = {"Status":{"Module":43,"FriendlyName":["monitorCasa"],"Topic":"Casa/PB/monitorCasa","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS1 = {"StatusPRM":{"Baudrate":4800,"GroupTopic":"sonoffs","OtaUrl":"http://10.10.0.110:1880/esp8266-ota/update/firmware_6.6.0.12_4mbit","RestartReason":"Software/System restart","Uptime":"0T00:42:36","StartupUTC":"2019-09-18T08:42:36","Sleep":50,"CfgHolder":4617,"BootCount":140,"SaveCount":745,"SaveAddress":"3F8000"}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS2 = {"StatusFWR":{"Version":"6.6.0.12(sonoff)","BuildDateTime":"2019-09-18T10:38:21","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Isabella_T","Isabella_H"],"TelePeriod":20,"Resolution":"558B88C0","SetOption":["002A8009","28B400000100068000005AFF000000170D00","00000200"]}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS4 = {"StatusMEM":{"ProgramSize":563,"Free":2508,"Heap":26,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640EF","FlashMode":3,"Features":["00000809","8FDAE397","043683A1","22B617CD","41001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34,49"}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS5 = {"StatusNET":{"Hostname":"Casa/PB/monitorCasa-6274","IPAddress":"10.10.0.101","Gateway":"10.10.0.254","Subnetmask":"255.255.255.0","DNSServer":"10.10.0.254","Mac":"80:7D:3A:32:B8:82","Webserver":2,"WifiConfig":2}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS6 = {"StatusMQT":{"MqttHost":"10.10.0.110","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32B882","MqttUser":"htpc","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS7 = {"StatusTIM":{"UTC":"Wed Sep 18 09:25:12 2019","Local":"Wed Sep 18 11:25:12 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":99,"Sunrise":"07:30","Sunset":"19:58"}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS9 = {"StatusPTH":{"PowerDelta":80,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS10 = {"StatusSNS":{"Time":"2019-09-18T11:25:12","ENERGY":{"TotalStartTime":"2019-05-25T00:00:00","Total":[11024.791,146.777,10878.015],"Yesterday":11.305,"Today":4.137,"Power":1317.1,"ApparentPower":1360.6,"ReactivePower":341.2,"Factor":0.97,"Voltage":234.5,"Current":5.802}}}
11:25:12 MQT: Casa/PB/monitorCasa/stat/STATUS11 = {"StatusSTS":{"Time":"2019-09-18T11:25:12","Uptime":"0T00:42:36","UptimeSec":2556,"Heap":26,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Isabella_T","BSSId":"D4:6E:0E:D0:43:EC","Channel":6,"RSSI":38,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
And the Tariff config:

````
11:27:09 CMD: tariff
11:27:09 MQT: Casa/PB/monitorCasa/stat/RESULT = {"Tariff":{"Off-Peak":[22,23],"Standard":[12,13],"Weekend":"OFF"}}
````

gajotnt, 2019-09-18:
I'm also having some readings being way off. 
Using the my SDM120 for both the domoticz virtual meter and the P1 smart (updated the sonoff to try out the new feature), for example the normal usage meter is giving me a total Usage of 9 kwh and the P1 is giving me 5.4kwh, almost half. Letting it run a couple more days.

pablozg, 2019-09-19:
It seems only are affected the off-peak period, during the standard period all data is ok:

[image]

pablozg, 2019-09-20:
I have created a PR #6462, with the new code no wrong data is reported to domoticz.

[image]

@gajotnt can you compile tasmota with the changes made in my PR and test it?

Thanks.

gajotnt, 2019-09-20:
Will try this weekend :)

pablozg, 2019-09-22:
Hi, for me is working well now, I'll wait to @gajotnt test result, before close the issue.

gajotnt, 2019-09-22:
Having some issues with Arduino ide. Gotta reinstall it again

pablozg, 2019-09-22:
@gajotnt You can download the lastest compiled binaries from http://thehackbox.org/tasmota/, they have the new changes merged and you only have to flash it.

gajotnt, 2019-09-22:
On it, gonna upload the bin and maybe later today or tomorrow will report

gajotnt, 2019-09-22:
Like this right? So far still seeing it register 0watts, but better wait some more hours. Also im not on the lastest beta of domoticz, dont know if it matters.
[image]

pablozg, 2019-09-22:
Please, create a new P1 sensor in domoticz, and configure the new index number in tasmota and see if it update properly.

gajotnt, 2019-09-22:
Ok, deleted it, made a new P1SmartMeter sensor and updated sonoff with new IDX.

gajotnt, 2019-09-22:
When return is above 0watts shouldnt the usage be 0watts?

[image]

Otherwise so far not really that off from the normal usage sensor

[image]

pablozg, 2019-09-22:
> When return is above 0watts shouldnt the usage be 0watts?

You are right, I have the same issue, but I think is related to the data update speed in domoticz and the formula they use to process the data, I don't know if they use the last reported data in the five minutes or they use a arithmetic average of all data received in the lapse of five minutes and then represent it in the chart.

> Otherwise so far not really that off from the normal usage sensor

I have the same behaviour, but if you see the totals in the two sensors, they are the same.

[image]

[image]

arendst, 2019-09-22:
Latest fix solves domoticz decimal resolution which would always be zero for a smart meter.

Also fixed partial entries for EnergyReset4 and EnergyRestet5.

Be awar that before implemnting this make a copy of your current usage values; after installation new code you will have to execute EnergeReset4 (and possibly EnergyReset5) to restore the correct values as now they are divided by 100 ;-).

gajotnt, 2019-09-22:
If i do this will the domoticz normal usage sensor still work like always?

arendst, 2019-09-22:
Yep

gajotnt, 2019-09-22:
Ok nice :)

gajotnt, 2019-09-22:
Now domoticz doesnt even show return 1 and 2. maybe because no return has been detected.

gajotnt, 2019-09-22:
Still getting some off place minor return values. 
[image]

pablozg, 2019-09-23:
It's strange, as you can see my return chart is totally flat, maybe something in your installation is injecting energy to the grid or a faulty meter.

[image]

gajotnt, 2019-09-23:
I have solar panels, but at night there is no sun, thats why I found it wierd, and if the house is consuming 638watts there can't be return. Maybe my sdm120 is faulty, still 11watts is nothing.

pablozg, 2019-09-23:
I have solar panels too, but as you can see, my sdm120 don't register any activity at night.

gajotnt, 2019-09-23:
If its working well in your side of the woods, then its something in my install.
12watts readings now and then wont change the readings in the long run, we're talking about 0.001Kwh, thats nothing :)

pablozg, 2019-09-23:
Ok, So I think we can close this issue and mark it as fixed

closed 2019-09-23 by pablozg

gajotnt, 2019-09-23:
Again some minor errors during the day, with the return, this time the usage should be zero.
Can be sinse domoticz logs in 5min intervals, in those 5 min the SD120 registred return and usage. Can this be the case?
[image]

pablozg, 2019-09-24:
I think so, because the counters used by tasmota to send the data to domoticz are acumulative.

So the both counters (usage/export) can be increased in the lapse of five minutes.

gajotnt, 2019-09-25:
Those spikes right after it going to 0watts are wierd. Cant find a suitable explantion for it.
PS: if the house goes over +/-3500watts the fuses go off.

[image]