arendst/Tasmota #6446: Automatically switching the supply voltage of a WS2812 LED strip.

Vujagig, 2019-09-18:
> Please take a few minutes to complete the requested information below. Our ability to provide assistance is greatly hampered without it. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

**Have you looked for this feature in other issues and in the wiki?**  
Yes

**Is your feature request related to a problem? Please describe.**  
_A clear and concise description of what the problem is._
No

**Describe the solution you'd like**  
_A clear and concise description of what you want to happen._
If the LEDs are "soft-off", an LED strip can quite well consume a few watts of standby power.
It would be good if you could use a SetOption command for the fact that a relay output with a short delay is switched.
WS2812 ON: Power on -> short wait -> WS2812 command on
WS2812 OFF: WS2812 command off -> short wait -> LED power off
SetOption xx 1..16 for GPIO no. the 0..8 for D-No.
SetOption yy 1-250 Delay time in ms
You only need a single command to completely switch off the LED

**Describe alternatives you've considered**  
_A clear and concise description of any alternative solutions or features you've considered._

**Additional context**  
_Add any other context or screenshots about the feature request here._
None

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2019-09-18:
This is doable via a rule.
Please, address this to the [Tasmota Support Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are meant for Tasmota Software Bug Reporting.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

Vujagig, 2019-09-18:
Oh thanks, it works :)

closed 2019-09-18 by Vujagig

raddi, 2020-01-06:
Did you manage do accomplish this? Could you post your rule? Thank you! :)