arendst/Tasmota #6447: HA mqtt auto discovery fails partially - HASS: failed to parse 'DS18B20-1'

qschneider, 2019-09-18:
### ISSUE DESCRIPTION - TROUBLESHOOTING
Home Assistant mqtt auto discovery fails partially on multiple temperature sensors (DS18B20) on a nodemcu V3.
I assume that xdrv_12_home_assistant.ino does not provide that scenario yet...

manually it should look  something like that, but i'am not firm enough with HA mqtt autodiscovery, so its just an assumption... 

homeassistant/sensor/63D985_TEMPERATURE1/config
{"device_class": "temperature", "name": "HeizungTemperatur Raum", "state_topic": "DVES_63D985/tele/SENSOR", "unit_of_measurement": "°C", "value_template": "{{ value_json['DS18B20-1'].Temperature }}"}
homeassistant/sensor/63D985_TEMPERATURE2/config
{"device_class": "temperature", "name": "HeizungTemperatur Boiler Vorlauf", "state_topic": "DVES_63D985/tele/SENSOR", "unit_of_measurement": "°C", "value_template": "{{ value_json['DS18B20-2'].Temperature }}"}
...

This existing results on that HA device (with missing temperature entities) looks like this.

[image]

Console output here:
13:24:33 CMD: setoption19 1
13:24:33 SRC: WebConsole from 192.168.44.1
13:24:33 MQT: DVES_63D985/stat/RESULT = {"SetOption19":"ON"}
13:24:34 MQT: homeassistant/light/63D985_LI_1/config =  (retained)
13:24:34 MQT: homeassistant/switch/63D985_RL_1/config =  (retained)
13:24:34 MQT: homeassistant/light/63D985_LI_2/config =  (retained)
13:24:34 MQT: homeassistant/switch/63D985_RL_2/config =  (retained)
13:24:34 MQT: homeassistant/light/63D985_LI_3/config =  (retained)
13:24:34 MQT: homeassistant/switch/63D985_RL_3/config =  (retained)
13:24:34 MQT: homeassistant/light/63D985_LI_4/config =  (retained)
13:24:34 MQT: homeassistant/switch/63D985_RL_4/config =  (retained)
13:24:34 MQT: homeassistant/light/63D985_LI_5/config =  (retained)
13:24:34 MQT: homeassistant/switch/63D985_RL_5/config =  (retained)
13:24:34 MQT: homeassistant/light/63D985_LI_6/config =  (retained)
13:24:34 MQT: homeassistant/switch/63D985_RL_6/config =  (retained)
13:24:34 MQT: homeassistant/light/63D985_LI_7/config =  (retained)
13:24:34 MQT: homeassistant/switch/63D985_RL_7/config =  (retained)
13:24:34 MQT: homeassistant/light/63D985_LI_8/config =  (retained)
13:24:34 MQT: homeassistant/switch/63D985_RL_8/config =  (retained)
13:24:34 HASS: failed to parse 'DS18B20-1'
13:24:34 MQT: homeassistant/sensor/63D985_status/config = {"name":"HeizungTemperatur status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"63D985_status","device":{"identifiers":["63D985"],"connections":[["mac","CC:50:E3:63:D9:85"]],"name":"HeizungTemperatur","model":"NodeMCU V3","sw_version":"6.6.0.12(33baadf-sonoff)","manufacturer":"Tasmota"},"~":"DVES_63D985/tele/"} (retained)
13:24:34 CFG: Saved to flash at F4, Count 103, Bytes 4096
13:24:35 MQT: DVES_63D985/tele/HASS_STATE = {"Version":"6.6.0.12(33baadf-sonoff)","BuildDateTime":"2019-09-10T22:04:34","Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Module":"NodeMCU V3","RestartReason":"Software/System restart","Uptime":"5T13:36:31","WiFi LinkCount":2,"WiFi Downtime":"0T00:00:07","MqttCount":4,"BootCount":58,"SaveCount":103,"IPAddress":"192.168.44.54","RSSI":"56","LoadAvg":27}
13:24:48 MQT: DVES_63D985/tele/STATE = {"Time":"2019-09-18T13:24:48","Uptime":"5T13:36:44","UptimeSec":481004,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":4,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":58,"LinkCount":2,"Downtime":"0T00:00:07"}}
13:24:48 MQT: DVES_63D985/tele/SENSOR = {"Time":"2019-09-18T13:24:48","DS18B20-1":{"Id":"000006DDD0F5","Temperature":16.6},"DS18B20-2":{"Id":"041470E982FF","Temperature":30.1},"DS18B20-3":{"Id":"041470E9DCFF","Temperature":28.6},"DS18B20-4":{"Id":"041470EA81FF","Temperature":27.8},"DS18B20-5":{"Id":"0414710F8BFF","Temperature":38.8},"DS18B20-6":{"Id":"041471600DFF","Temperature":25.2},"TempUnit":"C"}

effelle, 2019-09-18:
Assuming the result of `DVES_63D985/tele/SENSOR` is complete and there is not other stuff involved, this is not a problem with buffer size, since the limit is 500 bytes and your Sensor topic is 363 bytes long. 
Please remove all but the first DS18B20 from the configuration and then run again `Setoption19 1`. 
If is working fine, add DS18B20-2 and test again. 
Please be sure to set your weblog level to 4 using `weblog 4`

effelle, 2019-09-18:
Also, paste here the result of `192.168.44.54/cm?cmnd=status%200` from your web browser when all the sensor are configured, thanks.

qschneider, 2019-09-18:
Thanks for your effort so far.

I took a new nodemcu and sensors, because the other one was soldered together.

After added just one sensor first and run SetOption19 1.
This was the result as expected in HA.

[image]

After adding the second sensor, it looked like this.

[image]

the result of http://192.168.44.108/cm?cmnd=status%200 was

{"Status":{"Module":0,"FriendlyName":["Sonoff"],"Topic":"DVES_ED76DA","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:01:57","StartupUTC":"2019-09-18T18:50:23","Sleep":50,"CfgHolder":4617,"BootCount":11,"SaveCount":23,"SaveAddress":"F6000"},"StatusFWR":{"Version":"6.6.0.12(33baadf-sonoff)","BuildDateTime":"2019-09-10T22:04:34","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"},"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["FBI",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000A8009","280500000100060000005AFF000000000000","00000200"]},"StatusMEM":{"ProgramSize":558,"Free":444,"Heap":21,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164068","FlashMode":2,"Features":["00000809","8FDAE397","003683A0","22B617CD","01001BC0","00000001"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24","Sensors":"1,2,3,4,5,6,7,8,9,10,14,15,17,18,20,22,26,34"},"StatusNET":{"Hostname":"DVES_ED76DA-5850","IPAddress":"192.168.44.108","Gateway":"192.168.44.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.44.11","Mac":"84:F3:EB:ED:76:DA","Webserver":2,"WifiConfig":4},"StatusMQT":{"MqttHost":"192.168.44.11","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_ED76DA","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":30},"StatusTIM":{"UTC":"Wed Sep 18 18:52:20 2019","Local":"Wed Sep 18 19:52:20 2019","StartDST":"Sun Mar 31 02:00:00 2019","EndDST":"Sun Oct 27 03:00:00 2019","Timezone":"+01:00","Sunrise":"06:30","Sunset":"18:58"},"StatusSNS":{"Time":"2019-09-18T19:52:20","DS18B20-1":{"Id":"000006DDD0F5","Temperature":22.6},"DS18B20-2":{"Id":"01144FEE20AA","Temperature":23.4},"TempUnit":"C"},"StatusSTS":{"Time":"2019-09-18T19:52:20","Uptime":"0T00:01:57","UptimeSec":117,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:05"}}}

After that I run setoption19 0 and changed mqtt topic from %topic%/%prefix%/ back to %prefix%/%topic%/ to see if the HA object disappeared, but it stayed the same, except that the entities had an unknown state (did to take a screenshot of that).

After running setoption19 1 again I now see this in HA.
[image]
[image]

qschneider, 2019-09-18:
To me it seems that the auto discovery works if you using just 1 sensor.
If you start off with multiple sensors attached, you will end up with no sensors displayed in HA, like in my first post.
If you start adding more sensors to an device or keep on changing the discovery process it starts getting worse :-)

this leads me to the question on how to get rid off unnecessary entities in HA?

effelle, 2019-09-18:
So adding a third sensor (DS18B20-3) in your case, already lead to a parse error (`HASS: failed to parse 'DS18B20-3'` on log)?
Could you gently run again `Setoption 19 1` when you have two sensor and paste here the result of all the lines starting with `homeassistant/sensor/`?

> this leads me to the question on how to get rid off unnecessary entities in HA?

You can't easily. Is a known issue of Home Assistant auto discovery. The fastest way is to cancel completely all the MQTT autodiscovery you got until now (trash bin on the upper right of the page). Doing that you will loose all he discovery you got so far since it will wipe the DB of MQTT discovery inside HA.

qschneider, 2019-09-19:
continued testing.
up to 4 sensors are ok.

weblog after restart

11:02:28 APP: Boot Count 36
11:02:28 CFG: Saved to flash at FA, Count 59, Bytes 4096
11:02:33 MQT: DVES_ED76DA/tele/STATE = {"Time":"2019-09-19T11:02:33","Uptime":"0T00:00:15","UptimeSec":15,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":62,"LinkCount":1,"Downtime":"0T00:00:05"}}
11:02:33 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-19T11:02:33","DS18B20-1":{"Id":"000006DDD0F5","Temperature":22.1},"DS18B20-2":{"Id":"01144FC4C9AA","Temperature":20.6},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":20.4},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":20.4},"TempUnit":"C"}
11:02:26 MQT: homeassistant/light/ED76DA_LI_8/config =  (retained)
11:02:26 MQT: homeassistant/switch/ED76DA_RL_8/config =  (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_DS18B20-1_Id/config = {"name":"nodeMCU Test DS18B20-1 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-1_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":" ","val_tpl":"{{value_json['DS18B20-1'].Id}}"} (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_DS18B20-1_Temperature/config = {"name":"nodeMCU Test DS18B20-1 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-1_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-1'].Temperature}}"} (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_DS18B20-2_Id/config = {"name":"nodeMCU Test DS18B20-2 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-2_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":" ","val_tpl":"{{value_json['DS18B20-2'].Id}}"} (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_DS18B20-2_Temperature/config = {"name":"nodeMCU Test DS18B20-2 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-2_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-2'].Temperature}}"} (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_DS18B20-3_Id/config = {"name":"nodeMCU Test DS18B20-3 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-3_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":" ","val_tpl":"{{value_json['DS18B20-3'].Id}}"} (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_DS18B20-3_Temperature/config = {"name":"nodeMCU Test DS18B20-3 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-3_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-3'].Temperature}}"} (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_DS18B20-4_Id/config = {"name":"nodeMCU Test DS18B20-4 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-4_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":" ","val_tpl":"{{value_json['DS18B20-4'].Id}}"} (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_DS18B20-4_Temperature/config = {"name":"nodeMCU Test DS18B20-4 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-4_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-4'].Temperature}}"} (retained)
11:02:26 MQT: homeassistant/sensor/ED76DA_status/config = {"name":"nodeMCU Test status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"ED76DA_status","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]],"name":"nodeMCU Test","model":"NodeMCU V3","sw_version":"6.6.0.12(33baadf-sonoff)","manufacturer":"Tasmota"},"~":"DVES_ED76DA/tele/"} (retained)

result of http://192.168.44.108/cm?cmnd=status%200
{"Status":{"Module":0,"FriendlyName":["nodeMCU Test"],"Topic":"DVES_ED76DA","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0},"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://thehackbox.org/tasmota/release/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:47","StartupUTC":"2019-09-19T10:02:18","Sleep":50,"CfgHolder":4617,"BootCount":36,"SaveCount":59,"SaveAddress":"FA000"},"StatusFWR":{"Version":"6.6.0.12(33baadf-sonoff)","BuildDateTime":"2019-09-10T22:04:34","Boot":31,"Core":"STAGE","SDK":"2.2.2-dev(38a443e)"},"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost" [... cut ...] 05","DS18B20-1":{"Id":"000006DDD0F5","Temperature":22.5},"DS18B20-2":{"Id":"01144FC4C9AA","Temperature":20.6},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":20.4},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":20.4},"TempUnit":"C"},"StatusSTS":{"Time":"2019-09-19T11:04:05","Uptime":"0T00:01:47","UptimeSec":107,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":48,"LinkCount":1,"Downtime":"0T00:00:05"}}}

qschneider, 2019-09-19:
with 5 sensors or more I'am getting the parse error that looks like the one I had in my first post
11:06:47 HASS: failed to parse 'DS18B20-1'

weblog after restart

00:00:00 CFG: Loaded from flash at FA, Count 59
00:00:00 SRC: Restart
00:00:00 Project sonoff nodeMCU Test Version 6.6.0.12(33baadf-sonoff)-STAGE
00:00:00 DSB: Sensors found 5
00:00:00 SHT: Sensor did not ACK command
00:00:00 WIF: Checking connection...
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Connecting to AP1 FBI in mode 11N as DVES_ED76DA-5850...
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Checking connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Checking connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Checking connection...
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on DVES_ED76DA-5850 with IP address 192.168.44.108
11:06:45 NTP: Drift 0, (UTC) Thu Sep 19 10:06:45 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
11:06:45 HTP: Main Menu
11:06:46 MQT: Attempting connection...
11:06:46 MQT: Connected
11:06:46 MQT: DVES_ED76DA/tele/LWT = Online (retained)
11:06:46 MQT: DVES_ED76DA/cmnd/POWER = 
11:06:46 MQT: Subscribe to DVES_ED76DA/cmnd/#
11:06:46 MQT: Subscribe to sonoffs/cmnd/#
11:06:46 MQT: Subscribe to cmnd/DVES_ED76DA_fb/#
11:06:46 MQT: DVES_ED76DA/tele/INFO1 = {"Module":"NodeMCU V3","Version":"6.6.0.12(33baadf-sonoff)","FallbackTopic":"cmnd/DVES_ED76DA_fb/","GroupTopic":"sonoffs"}
11:06:46 MQT: DVES_ED76DA/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"DVES_ED76DA-5850","IPAddress":"192.168.44.108"}
11:06:46 MQT: DVES_ED76DA/tele/INFO3 = {"RestartReason":"Software/System restart"}
11:06:47 MQT: homeassistant/light/ED76DA_LI_1/config =  (retained)
11:06:47 MQT: homeassistant/switch/ED76DA_RL_1/config =  (retained)
11:06:47 MQT: homeassistant/light/ED76DA_LI_2/config =  (retained)
11:06:47 MQT: homeassistant/switch/ED76DA_RL_2/config =  (retained)
11:06:47 MQT: homeassistant/light/ED76DA_LI_3/config =  (retained)
11:06:47 MQT: homeassistant/switch/ED76DA_RL_3/config =  (retained)
11:06:47 MQT: homeassistant/light/ED76DA_LI_4/config =  (retained)
11:06:47 MQT: homeassistant/switch/ED76DA_RL_4/config =  (retained)
11:06:47 MQT: homeassistant/light/ED76DA_LI_5/config =  (retained)
11:06:47 MQT: homeassistant/switch/ED76DA_RL_5/config =  (retained)
11:06:47 MQT: homeassistant/light/ED76DA_LI_6/config =  (retained)
11:06:47 MQT: homeassistant/switch/ED76DA_RL_6/config =  (retained)
11:06:47 MQT: homeassistant/light/ED76DA_LI_7/config =  (retained)
11:06:47 MQT: homeassistant/switch/ED76DA_RL_7/config =  (retained)
11:06:47 MQT: homeassistant/light/ED76DA_LI_8/config =  (retained)
11:06:47 MQT: homeassistant/switch/ED76DA_RL_8/config =  (retained)
11:06:47 HASS: failed to parse 'DS18B20-1'
11:06:47 MQT: homeassistant/sensor/ED76DA_status/config = {"name":"nodeMCU Test status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"ED76DA_status","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]],"name":"nodeMCU Test","model":"NodeMCU V3","sw_version":"6.6.0.12(33baadf-sonoff)","manufacturer":"Tasmota"},"~":"DVES_ED76DA/tele/"} (retained)
11:06:48 HTP: Console
11:06:49 APP: Boot Count 37
11:06:49 CFG: Saved to flash at F9, Count 60, Bytes 4096
11:06:54 MQT: DVES_ED76DA/tele/STATE = {"Time":"2019-09-19T11:06:54","Uptime":"0T00:00:15","UptimeSec":15,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":52,"LinkCount":1,"Downtime":"0T00:00:05"}}
11:06:54 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-19T11:06:54","DS18B20-1":{"Id":"01144FC4C9AA","Temperature":20.6},"DS18B20-2":{"Id":"01144FCEEEAA","Temperature":20.5},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":20.5},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":20.4},"DS18B20-5":{"Id":"01144FEFDFAA","Temperature":20.3},"TempUnit":"C"}
11:07:04 WIF: Checking connection...
11:07:04 WIF: Connected
11:07:11 MQT: DVES_ED76DA/tele/HASS_STATE = {"Version":"6.6.0.12(33baadf-sonoff)","BuildDateTime":"2019-09-10T22:04:34","Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Module":"NodeMCU V3","RestartReason":"Software/System restart","Uptime":"0T00:00:32","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:05","MqttCount":1,"BootCount":37,"SaveCount":60,"IPAddress":"192.168.44.108","RSSI":"50","LoadAvg":19}
11:07:24 MQT: DVES_ED76DA/tele/STATE = {"Time":"2019-09-19T11:07:24","Uptime":"0T00:00:45","UptimeSec":45,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:05"}}
11:07:24 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-19T11:07:24","DS18B20-1":{"Id":"01144FC4C9AA","Temperature":20.6},"DS18B20-2":{"Id":"01144FCEEEAA","Temperature":20.5},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":20.4},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":20.5},"DS18B20-5":{"Id":"01144FEFDFAA","Temperature":20.3},"TempUnit":"C"}
11:07:24 WIF: Checking connection...
11:07:24 WIF: Connected

result of http://192.168.44.108/cm?cmnd=status%200

[... cut ...] .6},"DS18B20-2":{"Id":"01144FCEEEAA","Temperature":20.6},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":20.5},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":20.5},"DS18B20-5":{"Id":"01144FEFDFAA","Temperature":20.4},"TempUnit":"C"},"StatusSTS":{"Time":"2019-09-19T11:08:54","Uptime":"0T00:02:15","UptimeSec":135,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":50,"LinkCount":1,"Downtime":"0T00:00:05"}}}

effelle, 2019-09-19:
Puzzling. Can't replicate here since I have just 3 DS sensors. 
Please test this experimental build: do a `reset 2` and flash it over USB on your node. Configure it starting with the last know configuration (4 sensors), then change `seriallog` and `weblog` to level 4 and for last run `setoption19`.
Again, paste here all the lines about HA, including the debug lines starting with `SEN:` and `SUB:`.

[Hass_dbg.zip](https://github.com/arendst/Sonoff-Tasmota/files/3631053/Hass_dbg.zip)

qschneider, 2019-09-19:
Here we go... after reseting, flashing (witch 4 sensors attached) as you described I issued the following commands

changed SSID and PWD, save

configure other
  - template {"NAME":"NodeMCU V3","GPIO":[255,255,255,255,255,255,255,255,255,255,255,255,255],"FLAG":0,"BASE":18}
  - Activate checked
  - Friendly Name 1 nodeMCU Test
  - save

configure Module
  - D5 GPIO14 DS18x20 (4)
  - save
  
configure MQTT
  - Host 192.168.44.11
  - User test
  - Password test
  - topic DVES_%06X
  - save

configure Logging
  - serial log level 4
  - web log level 4
  - Telemetry period 30
  - save

console
  - setoption19 1

qschneider, 2019-09-19:
result was as expected...

00:00:05 HTP: Web server active on DVES_ED76DA-5850 with IP address 192.168.44.108
16:25:26 NTP: Drift 0, (UTC) Thu Sep 19 15:25:26 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
16:25:27 MQT: Attempting connection...
16:25:27 MQT: Connected
16:25:27 MQT: DVES_ED76DA/tele/LWT = Online (retained)
16:25:27 MQT: DVES_ED76DA/cmnd/POWER = 
16:25:27 MQT: Subscribe to DVES_ED76DA/cmnd/#
16:25:27 MQT: Subscribe to sonoffs/cmnd/#
16:25:27 MQT: Subscribe to cmnd/DVES_ED76DA_fb/#
16:25:27 MQT: DVES_ED76DA/tele/INFO1 = {"Module":"NodeMCU V3","Version":"6.6.0.12(sonoff)","FallbackTopic":"cmnd/DVES_ED76DA_fb/","GroupTopic":"sonoffs"}
16:25:27 MQT: DVES_ED76DA/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"DVES_ED76DA-5850","IPAddress":"192.168.44.108"}
16:25:27 MQT: DVES_ED76DA/tele/INFO3 = {"RestartReason":"Software/System restart"}
16:25:28 SEN: 'DS18B20-1'
16:25:28 SUB: 'Temperature'
16:25:28 MQT: homeassistant/sensor/ED76DA_DS18B20-2_Id/config = {"name":"nodeMCU Test DS18B20-2 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-2_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":" ","val_tpl":"{{value_json['DS18B20-2'].Id}}"} (retained)
16:25:28 SEN: 'DS18B20-2'
16:25:28 SUB: 'Id'
16:25:28 MQT: homeassistant/sensor/ED76DA_DS18B20-2_Temperature/config = {"name":"nodeMCU Test DS18B20-2 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-2_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-2'].Temperature}}","dev_cla":"temperature"} (retained)
16:25:28 SEN: 'DS18B20-2'
16:25:28 SUB: 'Temperature'
16:25:28 MQT: homeassistant/sensor/ED76DA_DS18B20-3_Id/config = {"name":"nodeMCU Test DS18B20-3 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-3_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":" ","val_tpl":"{{value_json['DS18B20-3'].Id}}"} (retained)
16:25:28 SEN: 'DS18B20-3'
16:25:28 SUB: 'Id'
16:25:28 MQT: homeassistant/sensor/ED76DA_DS18B20-3_Temperature/config = {"name":"nodeMCU Test DS18B20-3 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-3_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-3'].Temperature}}","dev_cla":"temperature"} (retained)
16:25:28 SEN: 'DS18B20-3'
16:25:28 SUB: 'Temperature'
16:25:28 MQT: homeassistant/sensor/ED76DA_DS18B20-4_Id/config = {"name":"nodeMCU Test DS18B20-4 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-4_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":" ","val_tpl":"{{value_json['DS18B20-4'].Id}}"} (retained)
16:25:28 SEN: 'DS18B20-4'
16:25:28 SUB: 'Id'
16:25:28 MQT: homeassistant/sensor/ED76DA_DS18B20-4_Temperature/config = {"name":"nodeMCU Test DS18B20-4 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-4_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-4'].Temperature}}","dev_cla":"temperature"} (retained)
16:25:28 SEN: 'DS18B20-4'
16:25:28 SUB: 'Temperature'
16:25:28 MQT: homeassistant/sensor/ED76DA_status/config = {"name":"nodeMCU Test status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"ED76DA_status","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]],"name":"nodeMCU Test","model":"NodeMCU V3","sw_version":"6.6.0.12(sonoff)","manufacturer":"Tasmota"},"~":"DVES_ED76DA/tele/"} (retained)
16:25:28 MQT: homeassistant/sensor/ED76DA_conn/config = {"name":"nodeMCU Test Connection","stat_t":"~STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","val_tpl":"{{[value_json.Wifi.SSId, value_json.Wifi.RSSI] | join (' (') + '%)' }}","dev_cla":"signal_strength","~":"DVES_ED76DA/tele/"} (retained)
16:25:30 APP: Boot Count 6
16:25:30 CFG: Saved to flash at F4, Count 16, Bytes 4096
16:25:35 MQT: DVES_ED76DA/tele/STATE = {"Time":"2019-09-19T16:25:35","Uptime":"0T00:00:15","UptimeSec":15,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":46,"LinkCount":1,"Downtime":"0T00:00:05"}}
16:25:35 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-19T16:25:35","DS18B20-1":{"Id":"01144FC4C9AA","Temperature":19.7},"DS18B20-2":{"Id":"01144FCEEEAA","Temperature":19.3},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":19.9},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":19.4},"TempUnit":"C"}

qschneider, 2019-09-19:
after that I disconnected the device, wire up the 5 sensor and started again.

Sadly, after that I got the same result as before....

00:00:05 HTP: Web server active on DVES_ED76DA-5850 with IP address 192.168.44.108
16:29:40 NTP: Drift 0, (UTC) Thu Sep 19 15:29:40 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
16:29:41 MQT: Attempting connection...
16:29:41 MQT: Connected
16:29:41 MQT: DVES_ED76DA/tele/LWT = Online (retained)
16:29:41 MQT: DVES_ED76DA/cmnd/POWER = 
16:29:41 MQT: Subscribe to DVES_ED76DA/cmnd/#
16:29:41 MQT: Subscribe to sonoffs/cmnd/#
16:29:41 MQT: Subscribe to cmnd/DVES_ED76DA_fb/#
16:29:41 MQT: DVES_ED76DA/tele/INFO1 = {"Module":"NodeMCU V3","Version":"6.6.0.12(sonoff)","FallbackTopic":"cmnd/DVES_ED76DA_fb/","GroupTopic":"sonoffs"}
16:29:41 MQT: DVES_ED76DA/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"DVES_ED76DA-5850","IPAddress":"192.168.44.108"}
16:29:41 MQT: DVES_ED76DA/tele/INFO3 = {"RestartReason":"Power on"}
16:29:42 MQT: homeassistant/light/ED76DA_LI_1/config =  (retained)
16:29:42 MQT: homeassistant/switch/ED76DA_RL_1/config =  (retained)
16:29:42 MQT: homeassistant/light/ED76DA_LI_2/config =  (retained)
16:29:42 MQT: homeassistant/switch/ED76DA_RL_2/config =  (retained)
16:29:42 MQT: homeassistant/light/ED76DA_LI_3/config =  (retained)
16:29:42 MQT: homeassistant/switch/ED76DA_RL_3/config =  (retained)
16:29:42 MQT: homeassistant/light/ED76DA_LI_4/config =  (retained)
16:29:42 MQT: homeassistant/switch/ED76DA_RL_4/config =  (retained)
16:29:42 MQT: homeassistant/light/ED76DA_LI_5/config =  (retained)
16:29:42 MQT: homeassistant/switch/ED76DA_RL_5/config =  (retained)
16:29:42 MQT: homeassistant/light/ED76DA_LI_6/config =  (retained)
16:29:42 MQT: homeassistant/switch/ED76DA_RL_6/config =  (retained)
16:29:42 MQT: homeassistant/light/ED76DA_LI_7/config =  (retained)
16:29:42 MQT: homeassistant/switch/ED76DA_RL_7/config =  (retained)
16:29:42 MQT: homeassistant/light/ED76DA_LI_8/config =  (retained)
16:29:42 MQT: homeassistant/switch/ED76DA_RL_8/config =  (retained)
16:29:42 HASS: failed to parse 'DS18B20-1' root
16:29:42 MQT: homeassistant/sensor/ED76DA_status/config = {"name":"nodeMCU Test status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"ED76DA_status","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]],"name":"nodeMCU Test","model":"NodeMCU V3","sw_version":"6.6.0.12(sonoff)","manufacturer":"Tasmota"},"~":"DVES_ED76DA/tele/"} (retained)
16:29:42 MQT: homeassistant/sensor/ED76DA_conn/config = {"name":"nodeMCU Test Connection","stat_t":"~STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","val_tpl":"{{[value_json.Wifi.SSId, value_json.Wifi.RSSI] | join (' (') + '%)' }}","dev_cla":"signal_strength","~":"DVES_ED76DA/tele/"} (retained)
16:29:44 APP: Boot Count 7
16:29:44 CFG: Saved to flash at FB, Count 17, Bytes 4096
16:29:49 MQT: DVES_ED76DA/tele/STATE = {"Time":"2019-09-19T16:29:49","Uptime":"0T00:00:15","UptimeSec":15,"Heap":28,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":40,"LinkCount":1,"Downtime":"0T00:00:05"}}
16:29:49 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-19T16:29:49","DS18B20-1":{"Id":"01144FC4C9AA","Temperature":19.7},"DS18B20-2":{"Id":"01144FCEEEAA","Temperature":19.3},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":19.9},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":19.4},"DS18B20-5":{"Id":"01144FEFDFAA","Temperature":19.4},"TempUnit":"C"}

effelle, 2019-09-19:
That's sad, but at least now I know where it happens. ;) 

@arendst Theo, seems the script stops on line 490 of `xdrv_12_home_assistant.ino`, because `!root.success()`. 
I build a bin with a different value for  `StaticJsonBuffer` and also `LOGSZ` on `sonoff.h`, both increased to 1024 for a quick test, hoping was something similar to #4354. 
I don't have enough DS18 to reproduce the issue. 
Any idea to solve this particular issue?

effelle, 2019-09-24:
@qschneider 
Quentin, I haven't forgotten you, just busy doing other stuff.
Perhaps I found something but since I'll get my Dallas just next week or so, I can't do more for now. 
Please flash this build and test it with `seriallog` and `weblog` set to `4`, thanks.
[firmware.zip](https://github.com/arendst/Sonoff-Tasmota/files/3647546/firmware.zip)

qschneider, 2019-09-25:
Hi, thank your for your continuing support, here are my findings...
setup as before, starting off with 4  sensors attached.
-
backlog template {"NAME":"NodeMCU","GPIO":[255,255,255,255,255,255,255,255,255,255,4,255,255],"FLAG":0,"BASE":18}; Module 0;
backlog SetOption3 1; mqtthost 192.168.44.11; mqttuser tasmota; mqttpassword xxx; topic DVES_%06X; powerretain 1;
backlog FriendlyName nodeMCU Test; TelePeriod 30; SerialLog 4; WebLog 4;
setoption19 1
-

result as expected...

00:00:05 HTP: Web server active on DVES_ED76DA-5850 with IP address 192.168.44.108
11:01:04 NTP: Drift 0, (UTC) Wed Sep 25 10:01:04 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
11:01:05 MQT: Attempting connection...
11:01:05 MQT: Connected
11:01:05 MQT: DVES_ED76DA/tele/LWT = Online (retained)
11:01:05 MQT: DVES_ED76DA/cmnd/POWER = 
11:01:05 MQT: Subscribe to DVES_ED76DA/cmnd/#
11:01:05 MQT: Subscribe to sonoffs/cmnd/#
11:01:05 MQT: Subscribe to cmnd/DVES_ED76DA_fb/#
11:01:05 MQT: DVES_ED76DA/tele/INFO1 = {"Module":"NodeMCU","Version":"6.6.0.12(sonoff)","FallbackTopic":"cmnd/DVES_ED76DA_fb/","GroupTopic":"sonoffs"}
11:01:05 MQT: DVES_ED76DA/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"DVES_ED76DA-5850","IPAddress":"192.168.44.108"}
11:01:05 MQT: DVES_ED76DA/tele/INFO3 = {"RestartReason":"Software/System restart"}
11:01:06 MQT: homeassistant/switch/ED76DA_RL_8/config =  (retained)
11:01:06 HASS: sensordata result: 'DS18B20-1' 
11:01:06 HASS: Ignoring Id sensor for 'DS18B20-1'
11:01:06 MQT: homeassistant/sensor/ED76DA_DS18B20-1_Id/config = {"name":"nodeMCU Test DS18B20-1 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-1_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/"} (retained)
11:01:06 MQT: homeassistant/sensor/ED76DA_DS18B20-1_Temperature/config = {"name":"nodeMCU Test DS18B20-1 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-1_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-1'].Temperature}}","dev_cla":"temperature"} (retained)
11:01:06 HASS: Ignoring Id sensor for 'DS18B20-2'
11:01:06 MQT: homeassistant/sensor/ED76DA_DS18B20-2_Id/config = {"name":"nodeMCU Test DS18B20-2 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-2_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/"} (retained)
11:01:06 MQT: homeassistant/sensor/ED76DA_DS18B20-2_Temperature/config = {"name":"nodeMCU Test DS18B20-2 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-2_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-2'].Temperature}}","dev_cla":"temperature"} (retained)
11:01:06 HASS: Ignoring Id sensor for 'DS18B20-3'
11:01:06 MQT: homeassistant/sensor/ED76DA_DS18B20-3_Id/config = {"name":"nodeMCU Test DS18B20-3 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-3_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/"} (retained)
11:01:06 MQT: homeassistant/sensor/ED76DA_DS18B20-3_Temperature/config = {"name":"nodeMCU Test DS18B20-3 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-3_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-3'].Temperature}}","dev_cla":"temperature"} (retained)
11:01:06 HASS: Ignoring Id sensor for 'DS18B20-4'
11:01:06 MQT: homeassistant/sensor/ED76DA_DS18B20-4_Id/config = {"name":"nodeMCU Test DS18B20-4 Id","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-4_Id","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/"} (retained)
11:01:06 MQT: homeassistant/sensor/ED76DA_DS18B20-4_Temperature/config = {"name":"nodeMCU Test DS18B20-4 Temperature","stat_t":"~SENSOR","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","uniq_id":"ED76DA_DS18B20-4_Temperature","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]]},"~":"DVES_ED76DA/tele/","unit_of_meas":"°C","val_tpl":"{{value_json['DS18B20-4'].Temperature}}","dev_cla":"temperature"} (retained)
11:01:06 MQT: homeassistant/sensor/ED76DA_status/config = {"name":"nodeMCU Test status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"ED76DA_status","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]],"name":"nodeMCU Test","model":"NodeMCU","sw_version":"6.6.0.12(sonoff)","manufacturer":"Tasmota"},"~":"DVES_ED76DA/tele/"} (retained)
11:01:08 APP: Boot Count 5
11:01:08 CFG: Saved to flash at F7, Count 13, Bytes 4096
[... cut ...]
11:06:43 MQT: DVES_ED76DA/tele/STATE = {"Time":"2019-09-25T11:06:43","Uptime":"0T00:05:45","UptimeSec":345,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":60,"LinkCount":1,"Downtime":"0T00:00:05"}}
11:06:43 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-25T11:06:43","DS18B20-1":{"Id":"01144FCEEEAA","Temperature":21.8},"DS18B20-2":{"Id":"01144FD2BEAA","Temperature":21.5},"DS18B20-3":{"Id":"01144FEE20AA","Temperature":21.7},"DS18B20-4":{"Id":"01144FEFDFAA","Temperature":21.6},"TempUnit":"C"}

qschneider, 2019-09-25:
after attaching the 5. sensor and replugging the device I got the same error message as before.

00:00:05 WIF: Connected
00:00:05 HTP: Web server active on DVES_ED76DA-5850 with IP address 192.168.44.108
11:08:18 NTP: Drift 0, (UTC) Wed Sep 25 10:08:18 2019, (DST) Sun Mar 31 02:00:00 2019, (STD) Sun Oct 27 03:00:00 2019
11:08:19 MQT: Attempting connection...
11:08:19 MQT: Connected
11:08:19 MQT: DVES_ED76DA/tele/LWT = Online (retained)
11:08:19 MQT: DVES_ED76DA/cmnd/POWER = 
11:08:19 MQT: Subscribe to DVES_ED76DA/cmnd/#
11:08:19 MQT: Subscribe to sonoffs/cmnd/#
11:08:19 MQT: Subscribe to cmnd/DVES_ED76DA_fb/#
11:08:19 MQT: DVES_ED76DA/tele/INFO1 = {"Module":"NodeMCU","Version":"6.6.0.12(sonoff)","FallbackTopic":"cmnd/DVES_ED76DA_fb/","GroupTopic":"sonoffs"}
11:08:19 MQT: DVES_ED76DA/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"DVES_ED76DA-5850","IPAddress":"192.168.44.108"}
11:08:19 MQT: DVES_ED76DA/tele/INFO3 = {"RestartReason":"External System"}
11:08:20 MQT: homeassistant/light/ED76DA_LI_1/config =  (retained)
11:08:20 MQT: homeassistant/switch/ED76DA_RL_1/config =  (retained)
11:08:20 MQT: homeassistant/light/ED76DA_LI_2/config =  (retained)
11:08:20 MQT: homeassistant/switch/ED76DA_RL_2/config =  (retained)
11:08:20 MQT: homeassistant/light/ED76DA_LI_3/config =  (retained)
11:08:20 MQT: homeassistant/switch/ED76DA_RL_3/config =  (retained)
11:08:20 MQT: homeassistant/light/ED76DA_LI_4/config =  (retained)
11:08:20 MQT: homeassistant/switch/ED76DA_RL_4/config =  (retained)
11:08:20 MQT: homeassistant/light/ED76DA_LI_5/config =  (retained)
11:08:20 MQT: homeassistant/switch/ED76DA_RL_5/config =  (retained)
11:08:20 MQT: homeassistant/light/ED76DA_LI_6/config =  (retained)
11:08:20 MQT: homeassistant/switch/ED76DA_RL_6/config =  (retained)
11:08:20 MQT: homeassistant/light/ED76DA_LI_7/config =  (retained)
11:08:20 MQT: homeassistant/switch/ED76DA_RL_7/config =  (retained)
11:08:20 MQT: homeassistant/light/ED76DA_LI_8/config =  (retained)
11:08:20 MQT: homeassistant/switch/ED76DA_RL_8/config =  (retained)
11:08:20 HASS: sensordata result: 'DS18B20-1' 
11:08:20 HASS: failed to parse 'DS18B20-1' root
11:08:20 MQT: homeassistant/sensor/ED76DA_status/config = {"name":"nodeMCU Test status","stat_t":"~HASS_STATE","avty_t":"~LWT","pl_avail":"Online","pl_not_avail":"Offline","json_attributes_topic":"~HASS_STATE","unit_of_meas":" ","val_tpl":"{{value_json['RSSI']}}","uniq_id":"ED76DA_status","device":{"identifiers":["ED76DA"],"connections":[["mac","84:F3:EB:ED:76:DA"]],"name":"nodeMCU Test","model":"NodeMCU","sw_version":"6.6.0.12(sonoff)","manufacturer":"Tasmota"},"~":"DVES_ED76DA/tele/"} (retained)
11:08:22 APP: Boot Count 6
11:08:22 CFG: Saved to flash at F6, Count 14, Bytes 4096
11:08:27 MQT: DVES_ED76DA/tele/STATE = {"Time":"2019-09-25T11:08:27","Uptime":"0T00:00:15","UptimeSec":15,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":52,"LinkCount":1,"Downtime":"0T00:00:05"}}
11:08:27 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-25T11:08:27","DS18B20-1":{"Id":"01144FC4C9AA","Temperature":85.0},"DS18B20-2":{"Id":"01144FCEEEAA","Temperature":21.8},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":21.5},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":21.8},"DS18B20-5":{"Id":"01144FEFDFAA","Temperature":21.6},"TempUnit":"C"}
11:08:37 WIF: Checking connection...
11:08:37 WIF: Connected
11:08:44 MQT: DVES_ED76DA/tele/HASS_STATE = {"Version":"6.6.0.12(sonoff)","BuildDateTime":"2019-09-24T13:49:54","Core":"STAGE","SDK":"2.2.2-dev(38a443e)","Module":"NodeMCU","RestartReason":"External System","Uptime":"0T00:00:32","WiFi LinkCount":1,"WiFi Downtime":"0T00:00:05","MqttCount":1,"BootCount":6,"SaveCount":14,"IPAddress":"192.168.44.108","RSSI":"56","LoadAvg":19}
11:08:57 MQT: DVES_ED76DA/tele/STATE = {"Time":"2019-09-25T11:08:57","Uptime":"0T00:00:45","UptimeSec":45,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"FBI","BSSId":"F0:B0:14:75:BF:2A","Channel":1,"RSSI":54,"LinkCount":1,"Downtime":"0T00:00:05"}}
11:08:57 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-25T11:08:57","DS18B20-1":{"Id":"01144FC4C9AA","Temperature":85.0},"DS18B20-2":{"Id":"01144FCEEEAA","Temperature":21.8},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":21.6},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":21.8},"DS18B20-5":{"Id":"01144FEFDFAA","Temperature":21.6},"TempUnit":"C"}
11:09:27 MQT: DVES_ED76DA/tele/SENSOR = {"Time":"2019-09-25T11:09:27","DS18B20-1":{"Id":"01144FC4C9AA","Temperature":85.0},"DS18B20-2":{"Id":"01144FCEEEAA","Temperature":21.8},"DS18B20-3":{"Id":"01144FD2BEAA","Temperature":21.6},"DS18B20-4":{"Id":"01144FEE20AA","Temperature":21.8},"DS18B20-5":{"Id":"01144FEFDFAA","Temperature":21.6},"TempUnit":"C"}

effelle, 2019-09-25:
Ok, I think we will talk again when I get my sensor then. Thanks for your feedback.

PR #6610 (2019-10-10): Fix for Multiple DS18 sensors under Home Assistant discovery
## Description:
This PR solves a problem when more than four Dallas sensors are present at the time of discovery. 
Also, `ID` is now an attribute of the `Temperature` sensor. The `ID` sensor will be created anyway leaving to the user the choice of the most suitable sensor for his configuration. 

[image]

**Related issue (if applicable):** fixes #6447

## Checklist:
  - [X] The pull request is done against the latest dev branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR.
  - [X] The code ch [... cut]
closed 2019-10-10 by arendst