arendst/Tasmota #6450: RF Sniffing not working

GatCode, 2019-09-18:
Hello Guys,
I just finished flashing my RF Bridge R2 V1.0 with 6.5.0 (because I had the same issue as: https://github.com/arendst/Sonoff-Tasmota/issues/3061 - last comment) and now everything works but I can't receive any RF codes even if the RF LED on the device flashes when I press a button on a RF device. I used the "RfRaw AAB155" command which is stated in the wiki. Like you can see on the screenshot, I get an valid answer but no codes.
Can someone help me out?

[image]

meingraham, 2019-09-18:
Please close this issue and read thoroughly the details regarding submitting an issue.

Our ability to provide assistance is greatly hampered since you did not take a few minutes to complete the  [troubleshooting template](https://github.com/arendst/Sonoff-Tasmota/blob/development/.github/ISSUE_TEMPLATE/Custom.md) with the requested information. The details requested potentially affect which options to pursue. The small amount of time you spend completing the template will also help the volunteers providing the assistance to you to reduce the time required to help you.

closed 2019-09-20 by andrethomas2

lobocobra, 2019-09-24:
After flashing, did you put back the switch on the board to the initial position?
=> I forgot that myself and wondered for 1h why I do not receive any codes...